mongoose-os-apps/shelly-homekit #468: Problems with Shelly 2.5 and wifi AP

AnakinSpain, 2021-02-07:
My Shelly 2.5 with firmware 2.7.3 doesn’t appear in my homekit nor scan lan. 

Then i decided to restore Wifi Shelly and stock firmware but i can not join to Shelly AP

it sounds clic-clac ( the relay) but wifi SSID doesn’t appear

i have two Shelly wirh the same problem

simonfield76, 2021-02-07:
It seems not correctly restored.

In my opinion you should flash original firmware via Ttl interface

Inviato da iPhone

AnakinSpain, 2021-02-07:
I have an TTL USB but I need a wire with a small connection to Shelly. Where I can buy it?

simonfield76, 2021-02-07:
I used the copper from an Ethernet cable (cat 6 as the copper is stronger)

Inviato da iPhone

AnakinSpain, 2021-02-07:
Finally I could update my shelly 2.5 from my MAC. I always do from my iPad or iPhone. But it responses very slow to load the web GUI.
`
235777527 shelly_main.cpp:554     Up 235.70, HAP 0/6/9 ns 6, RAM: 20684/41008; st 63; 4.1: c:0 mp:0.00 mt_ms:0 cp:0.00 tp:0.00 lemd:0 lhmd:0
243710632 shelly_main.cpp:554     Up 243.70, HAP 0/6/9 ns 6, RAM: 21328/41008; st 63; 4.1: c:0 mp:0.00 mt_ms:0 cp:0.00 tp:0.00 lemd:0 lhmd:0
1022526 mgos_rpc_channel_mq:203 0x3fff48b4 shellyswitch25-2D2745/rpc
1032014 mgos_rpc_channel_ua:313 0x3fff4c14 UART0
1055726 mgos_ade7953.c:96       ADE7953 silicon version: 0x02 (2)
1089681 shelly_main.cpp:304     === Creating accessories
1101925 shelly_hap_window_c:172 WC 1: not calibrated
1112187 shelly_main.cpp:331     === Starting HAP server (CN 6)
1180258 HAPPlatformTCPStrea:187 Listening on 9000
1244930 mgos_init.c:33          Init done, RAM: 37616 total, 23760 free, 23768 min free
1260469 esp_ota_backend.c:599   Committing ROM 1
1268298 esp_ota_backend.c:590   cur 1 prev 0 fwu 0
1333698 shelly_main.cpp:365     HAP server state: 1
1342344 mgos_mongoose.c:66      New heap free LWM: 23744
1354151 mgos_ota_core.c:1329    UID: 03c20312c21417f1, license: none
1368792 mgos_mdns.c:82          Listening on udp://:5353
1382003 mgos_mongoose.c:66      New heap free LWM: 21992
1390676 mgos_mdns.c:55          mDNS socket closed
1400204 mgos_mdns.c:82          Listening on udp://:5353
1410390 mgos_net.c:86           WiFi STA: connecting
1422604 mgos_mdns.c:55          mDNS socket closed
1432158 mgos_mdns.c:82          Listening on udp://:5353
4062858 esp_main.c:145          SDK: scandone
4957675 esp_main.c:145          SDK: state: 0 -> 2 (b0)
4978369 esp_main.c:145          SDK: state: 2 -> 3 (0)
4991031 esp_main.c:145          SDK: state: 3 -> 5 (10)
4998555 esp_main.c:145          SDK: add 0
5006040 esp_main.c:145          SDK: aid 14
5013620 esp_main.c:145          SDK: cnt 
5037717 esp_main.c:145          SDK: 
5047574 esp_main.c:145          SDK: connected with atorcha, channel 3
5060530 esp_main.c:145          SDK: ip:192.168.0.26,mask:255.255.255.0,gw:192.168.0.1
5072323 esp_main.c:145          SDK: ip:192.168.0.26,mask:255.255.255.0,gw:192.168.0.1
5089117 mgos_wifi.c:133         WiFi STA: Connected, BSSID e0:3f:49:ee:29:a8 ch 3 RSSI -53
5100017 mgos_net.c:90           WiFi STA: connected
5113350 mgos_net.c:101          WiFi STA: ready, IP 192.168.0.26, GW 192.168.0.1, DNS default
5128309 mgos_mongoose.c:66      New heap free LWM: 21336
5317554 mgos_mongoose.c:66      New heap free LWM: 20728
5563054 HAPPlatformTCPStrea:121 0x3fff69ec 192.168.0.189:49307 Incoming HAP connection, ns 0/0/12
5578126 HAPPlatformTCPStrea:276 0x3fff69ec 192.168.0.189:49307 Accepted HAP connection, ns 0/1/12 ts 0x3fff63f4
6821726 mgos_mongoose.c:66      New heap free LWM: 20552
6840664 HAPPlatformTCPStrea:121 0x3fff6ad4 192.168.0.248:49292 Incoming HAP connection, ns 0/1/12
6859424 mgos_mongoose.c:66      New heap free LWM: 19960
6873485 mgos_sntp.c:95          SNTP query to time.google.com
6882535 mgos_mongoose.c:66      New heap free LWM: 19208
6896053 mgos_mongoose.c:66      New heap free LWM: 18128
7099833 mgos_mongoose.c:66      New heap free LWM: 18056
7147019 mgos_mongoose.c:66      New heap free LWM: 17432
7212824 mgos_sntp.c:58          SNTP reply from 216.239.35.4: time 1612722495.942406, local 7.199557, delta 1612722488.742849
7407224:1612722496133444 HAPPlatformTCPStrea:121 0x3fff6744 192.168.0.193:51963 Incoming HAP connection, ns 1/1/12
7425953 HAPPlatformTCPStrea:121 0x3fff719c 192.168.0.242:57764 Incoming HAP connection, ns 2/1/12
7468202 mgos_mongoose.c:66      New heap free LWM: 16576
7509916 HAPPlatformTCPStrea:121 0x3fff779c 192.168.0.202:52486 Incoming HAP connection, ns 3/1/12
7748833 HAPPlatformTCPStrea:121 0x3fff6dc4 192.168.0.167:49266 Incoming HAP connection, ns 4/1/12
8386126 mgos_mongoose.c:66      New heap free LWM: 16280
8622543 HAPPlatformTCPStrea:276 0x3fff6ad4 192.168.0.248:49292 Accepted HAP connection, ns 4/2/12 ts 0x3fff69c4
9967938 shelly_main.cpp:543     Up 9.95, HAP 4/2/12 ns 2, RAM: 17620/37616; st 60; 4.1: c:0 mp:0.00 mt_ms:0 cp:0.00 tp:0.00 lemd:0 lhmd:0
10280546 mgos_mongoose.c:66      New heap free LWM: 16024
11700006 HAPPlatformTCPStrea:276 0x3fff6744 192.168.0.193:51963 Accepted HAP connection, ns 3/3/12 ts 0x3fff6d1c
14741418 HAPPlatformTCPStrea:276 0x3fff719c 192.168.0.242:57764 Accepted HAP connection, ns 2/4/12 ts 0x3fff773c
16068131 esp_main.c:145          SDK: pm open,type:0 0
16181420 HAPPlatformTCPStrea:105 0x3fff719c 192.168.0.242:57764 HAP connection closed, ns 2/4/12 i 2
16193258 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff773c nc 0x0 2/3/12
16424599 HAPPlatformTCPStrea:121 0x3fff7914 192.168.0.242:57765 Incoming HAP connection, ns 2/3/12
17200972 shelly_main.cpp:543     Up 17.19, HAP 3/3/12 ns 3, RAM: 18192/37616; st 60; 4.1: c:0 mp:0.00 mt_ms:0 cp:0.00 tp:0.00 lemd:0 lhmd:0
17785077 HAPPlatformTCPStrea:276 0x3fff779c 192.168.0.202:52486 Accepted HAP connection, ns 2/4/12 ts 0x3fff7b64
19118443 mgos_mongoose.c:66      New heap free LWM: 15336
19133583 HAPPlatformTCPStrea:105 0x3fff779c 192.168.0.202:52486 HAP connection closed, ns 2/4/12 i 2
[... 26 lines cut ...]
83414923 mgos_http_server.c:180  0x3fff6f0c HTTP connection from 192.168.0.193:52012
83447579 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.193:52012
84514805 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.193:52012
[... cut ...]
101886185 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.193:52015
102887960 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.193:52015
103889404 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.193:52015
104889514 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.193:52015
106092098 shelly_debug.cpp:203    Tailing
106294841 shelly_main.cpp:543     Up 106.28, HAP 0/6/12 ns 6, RAM: 17848/37616; st 60; 4.1: c:0 mp:0.00 mt_ms:0 cp:0.00 tp:0.00 lemd:0 lhmd:0
`

AnakinSpain, 2021-02-07:
[image]

AnakinSpain, 2021-02-07:
Now, after turn off and turn on again. I can see it 

[image]

AnakinSpain, 2021-02-07:
with 2.7.1 again:

It is conected to LAN but web GUI doesn't load.

AnakinSpain, 2021-02-07:
then I returned to 2.6.1 and few minutes later It blocked again. I press the button once and then the web GUI load, perfectly. Now I am waiting what happens in a few minutes

AnakinSpain, 2021-02-16:
This shelly must be briked or something. It respond too late. When i load the web gui it happens a few minutes later. Near router.

andyblac, 2021-03-03:
still an issue ?

AnakinSpain, 2021-03-07:
Yes. It is a problem still. I dont know how to reinstall firm again

AnakinSpain, 2021-03-08:
only load web config original firmware while I press the button on Shelly 2.5 I have to keep the button pressed if not, the web doesn't load

AnakinSpain, 2021-03-09:
https://youtu.be/bcMuouKD8sA

andyblac, 2021-03-09:
@rojer any ideas?

AnakinSpain, 2021-03-11:
Solved!!!

I was talking with Shelly assistance and I opened the case and wire antenna was wrong. When i close case it fails. Now i have TWO Shelly 2.5 witouth case and it works perfectly

Maybe someting wrong in case desing and wire wifi antenna

closed 2021-03-11 by AnakinSpain