mongoose-os-apps/shelly-homekit #479: Does this firmware work with shelly uni?

Ioan-Popovici, 2021-02-11:
I want to make a temperature and humidity switch with two channels out of it.
Is it possible to make it homekit compatible?

timoschilling, 2021-02-11:
Not at the moment but I will work on it in the next weeks.

Ioan-Popovici, 2021-02-19:
Thanks. No rush, I can use 2 shelly 1PMs, this was just an interesting idea. I want to control the temp and humidity in my fridge. I also have the shelly temp and humidity sensor so I can use that.

Thank you for you awesome work!

juanfal, 2021-04-07:
It would be fantastic shelly-homekit would work with shellyuni.  I've been struggling with shelly1 to read a LED from a AirConditioning wired remote control without interference and its just what shellyuni can do out of the box!!!

one4many, 2021-06-01:
Is it ready yet? ;-)
I would love to see support for the Uni as I have a few projects lined up for the garden (valves for the sprinkler etc).

rojer, 2021-06-02:
nope, not yet

tohmc, 2021-07-06:
Hi, it would be great if the UNI will have the garagedoor Funktion like Shelly 2.5.

many news on the implementation?

juanfal, 2021-07-07:
It would be really nice to have at least a template for it.  I understand it is quite open, and you can do crazy and not standard HomeKit recognisable things with it, but it is a pity having it out of the game.  If there were a template using it as a simple switch, say, and leaving comments where to tickle with it… It would be more than great

BadWolf42, 2021-08-14:
Hi, any news regarding this issue?
Thanks for your great work, I appreciate!

hfagelnour, 2021-12-21:
Up, plssssss

daitarn31974, 2022-01-21:
Interested too but it seems stuck

chef76, 2022-01-22:
Me toi

antaron1973, 2022-01-27:
@timoschilling: The Shelly Uni is really the only Shelly I can use for my use case. The HomeKit firmware is an extraordinary achievement and works great with my other Shelly 1. Is it possible also to support Shelly Uni?
Thanks, Maik.

MongooseFan, 2022-02-07:
I've purchased a Shelly UNI in order to interface with a Somfy IO transmitter (Izymo Transmitter) via dry contacts. It would be great if you guys supported the UNI! Thanks!

chef76, 2022-02-08:
You can use the dry contacts of the uni by Homekit Automation with http-request. Works perfekt!

MongooseFan, 2022-02-08:
Thanks, chef76. Could you provide more information? I don't understand what you mean exactly...   😕

chef76, 2022-02-08:
https://www.shelly-support.eu/forum/index.php?thread/775-collection-of-http-commands/

You can create an dummy-switch with hkknx and trigger an homekit automation, that calls the http request

PR #938 (2022-02-18): add support for Shelly UNI
closes #479

basic uni support with:
- 2 switches
- 2 inputs
- GDO mode

features for a later PR:
- temp sensors
- ADC
closed 2022-02-18 by rojer