mongoose-os-apps/shelly-homekit #487: update does not auto update

andyblac, 2021-02-12:
the update badge does not auto show hide on auto refresh
Ie

1. when on 2.7.2
2. open webUI
3. used my script to update
4. after device reboot
5. webui still shows “update”
6. refresh page
7. update badge gone

andyblac, 2021-02-13:
on reading the update code it seem it was done this way on purpose, @ZeevoX any reason that you only check once on manual page load ?

ZeevoX, 2021-02-13:
otherwise we were pinging rojer.me every second, it was slowing the autorefresh service down

andyblac, 2021-02-13:
ah ok, makes sense, i'll close this one then

rojer, 2021-02-13:
i think it makes sense to perform an update check if the version reported by the device has changed.
i.e. if Shelly.GetInfo returns a version different from the one that it was on the first update, UI should reset the update check timestamp and reload itself

andyblac, 2021-02-13:
@rojer like this https://github.com/mongoose-os-apps/shelly-homekit/pull/496

timoschilling, 2023-01-19:
@andyblac this is no longer reproducible for me. I think it was fixed as someone implemented a UI reload after device reboot. I would like to close this and the related pr, what do you think?

andyblac, 2023-01-20:
@timoschilling sure of you think it's fixed.

closed 2023-01-20 by andyblac