mongoose-os-apps/shelly-homekit #493: Improve Development Documentation

juangamnik, 2021-02-12:
I would like to add some developer documentation  describing the architecture in the wiki a llittlebit (so that one does not have to read the complete mongoose app but gets a good idea of how it works) and add some tests to the "Checks".

@rojer what do you think?

andyblac, 2021-02-12:
sure go for it. any help maintaining wiki i'd be greafull.

juangamnik, 2021-02-19:
I wait for the frontend rewrite with this too as this will influence the documentation.

louis49, 2021-03-08:
I am very interested and could make contributions. I am new to home automation and with MOS. Do you have a list of things to cover before you start to be able to help yourself?
Thank you

andyblac, 2021-03-08:
start here https://github.com/mongoose-os-apps/shelly-homekit/wiki/Development

closed 2021-04-16 by github-actions[bot]