mongoose-os-apps/shelly-homekit #494: More Testing

juangamnik, 2021-02-12:
I would like to add some end2end tests. I have not planned any details, yet. Is there interest?

rojer, 2021-02-13:
there's plenty of interest, for sure.
i think best start is to use the ubuntu port, it requires no additional hardware. after that we can consider adding tests on real devices.

juangamnik, 2021-02-13:
The idea is more or less to have some selenium/webdriver tests that try out some features. They can either be executed on a real device (configuring the target address of the web UI) or a simulated machine... is there a simulator (I do not know the capabilities of mongoose here, sorry)

rojer, 2021-02-13:
`make ShellyU` will build a binary that can be run on a Ubuntu linux system and represents a fully functioning device, except of course there are no real outputs. but web interface is fully functional (except for wifi) and the accessory can be discovered and added to the Home app. in addition, the binary is build with address sanitizer enabled, so it will catch leaks and other memory management bugs.

rojer, 2021-02-13:
in fact, i do most of my developemnt on the ShellyU port:

```
$ make ShellyU && rsync -av --exclude=*.gz fs_ShellyU/ build_ShellyU/fs/ && ASAN_SYMBOLIZER_PATH=$(realpath /usr/bin/llvm-symbolizer-*) LLVM_PROFILE_FILE=${PWD}/llvm.prof build_ShellyU/objs/*.elf --insecure --chroot build_ShellyU/fs
```

timoschilling, 2021-02-13:
Maybe we can use the GitHub action runner for Mac and use docker inside to run the ubuntu build. Then we can use the Mac home app to control the Ubuntu device.

juangamnik, 2021-02-13:
Sounds great, I will start with something easy and come back to you.

timoschilling, 2021-02-14:
Just a notice I’m currently working on a rewrite using the preact framework. Maybe this will effect your work. Maybe then we should wait with the tests?

juangamnik, 2021-02-14:
You are right. We are using react (and react native) with typescript a lot at work. I didn‘t know preact... sounds good. Please use Typescript and not plain JavaScript, ok? 😇

juangamnik, 2021-04-05:
@timoschilling is the frontend rewrite done?

timoschilling, 2021-04-05:
@juangamnik no work will continue in the next week