mongoose-os-apps/shelly-homekit #498: Testing 2.8.0

rojer, 2021-02-14:
Starting the 2.8 release cycle.

What are we testing:

 * Valve support (#433)
 * Doorbell support (#375)
 * Web UI changes, making sure stuff still works after cleanups and refactoring

Firmware: http://rojer.me/files/shelly/2.8.0-alpha1/

rojer, 2021-02-14:
@timoschilling one thing i noticed is that after the mobile layout change we have spinners overlap with button labels. it looks meh, can you please avoid that, at least in cases where we have enoughwidth and can afford it?

timoschilling, 2021-02-14:
👀

danny1605, 2021-02-14:
Hello i have extensively tested the 2.8.0-alpha1 version with a shelly 1 HAP Service Type Valve. I couldn't find any errors. Can I leave the alpha version installed until the current version comes out? Many thanks again for the migration

rojer, 2021-02-14:
thanks @danny1605 . leave it on the device, once 2.8.0 is released you will be able to update to it via regular update check.

andyblac, 2021-02-14:
valve type is limited to one device type, you should be able to choose in homekit sprinkler, valve, etc.

BlancoDeMallloca, 2021-02-15:
Installed the 2.8.0-alpha1. Where can I find the doorbell mode? I only see switch or garage opener mode.

andyblac, 2021-02-15:
> Installed the 2.8.0-alpha1. Where can I find the doorbell mode? I only see switch or garage opener mode.

it is located in "Detached" and in "Input1" as "Doorbell"

BlancoDeMallloca, 2021-02-15:
Thanks. Going to test it later.

andyblac, 2021-02-15:
Valve support seems ok for generic and irrigation, but Faucet and Shower Head do not work on my Shelly1, i just get no response.

BlancoDeMallloca, 2021-02-16:
Testes ist in a test environment and seems to work fine. No issues discovered.

emartife, 2021-02-18:
I have installed version 2.8 and I have configured the irrigation valve.
When I have added it to HomeKit it appears with the correct icon but it works as a switch.
In HomeKit, when you access the detail of the accessory, the default operating time does not appear and when it is working, the remaining watering time does not appear

timoschilling, 2021-02-18:
The time feature of valve is currently not supported.

emartife, 2021-02-19:
Thanks,
Will this functionality be in the final version 2.8?

timoschilling, 2021-02-19:
As it will require bigger changes on the current auto off implementation and on the UI, I don’t think so.

DerKoller-eu, 2021-02-21:
Hi, I tested the doorbell mode, works fine just for the HomeKit doorbell. If someone pushes the button outside, the HomePods ring and I get a push notification. The mechanical gong is not working anymore, it rings only after manually turning on the Switch that is presented in HomeKit. Here are my settings:
[image]

timoschilling, 2021-02-21:
@DerKoller-eu sounds wrong, but is actual expected behaviour. `Input Mode: Detached` means that there is not "connection" between the input and the output of the shelly.

Hotfix: create a HomeKit automatisation which turns on your "Shelly-Doorbell" in HomeKit when the Doorbell is triggered.

For the longterm we can add a explicit mode for that.

rojer, 2021-02-23:
pushed alpha2

rojer, 2021-02-23:
@andyblac roller shutter settings layout is a bit off after recent changes

[image]

rojer, 2021-02-23:
scratch that, pushed alpha3 with https://github.com/mongoose-os-apps/shelly-homekit/commit/20412b53325f8434ce62fede30a00e353fa7d3a1

lieberjott, 2021-02-23:
Quick question: Manual calibration for roller shutters (https://github.com/mongoose-os-apps/shelly-homekit/pull/449) won't be yet added?

andyblac, 2021-02-23:
no, it wont be added to 1.8.0, from what i understand, there is still work to be done on it.

rojer, 2021-02-23:
correct. won't make it into 2.8 but will be in 2.9

basroovers, 2021-03-03:
> @DerKoller-eu sounds wrong, but is actual expected behaviour. `Input Mode: Detached` means that there is not "connection" between the input and the output of the shelly.
>
> Hotfix: create a HomeKit [...]

I just tested the alpha3 version. Using an automation makes sense if you want something like a condition based on time of day (e.g. don't ring gong in the middle of the night). In order to get my gong sounding "properly" I needed to configure the auto off to a sub seconds delay.

rojer, 2021-03-06:
pushed alpha4. fixed layout for roller shutter section, plus networking changes in the backend (reverted to LWIP 1.4, reportedly it's more stable)

andyblac, 2021-03-06:
@everyone 2.8.0-alpha4 has connectivity issues, if you can not connect to device webui at all try a hard reboot (cut power), also DO NOT try to reflash alpha3, if you do, you will have to reset WiFi, as i found it doing so corupts wifi config.

after a hard reboot, webui can be reached, but it can be hit and miss just keep trying.

jeroendedeyne, 2021-03-07:
> @everyone 2.8.0-alpha4 has connectivity issues, if you can not connect to device webui at all try a hard reboot (cut power), also DO NOT try to reflash alpha3, if you do, you will have to reset [...]

jeroendedeyne, 2021-03-07:
Same issue here after updating to alpha 4

col-tom, 2021-03-07:
alpha4: If I set the shelly1 to „doorbell“ mode in the Dropdown this setting is lost after refreshing the web browser (probably it’s not correctly set in the UI?)

rojer, 2021-03-07:
connectivity issues were reported for alpha4, i pushed alpha5 with different networking stack configuration.

andyblac, 2021-03-07:
> connectivity issues were reported for alpha4, i pushed alpha5 with different networking stack configuration.

flashed to 2 devices, I'll keep an eye on it.

timoschilling, 2021-03-07:
> alpha4: If I set the shelly1 to „doorbell“ mode in the Dropdown this setting is lost after refreshing the web browser (probably it’s not correctly set in the UI?)

Can't confirm that, but get this error:
[image]

andyblac, 2021-03-07:
still having connectivity issues with beta5

col-tom, 2021-03-08:
>Can't confirm that, but get this error with the doorbell setting:

it's the dropdown where you set what it is (e.g. doorbell, motion, etc.) - that's always reset to the first item of the dropdown after a page refresh

timoschilling, 2021-03-08:
@col-tom after a reload:
[image]

pivale, 2021-03-08:
Lost http conectivity to most of my devices after alpha5. I can't access them by browser and update-script fails to connect.
I can ping them all though, so they are still alive and responsive.

andyblac, 2021-03-08:
> Lost http conectivity to most of my devices after alpha5. I can't access them by browser and update-script fails to connect.
> I can ping them all though, so they are still alive and responsive.
> 
> Power cycling didn't help...

and i bet homekit still works.

danny1605, 2021-03-08:
the same problem, a connection via browser is no longer possible.

andyblac, 2021-03-08:
can someone confirm something in alpha5, i think webui issues is linked with number of HAP connections, I can access webui, with no issues if HAP connections is 3 or less, as soon i hit 4 problems start, webui is hit and miss, the more HAP connections device has the worse it gets, over 5 and you loose tottal webui access.

col-tom, 2021-03-09:
@timoschilling ok, seems that is was an issue on my side - thanks anyway for looking into it

pivale, 2021-03-09:
> > Lost http conectivity to most of my devices after alpha5. I can't access them by browser and update-script fails to connect.
> > I can ping them all though, so they are still alive and responsive.
> > Power cycling didn't help...
> 
> and i bet homekit still works.

Yep, only http server seems to be affected... Homekit is still working

pivale, 2021-03-09:
> can someone confirm something in alpha5, i think webui issues is linked with number of HAP connections, I can access webui, with no issues if HAP connections is 3 or less, as soon i hit 4 problems start, webui is hit and miss, the more HAP connections device has the worse it gets, over 5 and you loose total webui access.

how can I help confirm that? currently all my shellys are unavailable (although working...)

andyblac, 2021-03-09:
> how can I help confirm that? currently all my shellys are unavailable (although working...)

reboot, open log screen in separate tab, keep an eye on it for connections

andyblac, 2021-03-09:
> @andyblac
> I can't see anything conspicuous, but the connection is lost
> [image]
> 
yup, 4 HAP connections, as i said above soon as i iit 4 connections i get issues.

rojer, 2021-03-09:
that's useful, thank you! i think i have an idea why

rojer, 2021-03-09:
Thanks to eagle eyed @andyblac this is now [fixed](https://github.com/mongoose-os-libs/lwip/commit/8725f4ecb2b8cee48760fe54dcb571257cb58194) - there was a limit on number of TCP connections, set to 5.
I pushed alpha6, please try it.

andyblac, 2021-03-09:
> Thanks to eagle eyed @andyblac this is now [fixed](https://github.com/mongoose-os-libs/lwip/commit/8725f4ecb2b8cee48760fe54dcb571257cb58194) - there was a limit on number of TCP connections, set to 5.
> I pushed alpha6, please try it.

so far so good, i'll keep an eye on it

pivale, 2021-03-10:
> Thanks to eagle eyed @andyblac this is now [fixed](https://github.com/mongoose-os-libs/lwip/commit/8725f4ecb2b8cee48760fe54dcb571257cb58194) - there was a limit on number of TCP connections, set to 5.
> I pushed alpha6, please try it.

Anyway we can reserve one connection always for webserver/script purposes?
It would prevent us from being left out, like now...

andyblac, 2021-03-10:
> Anyway we can reserve one connection always for webserver/script purposes?
>
> It would prevent us from being left out, like now...

HAP is limited to 12 connections, we have 20 TCP connections available now, no need too

pivale, 2021-03-10:
Cool, great to know 👍  Thanks for the info

rojer, 2021-03-11:
ok, i think 2.8 is ready. any objections?

andyblac, 2021-03-11:
> ok, i think 2.8 is ready. any objections?

all good here.

danny1605, 2021-03-11:
> ok, i think 2.8 is ready. any objections?

all of my 20 shellys work very well with the alpha 6 version. thank you for that

pivale, 2021-03-11:
All seems to be working as intended here with alpha6.
Only strange behaviour was my shelly plug S going into AP mode (1st time it happened)... Not sure it's even related

andyblac, 2021-03-11:
i'll close this one, now 2.8.0 has been released.

closed 2021-03-11 by andyblac

rojer, 2021-03-11:
@pivale 
```
59453953491 shelly_input.cpp:76     Input 0: change (state 1)
59460215133 shelly_main.cpp:554     Up 59460.20, HAP 0/4/12 ns 4, RAM: 22588/38616; st 30; 0.1: st:1 in_st:-1 inm:-1 ininv:0
59463964293 shelly_input.cpp:76     Input 0: long (state 0)
59463973460 shelly_main.cpp:216     0: Reset sequence detected
```

looks like noise on button input.

rojer, 2021-03-11:
submitted a fix, this will become 2.8.1. damn, i should've looked first

rojer, 2021-03-11:
@timoschilling i'm flipping logic of https://github.com/mongoose-os-apps/shelly-homekit/commit/023159de9a677b4d246e82f8e0f0832e7e15b7ee back to "noisy button by default".

andyblac, 2021-03-11:
> submitted a fix, this will become 2.8.1. damn, i should've looked first

haha, i know whats like 🤣

timoschilling, 2021-03-11:
@rojer as it's still configurable, it's fine for me. Maybe would you like to document why the noisy button is need and on which device.

rojer, 2021-03-11:
rather, where it was observed. makes sense.

ok, pushed 2.8.1. i think 2.8.0 was the shortest lived release :)