mongoose-os-apps/shelly-homekit #501: Better WiFi configuration

rojer, 2021-02-14:
We need a better, more advanced WiFi configuration.
1. We should allow configuring static IP settings: IP, netmask, gw
2. There should be up to 3 independent wifi configs. Mongoose OS already supports this, there are 3 identical sections: wiif.sta, wiif.sta1 and wifi.sta2 that will be tried one after another until a working one is found.

Suggested UI: 3 identical tabs in the WiFi section, 1 for each config (representing wifi.sta, wifi.sta1 and wifi.sta2 respectively), containing SSID, password, static IP checkbox that opens IP+netmask and gw if checked. tab that represents wifi currentky connected to (if any) should open by default, otherwise open the first on.

cc @timoschilling @ZeevoX

col-tom, 2021-02-15:
What about adding also a better, more understandable WiFi quality display like bad, good, .... (not just some negative numbers)

timoschilling, 2021-02-15:
> What about adding also a better, more understandable WiFi quality display like bad, good, .... 

@rojer if you have some range suggestions I can do that.

ZeevoX, 2021-02-15:
Taking Android as an example, it uses a range of -55 to -100 and splits it up evenly between 5 levels (zero to four bars) – https://android.googlesource.com/platform/frameworks/base/+/ccbf84f/wifi/java/android/net/wifi/WifiManager.java#944

timoschilling, 2021-02-16:
we have already a definition https://github.com/mongoose-os-apps/shelly-homekit/wiki/WiFi-Settings

andyblac, 2021-02-17:
I have been digging around in stock code on shelly device and found te following
```
                if (status) {
                    if (typeof Shelly.currentStatus == 'undefined')
                        return;
                    if (typeof Shelly.currentStatus.wifi_sta == 'undefined')
                        return;
                    var rssi = Shelly.currentStatus.wifi_sta.rssi;
                    if (rssi > -67)
                        $('#status').addClass('on');
                    else if (rssi > -75)
                        $('#status').addClass('on2');
                    else
                        $('#status').addClass('on1');
                } else {
                    $('#status').addClass('off');
                    $('#status').addClass('Access Point WiFi mode');
                }
```

rojer, 2021-02-20:
seems reasonable to follow stock in this case, that's what users are familiar with.

rojer, 2021-12-17:
implemented in 2.11.0-alpha2, see #868

closed 2021-12-17 by rojer