mongoose-os-apps/shelly-homekit #505: firmware version 2.7.2 shelly 2.5 trouble with chain actuators windows - stock firmware work well

gate546, 2021-02-15:


gate546, 2021-02-15:
firmware version 2.7.2 shelly 2.5 trouble with chain actuators windows - stock firmware work well 
just for to be sure I try many older firmware and the problem seem to be the same.
with original fw all work well.
For some reasons the calibration process don't run well, the process start and stop continuosly

andyblac, 2021-02-15:
Take a lot here, if this sounds your issue, please close thus issue, and chat in that one, if not please give more information about your problem.

gate546, 2021-02-15:
> Take a look here #404, if this sounds your issue, please close this issue , and chat in that one, if not please give more information about your problem.

OK 
You are right, the issue is the same.

andyblac, 2021-02-15:
ok i'll close this issue.

gate546, 2021-02-18:
After some test with fork of @juangamnik and some post on #404,  juangamnik suggets me to reopen this issue.
Seem that the solution will be to enhance the calibration of homelit firmware instead of adding a manual calibration.
I don't know if this is the correct way to proceed, however I repeat here my observations that I have already written in # 404.
This is what happens to me and what I understand.
I have read in some posts that the problem of incorrect calibration with chain drives is due to a current draw problem, as if there was no current draw.
According to my observations, the problem is different, I will try to share what I see.
I have installed many Shelly 2.5 PMs to operate as roller shutters, all installations work very well with the original firmware, both way,  when Shelly operates the motor directly, in the case of roller shutters, and when Shelly operates by the way of intermediate automatism, as in the case of chain drives.
With the HomeKit firmware, on the other hand, this situation change, when the motors are activated directly everything is fine, when operate on a chain system the calibration does not work.
I have observed that my chain drive has a little delay from when the push button is pressed to when the motor starts moving, this is likely the problem.
The calibration function, when the process start, expects to immediately see an absorption of current and instead this it does not happen because, due to the electronics inside the drive, the current is seen after a while, in my case I think more or less a second, and this make that the process fails.
Since the original firmware works well, and observing the operation of the chain mechanism, I suppose that inserting a delay in the code that read the current absorption during the initial calibration process maybe will address this issue.

juangamnik, 2021-02-18:
> Seem that the solution will be to enhance the calibration of homekit firmware instead of adding a manual calibration.

That may be.
@contributors, please have a look whether this makes sense to you. My understanding is, that an enhancement of the calibration might be the better solution.

gate546, 2021-02-18:
Thak you 

Why seem that the issues #505 is CLOSED?
 
Is my browser trouble?

closed 2021-02-21 by andyblac