mongoose-os-apps/shelly-homekit #506: Shelly 1 sporadically no longer accessible - power interruption necessary.

Marcus-1976, 2021-02-15:
Hello,
I am brand new here in the forum. 
Maybe you can help me.
I have some Shelly 1 and ShellyPlug in use.
They are all flashed with the Homekit firmware you programmed.
The current firmware 2.7.3 is installed on all Shellys.

Unfortunately, one Shelly 1 keeps giving me problems.
I have already replaced it, but the same thing happens with the new one.

Shelly 1 is connected to the heating valve in our flat and controls it, depending on the HomeKit automation.

At some point, the Shelly 1 is simply no longer accessible via HomeKit. It can also no longer be reached via the fixed IP address. 
The AccessPoint from Fortgate is about 4 m in sight of it.
The Shelly 1 shows this in the configuration: 
RSSI: -63Host: shelly1-8CAAB5553DCD.localIP: 10.30.0.116

To make the Shelly 1 accessible again, I have to disconnect it briefly from the power supply. Then it works again. 
I can't see any connection as to when it is no longer accessible. It happens sporadically. 

Do you have any idea what is going wrong?

I have about 25 Shellys in use, all with fixed IP addresses, and this doesn't happen with any of them.
The previously installed Shelly, which was initially connected to the heating valve, is now installed for the kitchen lighting - it has not hung up since then.
So it can't be due to the hardware.

Thank you 

Regards
Marcus

Translated with www.DeepL.com/Translator (free version)

andyblac, 2021-02-15:
sounds like a wifi issue, but -65 should not casuing any issues, is there any differant about that location ?, like differant access point, anything that could be causing interference ?

Marcus-1976, 2021-02-15:
Hello andyblac,

thank you for your quick reply.
It is indeed only the one Fortigate access point that is installed centrally in the flat. There is no other access point. The AP sends out a WLAN SSID for 2.4GHz and 5GHz.
Many of the other Shellys are much further away from the AP and always work without problems.

The previous Shelly, which was installed on the heating valve, also always showed the identical problem. It had the IP address: 10.30.0.132.
I have a free range from 10.30.0.110 to 10.30.0.149 for the HomeKit devices, so there are no overlaps or duplicate IP addresses.

I am absolutely at a loss, especially as the automation itself does not work reliably in the Home app.
The thermometer used is the QingPing Temp & RH in the H version.
The heating should heat up to 22 degrees by automation daily from Monday to Friday at 5:30 a.m. and switch off at 10:00 p.m., but not drop below 18 degrees temperature.
Heating during the day also works without any problems, as does switching off at 22:00, but starting during the week at 05:30, as well as at the weekend from 07:30, never works. At one point it worked, but for weeks it hasn't worked at all. I have to start the heating manually every time the alarm goes off at 06:00.

Translated with www.DeepL.com/Translator (free version)

simonfield76, 2021-02-15:
You could try setting a second ssid only with 2.4 ghz active if your router has this option.

Inviato da iPhone

Marcus-1976, 2021-02-15:
Hello simonfield76,

unfortunately I can't set different SSIDs for 2.4GHz and 5GHz, the AP doesn't support that.

But if that were the problem, other shellys should also cause problems?

The next Shelly, which is actually installed right next to the heating Shelly, is the Shelly that controls the light in the kitchen and always does so without any problems.
The two Shellys are installed about 30 cm next to each other.

Translated with www.DeepL.com/Translator (free version)

andyblac, 2021-02-15:
@Marcus-1976 is there anything close to the device that could be casuing interearance ?, as @simonfield76 seperating the 5Ghz and 2.4Ghz would help out shellys, it has been known to casue issues, but as you say it would be happing to the other shelly devices you have.

it definatly sounds like wifi issues.

Marcus-1976, 2021-02-15:
Until last week, the heating shelly was installed in a flush-mounted box (switch box) directly next to the light shelly, not 3 cm apart.
Now the heating Shelly is about 30 cm away from the light Shelly, but the situation has not changed.
As I said, changing the shelly hardware didn't help either.
The following Shellys are installed in the kitchen:
- Light Shelly (3 automations are controlled with it),
- Heating Shelly (only controlled by the Home app),
- 2x Shelly for kitchen cupboard lighting (is also controlled by the light Shelly).
- 2x Shelly for two different sockets (devices that draw standby current)
- 1x Shelly Plug for the coffee machine

The previous heating Shelly is still installed in the kitchen, so this is one of the two that now controls the kitchen cupboard lighting. 

If it's a wifi problem, I don't understand why some other shelly isn't causing these problems.

I'm still a beginner in the HomeKit matter, only started learning about it around mid January 2021. 
Unfortunately, the problem with the heating shelly has existed since the beginning.

andyblac, 2021-02-15:
is it near your microwave oven it could be that.

andyblac, 2021-02-15:
something near the device it causing interfearance to the wifi signal.

Marcus-1976, 2021-02-15:
No, no microwave.
Last week there was only a building site, no kitchen or kitchen furniture was installed at all, because we got a new kitchen on Friday, and it happened again that the Shelly was suddenly no longer accessible.

andyblac, 2021-02-15:
> No, no microwave.
> Last week there was only a building site, no kitchen or kitchen furniture was installed at all, because we got a new kitchen on Friday, and it happened again that the Shelly was suddenly no longer accessible.

sounds like you may need an aditional access point near that area.

Marcus-1976, 2021-02-15:
The access point is right next to the kitchen door. 
The other Shellys, which are distributed throughout the flat, have far worse network connections, see here: 

Sleeping Room:
RSSI: -71Host: shelly1-483FDA91AEA7.localIP: 10.30.0.129
RSSI: -81Host: shellyplug-s-B48A93.localIP: 10.30.0.135
 RSSI: -80Host: shelly1-483FDA9187CA.localIP: 10.30.0.136

And these always work.

jsiegenthaler, 2021-02-16:
Try reducing the WiFi signal strength by 19dB. Either just at the problem Shelly1 or centrally. You can do this by loosely covering it with tin foil. Make sure all terminals and wiring are insulated to avoid contact.

spiegelt, 2021-02-25:
Since firmware 2.7.3 I have the same problems with my five Shellys1s, these are on three different access points partly with very good signal quality (-37) and partly less good signal quality (-76).  Within 12 hours you can no longer be reached in the HomeKit and the web interface only opens very slowly after the reboot everything is fine again for a few hours in the previous five versions it was much better.

Marcus-1976, 2021-03-02:
Hello all,

I'm getting back to you with my initial problem with the Shelly switching my heating valve.
I have installed the Shelly in an electrical junction box.
I have now wrapped this with aluminium foil and it has not crashed for a week.

Unfortunately, what has been happening since I restarted my W-LAN access point is that every now and then one or more Shellys simply hang up and reboot by themselves.
Sometimes, however, I have to help by briefly "switching" the switching contact of the Shellys - otherwise the Shellys obviously don't boot on their own.
They are then simply no longer accessible - in Homekit - no connection.

Regards
Marcus

Translated with www.DeepL.com/Translator (free version)

closed 2021-04-10 by github-actions[bot]