mongoose-os-apps/shelly-homekit #507: Reverse status in HomeKit

kamilpop, 2021-02-15:
I have several Shelly 2.5 installed as blind control on the window and a sun awning. In HomeKit, these are displayed as active or open when retracted. But this should be inactive (gray). How can I change it?

[image]

[image]
[image]

andyblac, 2021-02-16:
have you tried swap outputs ?

closed 2021-02-16 by andyblac

jankornblum, 2022-07-17:
This is great! Thanks for this really excellent and helpful firmware at all :)