mongoose-os-apps/shelly-homekit #510: [Valve] Faucet and Shower Head broken

andyblac, 2021-02-15:
as reported here https://github.com/mongoose-os-apps/shelly-homekit/issues/498#issuecomment-779412674

timoschilling, 2021-02-16:
I can reproduce this, but I can't fix it. Maybe @rojer can take a look or we drop the support for "Shower Header" and "Water Faucet".

rojer, 2021-02-16:
@timoschilling can you elaborate on the issue?

timoschilling, 2021-02-16:
The only thing I can say is:
1. peering a valve in shower or faucet mode didn't worked
2. peering a valve in generic or irrigation mode worked 
3. after changing from a working generic or irrigation to shower or faucet has the following effects:
a. The valve type (characteristic) didn't switched (can not be seen in the Home app, but in the Home+ app, maybe in others too)
b. If I tried to turn on the shower or faucet it switches from "off" to "starting..." and then after some time to "not responding"
The switch from "off" to "starting" is expect, but then it should switch from "starting" to "running". If I do the same with a status on generic, switching that to shower and then turn it off, results in "running" -> "stopping", here is last step to "off" is missing.

The behaviour of a valve is a bit like the one of a garage door opener, it has active similar to target door state and in use similar to current target state.
```
!active && !inuse == off
active && !inuse == starting
active && inuse == running
!active && inuse == stopping
!active && !inuse == off
```

The behaviour of the UI looks like an issue with the InUser characteristic.

PR #523 (2021-02-22): remove support for valve type Shower Head and Water Faucet
closes #510, I can't find the problem, so I provide a fix which removes the support for valve type Shower Head and Water Faucet. Generic Valve and Irrigation will still work.

If no one can fix #510, we can remove the support of the non working types to unblock v2.8.0.
closed 2021-02-22 by rojer