mongoose-os-apps/shelly-homekit #511: Shelly i3 with PIR sensor

Ludelag, 2021-02-16:
Hello, I just wold like to ask if someone has tasted the shelly i3 with a PIR sensor using Mongose OS App, if yes how it works and wiring schema? any recomendations of PIR sensor or could be use any?
Thank you

rojer, 2021-02-20:
the "motion sensor" mode was developed specifically to accommodate the two common PIR sensors - "level" output and "pulse" output types. "level" output types close the output circuit and keep it closed while motion is detected, pulse output type sensors send pulses at regular intervals while motion is present and cease to do so when there is no motion.
both types, in normally closed and normally open configurations are supported
https://github.com/mongoose-os-apps/shelly-homekit/wiki/Input-Switch-Settings#motion-sensor--occupancy-sensor--contact-sensor--doorbell

rojer, 2021-02-20:
for normally closed sensors enable the "inverted input" option.

please reopen if you encounter any issues.

closed 2021-02-20 by rojer