mongoose-os-apps/shelly-homekit #515: Device getting warmer compared to original firmware (1+2.5)

janbl, 2021-02-16:
My feeling is that the power consumption in idle mode is higher compared to the original firmware. I did not measure it but the device is getting a lot warmer.
Can we reduce the activity or is it bound to the home kit communication?
I have tested with Shelly 1 and Shelly 2.5

andyblac, 2021-02-16:
i have no issues with temps on 2.5
[image]

what are yours at?

rojer, 2021-02-16:
yeah, i'd like to see some evidence first. homekit handshake is computationally intensive, but once it's done and connection is idle, it shouldn't be doing much.
that said, we do not put CPU or wifi radio into any kind of power saving mode but as far as i know, neither does stock fw.

chstoll, 2021-02-16:
I do have one which is always above 70°C : 
[image]
This one runs in switch mode. All my other Shelly 2.5 run in Rollershutter mode and are around 60-65°C. The are all mounted behind switches inside the wall.

andyblac, 2021-02-16:
my mine are inside the cavity, having said that, I think your isssue is location more than anything.

janbl, 2021-02-17:
Mine are around 60°C outside the wall and 65° inside. It's winter and they are inside the outer wall. I am concerned about summer temperatures.
But for me the questions is about the difference to the original firmware. In comparison it is 10-15°C warmer. Which is a lot in electronics.

col-tom, 2021-02-17:
What is the worse what could happen if temp goes up and up? fire?
Is this the same for shelly1 (without temp sensor) ?

andyblac, 2021-02-17:
> What is the worse what could happen if temp goes up and up? fire?
> Is this the same for shelly1 (without temp sensor) ?

it has a safety cut off iirc, @rojer can confirm temps but iirc it around 90˙c

found it: https://github.com/mongoose-os-apps/shelly-homekit/blob/21c70ee330351ea6ddf46d5361e5a0260e2a2f59/mos.yml#L38

col-tom, 2021-02-17:
Also for shelly1 ?

andyblac, 2021-02-17:
> Also for shelly1 ?

dont think so, shelly1 does not have hardware needed.

col-tom, 2021-02-17:
and what happens then? 😀

andyblac, 2021-02-17:
thats one for @rojer

rojer, 2021-02-20:
there is no thermal cutoff for shelly1 because there is no NTC on board to measure the temperature.

timoschilling, 2021-02-20:
What does ntc stands for?

andyblac, 2021-02-20:
> What does ntc stands for?

> NTC stands for “Negative Temperature Coefficient. NTC thermistors are resistors with a negative temperature coefficient, which means that the resistance decreases with increasing temperature. They are primarily used as resistive temperature sensors and current-limiting devices. The temperature sensitivity coefficient is about five times greater than that of silicon temperature sensors (silistors) and about ten times greater than those of resistance temperature detectors (RTDs). NTC sensors are typically used in a range from −55°C to 200°C.

romanbalsiger, 2021-02-21:
so far i have deployed 5 shellies:
1 shelly 1 behind a wall switch
2 shelly 2.5 behind a wall switch in switch mode - one at 66°C the other at 68°C
1 shelly 2.5 on the mains panel in switch mode - this one is at 72°C!
1 shelly i3 behind a wall switch - 83°C!

i have never checked with original firmware

rojer, 2021-02-21:
shelly i3 does tend to get warmer, i've noticed that.

> i have never checked with original firmware

please do, otherwise there is nothing to discuss here.  these values are well within safety range.
even then, you need to use thermocouple to measure actual temperature as stock and hk firmware may report slightly different temperature values as they use somewhat different NTC calculation algorithm.

rojer, 2021-02-21:
i'm closing the issue, discussion is wandering about without any concrete comparisons to stock.
absolute numbers are within range, it's ok.

closed 2021-02-21 by rojer

romanbalsiger, 2021-02-21:
where do i see temperatures in the original firmware...?

chstoll, 2021-02-24:
The shelly I mentioned earlier is getting warmer:
[image]

I rebootet it, but it makes no difference.

tux011, 2021-04-03:
I just want to confirm that the FW 20210315-093826/2.8.2-gc0f7426-master gets hotter compared to the Stock FW.
Shelly 2.5 with stock FW consumes 0.8W in standby and the FW 2.8.2 from here consumes 1.2W - under the same conditions.
Idle - both relays off....

timoschilling, 2021-04-03:
@tux011 do you habe the web ui Open during the test time?

chstoll, 2021-05-01:
[image]

I think I found the problem. I use this shelly to remote control an other roller shutter shelly. So I set this in normal switch mode an then created 4 automations in Homekit (If switch 1 gets triggerd on or off then open roller shutter, if switch 2 gets triggerd on or off then close roller shutter). But those automations seem to penetrate the shelly permanatly (polling for the state?) and the cpu temparuter gets so high that the shelly is failing (I think i maybe roasted this one, even with stock firmware it isn't stable any more). Is there some other way to remote control a roller shutter shelly from another switch? It is a shutter from a roof window which I want to control from downstairs with a physical switch.

rojer, 2021-05-02:
only extreme amount of traffic would be able to produce this kind of thing... it'd be interesting to see debug log from this shelly.

chstoll, 2021-05-05:
I set this up again on another shelly (only for a short period of time) but the logfile is only the same line over and over again:

376036576827 shelly_main.cpp:569     Up 376036.56, HAP 0/3/12 ns 3, RAM: 20484/38216; st 80; 0.1: st:1 in_st:0 inm:0 ininv:0; 0.2: st:1 in_st:0 inm:0 ininv:0
376044577257 shelly_main.cpp:569     Up 376044.56, HAP 0/3/12 ns 3, RAM: 20484/38216; st 80; 0.1: st:1 in_st:0 inm:0 ininv:0; 0.2: st:1 in_st:0 inm:0 ininv:0
376052576899 shelly_main.cpp:569     Up 376052.56, HAP 0/3/12 ns 3, RAM: 20484/38216; st 80; 0.1: st:1 in_st:0 inm:0 ininv:0; 0.2: st:1 in_st:0 inm:0 ininv:0
376060576864 shelly_main.cpp:569     Up 376060.56, HAP 0/3/12 ns 3, RAM: 20484/38216; st 80; 0.1: st:1 in_st:0 inm:0 ininv:0; 0.2: st:1 in_st:0 inm:0 ininv:0
376068576924 shelly_main.cpp:569     Up 376068.56, HAP 0/3/12 ns 3, RAM: 20484/38216; st 80; 0.1: st:1 in_st:0 inm:0 ininv:0; 0.2: st:1 in_st:0 inm:0 ininv:0
376076577113 shelly_main.cpp:569     Up 376076.56, HAP 0/3/12 ns 3, RAM: 20852/38216; st 80; 0.1: st:1 in_st:0 inm:0 ininv:0; 0.2: st:1 in_st:0 inm:0 ininv:0
376084577227 shelly_main.cpp:569     Up 376084.56, HAP 0/3/12 ns 3, RAM: 20852/38216; st 80; 0.1: st:1 in_st:0 inm:0 ininv:0; 0.2: st:1 in_st:0 inm:0 ininv:0
376092576896 shelly_main.cpp:569     Up 376092.56, HAP 0/3/12 ns 3, RAM: 20852/38216; st 80; 0.1: st:1 in_st:0 inm:0 ininv:0; 0.2: st:1 in_st:0 inm:0 ininv:0
376100576899 shelly_main.cpp:569     Up 376100.56, HAP 0/3/12 ns 3, RAM: 20852/38216; st 80; 0.1: st:1 in_st:0 inm:0 ininv:0; 0.2: st:1 in_st:0 inm:0 ininv:0
376108577347 shelly_main.cpp:569     Up 376108.56, HAP 0/3/12 ns 3, RAM: 20852/38216; st 80; 0.1: st:1 in_st:0 inm:0 ininv:0; 0.2: st:1 in_st:0 inm:0 ininv:0
376116577341 shelly_main.cpp:569     Up 376116.56, HAP 0/3/12 ns 3, RAM: 20852/38216; st 80; 0.1: st:1 in_st:0 inm:0 ininv:0; 0.2: st:1 in_st:0 inm:0 ininv:0
376124577131 shelly_main.cpp:569     Up 376124.56, HAP 0/3/12 ns 3, RAM: 20852/38216; st 80; 0.1: st:1 in_st:0 inm:0 ininv:0; 0.2: st:1 in_st:0 inm:0 ininv:0
376132576852 shelly_main.cpp:569     Up 376132.56, HAP 0/3/12 ns 3, RAM: 20852/38216; st 80; 0.1: st:1 in_st:0 inm:0 ininv:0; 0.2: st:1 in_st:0 inm:0 ininv:0
376140576905 shelly_main.cpp:569     Up 376140.56, HAP 0/3/12 ns 3, RAM: 20852/38216; st 80; 0.1: st:1 in_st:0 inm:0 ininv:0; 0.2: st:1 in_st:0 inm:0 ininv:0
376148576893 shelly_main.cpp:569     Up 376148.56, HAP 0/3/12 ns 3, RAM: 20852/38216; st 80; 0.1: st:1 in_st:0 inm:0 ininv:0; 0.2: st:1 in_st:0 inm:0 ininv:0
376156576806 shelly_main.cpp:569     Up 376156.56, HAP 0/3/12 ns 3, RAM: 20852/38216; st 80; 0.1: st:1 in_st:0 inm:0 ininv:0; 0.2: st:1 in_st:0 inm:0 ininv:0
376164577611 shelly_main.cpp:569     Up 376164.56, HAP 0/3/12 ns 3, RAM: 20852/38216; st 80; 0.1: st:1 in_st:0 inm:0 ininv:0; 0.2: st:1 in_st:0 inm:0 ininv:0
376172576769 shelly_main.cpp:569     Up 376172.56, HAP 0/3/12 ns 3, RAM: 20852/38216; st 80; 0.1: st:1 in_st:0 inm:0 ininv:0; 0.2: st:1 in_st:0 inm:0 ininv:0
376180577119 shelly_main.cpp:569     Up 376180.56, HAP 0/3/12 ns 3, RAM: 20852/38216; st 80; 0.1: st:1 in_st:0 inm:0 ininv:0; 0.2: st:1 in_st:0 inm:0 ininv:0
376188577167 shelly_main.cpp:569     Up 376188.56, HAP 0/3/12 ns 3, RAM: 20852/38216; st 80; 0.1: st:1 in_st:0 inm:0 ininv:0; 0.2: st:1 in_st:0 inm:0 ininv:0
376196577271 shelly_main.cpp:569     Up 376196.56, HAP 0/3/12 ns 3, RAM: 20852/38216; st 80; 0.1: st:1 in_st:0 inm:0 ininv:0; 0.2: st:1 in_st:0 inm:0 ininv:0
376204576804 shelly_main.cpp:569     Up 376204.56, HAP 0/3/12 ns 3, RAM: 20852/38216; st 80; 0.1: st:1 in_st:0 inm:0 ininv:0; 0.2: st:1 in_st:0 inm:0 ininv:0
376212576980 shelly_main.cpp:569     Up 376212.56, HAP 0/3/12 ns 3, RAM: 20852/38216; st 80; 0.1: st:1 in_st:0 inm:0 ininv:0; 0.2: st:1 in_st:0 inm:0 ininv:0
376220576843 shelly_main.cpp:569     Up 376220.56, HAP 0/3/12 ns 3, RAM: 20852/38216; st 80; 0.1: st:1 in_st:0 inm:0 ininv:0; 0.2: st:1 in_st:0 inm:0 ininv:0
376228577111 shelly_main.cpp:569     Up 376228.56, HAP 0/3/12 ns 3, RAM: 20852/38216; st 80; 0.1: st:1 in_st:0 inm:0 ininv:0; 0.2: st:1 in_st:0 inm:0 ininv:0
376236577220 shelly_main.cpp:569     Up 376236.56, HAP 0/3/12 ns 3, RAM: 20852/38216; st 80; 0.1: st:1 in_st:0 inm:0 ininv:0; 0.2: st:1 in_st:0 inm:0 ininv:0
376244576994 shelly_main.cpp:569     Up 376244.56, HAP 0/3/12 ns 3, RAM: 20852/38216; st 80; 0.1: st:1 in_st:0 inm:0 ininv:0; 0.2: st:1 in_st:0 inm:0 ininv:0
376252577008 shelly_main.cpp:569     Up 376252.56, HAP 0/3/12 ns 3, RAM: 21112/38216; st 80; 0.1: st:1 in_st:0 inm:0 ininv:0; 0.2: st:1 in_st:0 inm:0 ininv:0
376260577080 shelly_main.cpp:569     Up 376260.56, HAP 0/3/12 ns 3, RAM: 21112/38216; st 80; 0.1: st:1 in_st:0 inm:0 ininv:0; 0.2: st:1 in_st:0 inm:0 ininv:0
376268576764 shelly_main.cpp:569     Up 376268.56, HAP 0/3/12 ns 3, RAM: 21112/38216; st 80; 0.1: st:1 in_st:0 inm:0 ininv:0; 0.2: st:1 in_st:0 inm:0 ininv:0

May the shelly above had some other problem, this one is not getting as hot (still 80°).