mongoose-os-apps/shelly-homekit #517: Shelly 2.5 bipolar switch mode

AG-dingoant, 2021-02-17:
Hi,
I have outdoor lights with bipolar switches (it is a rule in my country). I would to use a shelly 2.5 to command the outdoor lights. One of the switches will be used for the Pole and the second for the Neutral. It would be nice to have a bipolar mode that 

- has only 1 switch in HK
- links the switch 1 to the switch 2 to automate the switch on

Do not hesitate to request more details if my request is not clear enough :-)

timoschilling, 2021-02-17:
You can solve that in the Home App by your own. You can create grouped devices, which looks like one device but are 2 devices which are controlled
as one

timoschilling, 2021-02-18:
update: you need to wire switch 1 and switch 2 of the shelly to you wall switch.

AG-dingoant, 2021-02-18:
Hi Timo,
Thanks for your answer. I just discovered the group function in HK :-). I will test the wiring next week when my new Shelly order will be delivered (I have no Shelly 2.5 available now :-))

freddair, 2021-02-18:
Im'm curious how you want to switch a neutral wire using a shelly 2.5, as it can only switch the line wire it uses for its operation.

I can only think of switching a neutral wire using dry contacts like with a shelly 1.

timoschilling, 2021-02-18:
updated my first post

AG-dingoant, 2021-02-19:
My order is on its way and should arrive somewhere next week. I will test and provide some feedbacks on if and how it is possible.

timoschilling, 2021-03-02:
I think this issue can be closed as the feature request isn't resolvable with a 2.5

closed 2021-03-02 by andyblac