mongoose-os-apps/shelly-homekit #518: Unable to Flash Devices

Ioan-Popovici, 2021-02-18:
I have a bunch of 1PMs and 2.5 switches but I can't seem to be able to flash them via script or link.

I put them on the same vlan and reverted to factory settings but still no go.

with the script I get the following error:

[image]

Ioan-Popovici, 2021-02-18:
I tried older firmware and even put up a web server in my environment and tried to flash it.
Maybe there's something wrong on my network

Ioan-Popovici, 2021-02-18:
I'm starting to think that maybe they changed something. I even started a local phyton server and connected to it offline and it still doesn't work...

timoschilling, 2021-02-18:
I can confirm you error. It's a bug in the flash script.
 Please checkout the commit 2dda62d (`git checkout 2dda62d`) and run the flash script again.
For those  are not using git, copy the flash script from here: https://github.com/mongoose-os-apps/shelly-homekit/blob/2dda62d2c4255d204efde004e3557b9b179ffdb6/tools/flash-shelly.py

Note: The bug was added somewhere between 2dda62d2c4255d204efde004e3557b9b179ffdb6 and 21c70ee330351ea6ddf46d5361e5a0260e2a2f59.

andyblac, 2021-02-18:
@Ioan-Popovici can you run the script again, but add ` --log-file=debug.log v=5`  ie `./flash-shelly.py -a  --log-file=debug.log v=5`
can post the debog.log file

andyblac, 2021-02-18:
think i found issue please try 
[flash-shelly.py.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/6002099/flash-shelly.py.zip)

Ioan-Popovici, 2021-02-18:
Hi, thanks for your prompt response.
I can confirm it works on the 1pm with the latest stock firmware.

Amazing job with the firmware, it works perfect, even across vlans if you enable mDNS. I know now that this is a homekit limitation but maybe it should be in the docs.

Thanks again,
Ioan

Ioan-Popovici, 2021-02-19:
Just tested on shelly 2.5 , works perfectly now. The only thing is that it does not find devices if a password is set on the web interface. Maybe an option would be added in the future to support passwords but that's for an other issue.

Thanks again

PR #519 (2021-02-20): [Flashscript] Version 2.3.9
do not lookup remote version info when using local files.
add uptime to Stock firmware.
fix flashing stock firmware fixes #518 
simplify and give more descriptive messages.
tweak some debug info.
closed 2021-02-19 by Ioan-Popovici