mongoose-os-apps/shelly-homekit #521: Shelly1 display as puls switch in Homekit

heili1094, 2021-02-19:
Is ist possible to se the shelly1 as a puls switch in Homekit which changes the state. For instance i push the button in Homekit and it shows "open" i push it again and it says "closed". Now it's not it show always "on" or "off":

Thanks

andyblac, 2021-02-19:
> Is ist possible to se the shelly1 as a puls switch in Homekit which changes the state. For instance i push the button in Homekit and it shows "open" i push it again and it says "closed". Now it's not it show always "on" or "off":
> 
> Thanks

sorry i dont fully understand, can you give an example of use ?

timoschilling, 2021-02-26:
I think this is wat he want:
[image]

timoschilling, 2021-02-26:
Set input mode to detached and then hap type on the input to contact sensor

MagOO33, 2021-03-24:
Hello, I think that what was requested was to use a shelly1 as a pulse generator for a garden gate which opens or closes by a short (1 second, for example) dry contact between two pins.
For example, Somfy brand does that, no need for a position sensor because it detects the mechanical stop.
I personally did it by configuring a switch with the auto off feature.
In homekit, it shows the real electrical state of the switch : one second open then close, whatever the gate's state is (opening or closing). It works but it's not really clean.
So, the question is : could it be possible with the shelly1 to electrically "pulse" and shows a fake "open" at the first time and a fake "closed" after a second pulse and so on ... ?
Thanks

closed 2021-05-01 by github-actions[bot]