mongoose-os-apps/shelly-homekit #522: URLs are offline

denbec, 2021-02-20:
http://rojer.me/files/shelly/shelly-homekit-Shelly1.zip produces a 404.

rojer, 2021-02-20:
seems to be working just fine for me, and i can see requests in the logs from other users.
can you run `curl -v http://rojer.me/files/shelly/shelly-homekit-Shelly1.zip` from the terminal and paste output here?

timoschilling, 2021-02-20:
I can confirm a 404

rojer, 2021-02-20:
ah, ipv6... ok, will take a look

timoschilling, 2021-02-20:
Yes it's IP6 problem, `curl --ipv4` works

rojer, 2021-02-20:
ok, missing listener spec in the server config, fixed. this wasn't a real problem as devices are ipv4-only anyway.

closed 2021-02-20 by rojer

timoschilling, 2021-02-20:
Devices are ipv4 right, but updates are happen from the UI and the browsers are using ipv6.

rojer, 2021-02-20:
ah, you are right. anyway, thanks @denbec  for reporting and @timoschilling for confirming, this is now fixed.