mongoose-os-apps/shelly-homekit #525: shelly 2.5 firmware version 2.7.2 trouble with chain actuators windows - original firmware work well

gate546, 2021-02-21:
@contributors, please have a look whether this makes sense to you.
My understanding is, that an enhancement of the calibration might be the better solution.

I have a trouble with a chain system, the calibration process does not end normally
I have made some test with fork of @juangamnik but that code donìt work well in my case.
I have read in some posts that the problem of incorrect calibration with chain drives is due to a current draw problem, as if there was no current draw.
According to my observations,  at least in my case, the problem is different, I will try to share what happens to me and what I understand.
I have installed many Shelly 2.5 PMs to operate as roller shutters, all installations work very well with the original firmware in both cases, when Shelly operates the motor directly, in the case of roller shutters, and when Shelly operates by the way of intermediate automatism, as in the case of chain drives.
With the HomeKit firmware, on the other hand, this situation change, when the motors are activated directly everything is fine, when operate on a chain system the calibration does not work.
I have observed that my chain drive has a little delay from when the push button is pressed to when the motor starts moving, this is likely the problem.
The calibration function, when the process start, expects to immediately see an absorption of current and instead this does not happen because, due to the electronics inside the drive, the current is seen after a while, in my case I think more or less a second, and this make that the process fails.
Since the original firmware works well, and observing the operation of the chain mechanism, I suppose that inserting a delay in the code that read the current absorption during the initial calibration process maybe will address this issue.

timoschilling, 2021-02-21:
repost of https://github.com/mongoose-os-apps/shelly-homekit/issues/505

andyblac, 2021-02-21:
seems #505 was edited, i have left that one closed.

andyblac, 2021-03-03:
@everyone, can anyone help here

gate546, 2021-04-03:
I am still waiting for a solution
the last firmware don't address this issue

closed 2021-05-12 by github-actions[bot]