mongoose-os-apps/shelly-homekit #531: Rebooting in 2.7.1 and 2.6.1

jerzykolysz, 2021-02-26:
Have number of Shelly 2.5 I updated recently from 2.0.6 to 2.7.3 and they start to reboot during operation very often. 
I upload 2.6.1 and situation is slightly better but it still rebooting from time to time. 
Now I downgraded it to **2.0.11** and it working quite stable. 
Regards, 
JK

BlancoDeMallloca, 2021-02-27:
I do not experience this issue with version 2.7.3.  Got several 2.5 with rollershutter and switch mode. 

May the debug log would help the developers

andyblac, 2021-02-27:
@jerzykolysz Enable debug logs, open a web browser to collect logs, and leave open to collect the log we will need to see the part where it is rebooting. post log once you have caught it.

jerzykolysz, 2021-02-27:
Sure, I will. 

Pozdrawiam,
Jerzy Kołysz 
Wysłane z iPhone'a

closed 2021-04-06 by github-actions[bot]