mongoose-os-apps/shelly-homekit #533: Http toggle

JCalvi, 2021-02-27:
I can get on and off to work but how do I toggle?

http://shelly1-test.local/rpc/Shelly.SetState?id=1&type=0&state=%7b%22state%22%3atrue%7d turns on
http://shelly1-test.local/rpc/Shelly.SetState?id=1&type=0&state=%7b%22state%22%3afalse%7d turns off

http://shelly1-test.local/rpc/Shelly.SetState?id=1&type=0&state=%7b%22toggle%22%3atrue%7d does not work

andyblac, 2021-02-27:
don't think @rojer has coded toggle for statefull switch, only statless, i guess your using a monentary input on a statefull switch ?

JCalvi, 2021-02-27:
Switch is wired as edge so can be turned on/off from physical as well as web access.

web access is linked to remote control button via htpc so I can turn a light on/off via the wall switch or the remote.

JCalvi, 2021-02-28:
Note: this could also be a repeat of https://github.com/mongoose-os-apps/shelly-homekit/issues/461 as I suspect he had the same desire and similar use case as me.

EmmieFromTheBlock, 2021-05-08:
@rojer Would be awesome to have a toggle function via rest api for the type 0 switch!

andyblac, 2021-09-01:
this is/will be required once lights bulb support are added too.

JCalvi, 2021-09-04:
@roger any plans for adding this feature? 
I am still on stock firmware for now and not complaining, just like to know if its on your radar to be done and I should wait or whether I need to consider other options.

timoschilling, 2021-09-04:
I think it will be come, I’m also interested to implement it for my self, but no estimation for now

JCalvi, 2021-09-04:
Thanks Timo, I’ll wait then.

closed 2023-02-27 by timoschilling