mongoose-os-apps/shelly-homekit #535: Shelly 1 garage door opener - push button does not work

Germanytraveller, 2021-03-01:
Hello,

first of all: i‘m totally new to this whole thing. 

I‘ve successfully flashed and installed a Shelly 1 as a garage door opener and integrated it into my home app. It works perfect in the home app and the garage‘s radio transmitter works fine with it, too.

But the wall-mounted cable- push button (connected to L and SW) does not work. When I push this button, Shelly gets the command and the home app sends a message, that the garage door is opening/closing, but nothing happens.

gregDEweb, 2021-03-01:
Sure that you connected it according to the drawing? L and SW for the Button is not correct. 
[image]

Germanytraveller, 2021-03-02:
yes, i connected it exactely like the drawing. The Shelly does create a command as well, when the button is pushed. But it seems as if this command does not reach the garage-controller while Siri- and radio-controller-initiated shelly-commands do reach the garage-controller.

[image]

AG-dingoant, 2021-03-02:
Hi,
you should connect the button directly on I and O or directly on the door opener. The L and SW are used for a magnet contactor that is used to know the status of the door (open or closed).

gregDEweb, 2021-03-02:
You are not sending a command since you wired it up wrong. By pressing the button you are sending a signal which is changing the status. 
Make sure the limit switch is connected to L and SW and the Push Button to I and 0.
I assume you are from Germany and if it helps i reply in German if needed.

Germanytraveller, 2021-03-06:
Thank you for the replies. I now bought a magnetic contact and wired everything as shown in the picture.

The push button now works, but now i’ve got a new problem. The status of the garage door shown in the home app always turns to “closed” after a few seconds, no matter what position the door is at.
The wired magnetic contact is positioned at the bottom end of the door-rail to ensure that no magnetic component passes the wired-magnet while the door moves.
When the door is closed the distance between the two magnets is 3-4mm. 

What do I do wrong?

gregDEweb, 2021-03-06:
Make sure you did set the limit switch type to the right type. Normally closed or normally open. This is part of the settings. You should also set the movement time. 
Check if this solves the issue.

timoschilling, 2021-03-09:
Can you test the alpha?

Germanytraveller, 2021-03-18:
Sorry for the delay...

i can test it, but you have to tell me how to get the alpha to the Shelly.

timoschilling, 2021-03-18:
@Germanytraveller it's no longer a alpha, please just update to 2.8.2

Trasmontinho, 2021-03-18:
Hi there, 

I'm facing the same issue with the 2.8.2 version. 

Let me expose it here. 

So basically, when the door is closed the sensor is returning to the shelly 0v. When the door is opened, The sensor is returning 35v dc. 

When I toggle the door, The door is opening and few seconds after I see the state door on the shelly to closed (not on the homekit app). 

I don't understand why. May Have you the chance to help me on that ? 

Thanks, 

Regards, 

Sam

Trasmontinho, 2021-03-18:
to add to my previous message below the logs : 

```
167146575 shelly_output.cpp:58    Output 1: off -> on (GDO:open)
167154237 shelly_hap_garage_d:307 GDO 1: Tgt State: closed -> open (1 -> 0) (HAPopen)
167161081 shelly_hap_garage_d:222 GDO 1: Cur State: closed -> opening (1 -> 2)
167453389 shelly_output.cpp:58    Output 1: on -> off (pulse_off)
168675444 mgos_http_server.c:180  0x3fff687c HTTP connection from 192.168.1.98:49848
168719799 mgos_http_server.c:180  0x3fff754c HTTP connection from 192.168.1.98:52310
168726707 mgos_http_server.c:187  0x3fff754c GET /status
168786161 mgos_http_server.c:180  0x3fff75fc HTTP connection from 192.168.1.98:52311
168794771 mgos_http_server.c:187  0x3fff75fc GET /
168803536 mongoose.c:6012         '/index.html.gz' '/index.html' 'text/html'
173649177 shelly_main.cpp:560     Up 173.64, HAP 0/3/12 ns 3, RAM: 21476/38552; st 0; 5.1: cur:opening tgt:open cl:1 op:-1
174196419 shelly_hap_garage_d:307 GDO 1: Tgt State: open -> closed (0 -> 1) (ext)
174203561 shelly_hap_garage_d:222 GDO 1: Cur State: opening -> closed (2 -> 1)
181649264 shelly_main.cpp:560     Up 181.64, HAP 0/3/12 ns 3, RAM: 21656/38552; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
189649396 shelly_main.cpp:560     Up 189.64, HAP 0/3/12 ns 3, RAM: 20668/38552; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
197649161 shelly_main.cpp:560     Up 197.64, HAP 0/3/12 ns 3, RAM: 22264/38552; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
205649068 shelly_main.cpp:560     Up 205.64, HAP 0/3/12 ns 3, RAM: 22264/38552; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1

```

Germanytraveller, 2021-03-18:
Same here, problem is still the same after updating.

Trasmontinho, 2021-04-20:
Hi there, 

Any news on this issue ?

closed 2021-05-28 by github-actions[bot]

Joddik, 2022-08-25:
Issue is still present! Seems like a bug with the input on the shelly. It works fine on shelly's webserver, with correct status. The problem starts when I've added it to homekit. Seems like the switch status doesn't update. The strange thing is that is does update the input status if I give it a open/close command from homekit.