mongoose-os-apps/shelly-homekit #536: I3 with mongoose firmware

Ludelag, 2021-03-01:
How will behave I3 with the mongoose firmware? I have 4 I3 connected to a switched pir sensor and all work thru homebridge as simple button, when pir sensor detects presence it trigger the button

andyblac, 2021-03-03:
the i3 is a supported device, if you have flash it and try for yourself.

Ludelag, 2021-03-03:
what to spect from the flashing?

timoschilling, 2021-03-03:
I that don’t have that setup you are asking for, but I expect that it works. Try it. If it’s not working you can easily flash back to stock firmware, it’s just one click in the UI.

Ludelag, 2021-03-03:
Thank you Timo, yes I Know how to get back to the stock firmware. I am asking if some got the experience already can advice me before get into it. Maybe this weekend I will give a try. I will keep posted here in this issue the results and what I will find.

jobl1978, 2021-03-14:
I use the i3 with the HomeKit firmware as a scene switch (long press) to activate my "good night" scene. Works perfekt!

closed 2021-04-21 by github-actions[bot]