mongoose-os-apps/shelly-homekit #537: Max Connections more then 12?

StevenBeck93, 2021-03-02:
Hello,

we want to use the Shelly 1 for a Garage Door with more then 12 People.
Is there a way, to change the "Max Connections" from 12 to another number?

Thank you very much!

andyblac, 2021-03-02:
no, but unless you have 12 poeple actually using the door all a the same time it should drop connnections as required.

timoschilling, 2021-03-02:
And even are 12 people are connected the 13 will share the connection of one of the other.

andyblac, 2021-03-02:
> And even are 12 people are connected the 13 will share the connection of one of the other.

should be talking about 12+ devices not people, you are correct.

StevenBeck93, 2021-03-02:
Thanks for the quick response. 

The usecase is, that we have a hall with automatic door. Now we create a "Home" for this Location and will share this "Home" to more then 12 people, that everyone can open the Hall Door. 

So we dont need a key for each person.

At the moment we have 4 persons with connection to this home. But we dont know, if we extend the limit?

StevenBeck93, 2021-03-02:
Hello,

perfect. Then we will try it :) 

Yes we have a HomePod as Base in this "Home".

Thanks to all of you!

closed 2021-03-02 by StevenBeck93

partytrik, 2021-03-16:
When we have more people here (at the moment 4 visitors) everyone's homekit app always starts reconnecting, every time you open the app. That's very annoying. This is now the problem with roughly 20 iPhones/iPads/Macs/HomePods/Apple TVs in the house. And worse, when the homepods don't have a connection and you can't use siri anymore. There really is no way to add more connections? I understand how it's supposed to work, but the devices are sometimes gone for 15min. And with less people in the house the problem doesn't seem to exist. Or it's all 2.8.2's fault.