mongoose-os-apps/shelly-homekit #539: shelly1-homekit

ma-jkl, 2021-03-04:
Hi all, I am total amateur. But I do have a problem with my Shelly. I have installed it into the HomeKit using the latest software. When I am in the wifi shelly - there is no problem. But when I am on data (LTE) the connection is lost. Does anyone of you have the same issue? 
Thank you 
Michal

andyblac, 2021-03-04:
do you have have a homekit hub (HomePod, AppleTV, iPad in hub mode) ?

ma-jkl, 2021-03-04:
Hi, nope. I add the device into my iPhone.

andyblac, 2021-03-04:
ok, then thats why, in order to use HomeKit outside of WiFi you need a HomeKit hub.

ma-jkl, 2021-03-04:
Ok, thank you. I will try to find more info about this how to do that. Thank you Andy.

ma-jkl, 2021-03-04:
I do have an Apple TV

ma-jkl, 2021-03-04:
[image]

andyblac, 2021-03-04:
no problem. you use that, 

Setting up your HomeKit Hub

Using an Apple TV (4th-gen or 4K)

1. On your iOS device enable two-factor authentication for your Apple ID. Also, go to the iCloud settings menu and check keychain access is enabled.

2. Sign into the same iCloud account on the Apple TV ('Settings' > 'Accounts' to check).

3. iCloud should do the rest but you can go to 'Settings' > 'Accounts' > i'Cloud' to check if HomeKit is connected. You can also check the status of your Home hubs by tapping in the right-hand corner of the Apple Home app screen.

andyblac, 2021-03-04:
but please bare in mind AppleTV is not good at HomeKit automations, so if you decide to get more advanced and start doing Home Automations, I recomend you getting a HomePod mini.

ma-jkl, 2021-03-04:
Andy, thank you. I have checked everything. The shelly works from all devices. Apple TV, iPhone. When I had software version 2.0 it worked from the data. I am using shelly to open gate to my house. But now when I am out of the wifi on phone (LTE) the connection disappears. I see Apple TV running but shelly is writing do response. It means that there is some issue that shelly does not communicate out of my wifi now after latest update.

ma-jkl, 2021-03-04:
debut log enable is not the solution right?

andyblac, 2021-03-04:
how may connection are used ? open webui of device go to homekit section.

ma-jkl, 2021-03-04:
[image]

andyblac, 2021-03-04:
also goto http://deviceip/debug/info should see something like

```
pp: switch1 2.8.0-alpha3 20210223-015045/2.7.3-103-g20412b5-master
Uptime: 576220.54
RAM: 18868 free, 11144 min free
HAP server port: 9000
HAP config number: 32
HAP connection stats: 0/7/12
HAP connections:
  192.168.3.11:64121 nc 0x3fff7944f 0x200000 io 3113 ts 0x3fff7a64 rd 3113
  192.168.3.29:49181 nc 0x3fff7224f 0x200000 io 8177 ts 0x3fff66ec rd 8177
  192.168.3.61:63541 nc 0x3fff6854f 0x200000 io 4179 ts 0x3fff5af4 rd 4179
  192.168.3.16:51530 nc 0x3fff679cf 0x200000 io 576190 ts 0x3fff7a8c rd 576169
  192.168.3.14:50239 nc 0x3fff79dcf 0x200000 io 4090 ts 0x3fff791c rd 4090
  192.168.3.40:64530 nc 0x3fff6a94f 0x200000 io 576218 ts 0x3fff6244 rd 576197
  192.168.3.31:50085 nc 0x3fff72e4f 0x200000 io 131464 ts 0x3fff6f0c rd 576200
 Total: 7
HAP sessions:
  0: s 0x3ffef4c8 ts 0x3fff66ec o 1 st 1 ts 568042670
  1: s 0x3ffef808 ts 0x3fff5af4 o 1 st 1 ts 572040727
  2: s 0x3ffefb48 ts 0x3fff7a64 o 1 st 1 ts 573107519
  4: s 0x3fff01c8 ts 0x3fff6f0c o 1 st 1 ts 444760630
  5: s 0x3fff0508 ts 0x3fff6244 o 1 st 1 ts 23021
  8: s 0x3fff0ec8 ts 0x3fff791c o 1 st 1 ts 572130492
  9: s 0x3fff1208 ts 0x3fff7a8c o 1 st 1 ts 50940
 Total: 7
```

is your appletv ip listed. ?

ma-jkl, 2021-03-04:
Thank you, I will need to go to my friend - he is IT because it is too much for me. The link do not work.

andyblac, 2021-03-04:
what is the ip of your shelly device ?

ma-jkl, 2021-03-04:
192.168.1.27

ma-jkl, 2021-03-04:
in my network

andyblac, 2021-03-04:
try http://192.168.1.27/debug/info

ma-jkl, 2021-03-04:
App: switch1 2.7.3 20210124-130526/2.7.3-g883914a-master
Uptime: 3968.08
RAM: 22636 free, 18968 min free
HAP server port: 9000
HAP config number: 1
HAP connection stats: 0/3/12
HAP connections:
  192.168.1.31:54564 nc 0x3fff61e4f 0x200000 io 35 ts 0x3fff6364 rd 34
  192.168.1.40:56342 nc 0x3fff5ce4f 0x200000 io 123 ts 0x3fff5804 rd 123
  192.168.1.21:49321 nc 0x3fff577cf 0x200000 io 24 ts 0x3fff6034 rd 24
 Total: 3
HAP sessions:
  0: s 0x3ffef4c8 ts 0x3fff5804 o 1 st 1 ts 3844509
  1: s 0x3ffef808 ts 0x3fff6034 o 1 st 1 ts 3943944
  2: s 0x3ffefb48 ts 0x3fff6364 o 1 st 1 ts 3933254
 Total: 3

andyblac, 2021-03-04:
what is the IP of your appletv ?

ma-jkl, 2021-03-04:
192.168.1.21

ma-jkl, 2021-03-04:
Apple TV is connected via cable

andyblac, 2021-03-04:
hmm it does have a connection, so it should be working. you now have 3, before it was 2, can you try data again ?

ma-jkl, 2021-03-04:
App: switch1 2.7.3 20210124-130526/2.7.3-g883914a-master
Uptime: 4177.84
RAM: 21756 free, 18968 min free
HAP server port: 9000
HAP config number: 1
HAP connection stats: 0/3/12
HAP connections:
  192.168.1.31:54564 nc 0x3fff61e4f 0x200000 io 245 ts 0x3fff6364 rd 244
  192.168.1.40:56342 nc 0x3fff5ce4f 0x200000 io 333 ts 0x3fff5804 rd 333
  192.168.1.21:49321 nc 0x3fff577cf 0x200000 io 234 ts 0x3fff6034 rd 233
 Total: 3
HAP sessions:
  0: s 0x3ffef4c8 ts 0x3fff5804 o 1 st 1 ts 3844509
  1: s 0x3ffef808 ts 0x3fff6034 o 1 st 1 ts 3943944
  2: s 0x3ffefb48 ts 0x3fff6364 o 1 st 1 ts 3933254
 Total: 3

ma-jkl, 2021-03-04:
from my point of view - when I am not in the same network - shelly is unavailable

andyblac, 2021-03-04:
in the HomeApp, tap on he HOUSE icon (top left), goto Home Settings. scroll down to "Hubs & Bridges" tap on it post a screenshot please.

ma-jkl, 2021-03-04:
it seems that shelly is now not comunicating out of my home network.

ma-jkl, 2021-03-04:
[image]

andyblac, 2021-03-04:
I assume appleTV syas "Connected" ?

ma-jkl, 2021-03-04:
připojeno = connected

ma-jkl, 2021-03-04:
yes please

andyblac, 2021-03-04:
it should be woking then, should be no reason for it not to be working outside of wifi. do you have any Router firewals ?

ma-jkl, 2021-03-04:
I have Amplify router. I did not make any changes - only updated the shelly software.

ma-jkl, 2021-03-04:
Thank you very much for your help. Not happy about this situation because we open the gate for food delivers, fedex etc when we are out of home and we can not use it :)

andyblac, 2021-03-04:
do you have other homekit accessories ? are they working ?, what was the last version that worked for you?

ma-jkl, 2021-03-04:
i have only Apple TV and I see it when I am out of my wifi.

andyblac, 2021-03-04:
have you tried rebooting your Appletv, i seem to remeber someone else have the same issue, it was down to his AppleTV

ma-jkl, 2021-03-04:
ok. I will try.

andyblac, 2021-03-04:
here this guy had same issue but on a shelly 2.5, it was a AppleTV issue https://github.com/mongoose-os-apps/shelly-homekit/issues/457
maybe ask him how he fixed it ?

ma-jkl, 2021-03-04:
I did reboot - no change. Thank you, I will look at it. I really appreciate you help. Big thanks to you

andyblac, 2021-03-04:
i'll close this for now. as it seems a homekit issue.

closed 2021-03-04 by andyblac

ma-jkl, 2021-03-05:
problem solved:
I contacted apple support and did not know why it happened. At the and I have tried to completely remove HomeKit and add a new one and setup it from the begin and Shelly works now out of home wifi from LTE or other wifi!

SanchoCZ, 2021-07-27:
Hi. I have same problem. Can I take question?

ma-jkl, 2021-07-27:
Yes. For sure. You speak CZ as well?

SanchoCZ, 2021-07-27:
Ahoj, mam stejný Problem ale odstranění Domácnosti nepomohlo
Mam dalsi certifikovane prislusenstvi a to vidim jak pres wifi tak LTE

SanchoCZ, 2021-07-28:
> Yes. For sure. You speak CZ as well?
> […](#)

Odstranoval jsi to jen v iPhonu?