mongoose-os-apps/shelly-homekit #540: Shelly 1L physical switch is not working

benlaniado, 2021-03-05:
Hey All,
Before adding the homekit firmware, I could turn the light on and off with the app and with the switch.

Now I can turn the light on and off from homekit or from the web interface. 
But the physical switch is not doing anything.
Input mode is set to edge.
I can't add the device to the shelly app now.

Should I revert back to the stock version?

wawoodwa, 2021-03-05:
Have you tried changing the switch from edge to toggle?

Standard single throw switch?  Not a three way correct?

The mongoose HomeKit firmware does not work with the Shelly App.  If you want to use the Shelly App as well as HomeKit, you will need to revert to stock firmware and implement homebridge with the Shelly plugin (I don't run it this way, so I can't give anymore pointers on that, but there is a lot of info on GitHub).

benlaniado, 2021-03-05:
> Have you tried changing the switch from edge to toggle?
>
> Standard single throw switch? Not a three way correct?
>
> The mongoose HomeKit firmware does not work with the Shelly App. If you want to use [...]

I tried all the options :(
I will try homebridge. Thanks

wawoodwa, 2021-03-05:
Hmmm...

I use the Shelly 1L for my lights at home, with switch 1 set to toggle and switch 2 disabled.  Sorry it isn't working out for you.  But yeah, if you are using the Shelly app, homebridge is the only way to go with the stock firmware.

benlaniado, 2021-03-05:
I will try the homebridge option and if it won’t work I will retry the
HomeKit firmware. Thanks

timoschilling, 2021-03-05:
With the HomeKit firmware you can only use switch 1 to control the light. Switch 2 works only in detached mode. Maybe your wall switch is connected to Switch 2? If yes, please wire it to Switch 1. Or post a screenshot from your config UI.

AG-dingoant, 2021-03-06:
Why is it not possible to have the two switches to control the light with the homekit firmware? I have one Shelly 1L with two switches and I would like to have the two switches working.

timoschilling, 2021-03-06:
It’s because of how the things are working internally, at the moment input and output are one unit so isn’t easy to build a second input. 

But you can use the second switch as a stateless switch and add an action in Home to control the output. There is a wiki entry about how to control a light/outlet via stateless switch.

andyblac, 2021-03-06:
closing this as duplicate issue https://github.com/mongoose-os-apps/shelly-homekit/issues/346

closed 2021-03-06 by andyblac

timoschilling, 2021-03-11:
@benlaniado maybe I found the bug, how was you 1L wired, did you use S1 or S2? Is there a "V2" sticker on you 1L?

benlaniado, 2021-03-11:
> @benlaniado maybe I found the bug, how was you 1L wired, did you use S1 or S2? Is there a "V2" sticker on you 1L?

@timoschilling Thank you for taking the time to investigate the issue. 
I have 2 Shelly 1L that was installed by electrician. I have the same issue with both of them. Is there a programmatic way I can get this data? Or some other way can I help you debug?

timoschilling, 2021-03-12:
@benlaniado Sorry wrong alarm, I reinvestigate my issue and it's solved, was something wrong in the wiring, sorry can't help you.