mongoose-os-apps/shelly-homekit #543: Home Assistant integration?

uspino2, 2021-03-09:
I use my Shelly 2.5 as a garage opener through Home Assistant and from there into HomeKit, but this firmware so much better (double reed, obstruction detection, etc.). If I flash this firmware, is there a way I can still integrate the Shelly simultaneously into Home Assistant? (My house still requires Home Assistant integration because everything -including the garage door- is controlled or supervised from a Home Assistant touch panel). Thanks!

timoschilling, 2021-03-09:
I expect a backwoods compatible API wouldn’t  work nicely, because it can’t reflect all features of this firmware. I think the better option is to develop a shelly-homekit home bridge plugin. But I expect that it is hard to find someone how will do that, because one goal of this firmware is to make bridges obsolete.

Ioan-Popovici, 2021-03-11:
No, but you can integrate it with HA and then expose it to homekit. This is what I do with my bathroom fan which is controlled by humidity and temp levels with Node Red in HA. Or, you could just use stock firmware with MTTQ or shelly plugin.

uspino2, 2021-03-12:
> No, but you can integrate it with HA and then expose it to homekit. This is what I do with my bathroom fan which is controlled by humidity and temp levels with Node Red in HA. Or, you could just use stock firmware with MTTQ or shelly plugin.

Regrettably, HomeKit integration for garage cover in Home Assistant is very very limited: only gives HomeKit `open` and `closed` states. This firmware also provides `opening`, `closing` and `obstruction` states.

Ioan-Popovici, 2021-03-12:
Can't you get that with MTTQ and the original firmware? That's how I get the data from my Shelly HT sensor that controls the fan. It shows up in HomeKit just fine, even the battery reading.

uspino2, 2021-03-12:
Yes, I've done it with MQTT and templates. The issue is how Home Assistant has implement HomeKit integration for garage doors. It's currently very limited. It doesn't even let HomeKit alert you of obstructions...

andyblac, 2021-03-12:
ok clossing as not possible.

closed 2021-03-12 by andyblac