mongoose-os-apps/shelly-homekit #544: Setting a Shelly 2.5 as a Valve

homekitguy, 2021-03-09:
Hello i want to show my Shelly as a valve in HomeKit but i can’t choose it. What should i do???
[image]

timoschilling, 2021-03-09:
Are you using 2.7.X or 2.8.0-alphaX? Valve is only supported in 2.8, but not officially released yet, there is a alpha available, but be careful installing it, there are some connection issues.

homekitguy, 2021-03-09:
Yeah its 2.7 will the 2.8 support doorbell? Because i saw it here too: https://github.com/mongoose-os-apps/shelly-homekit/wiki/Input-Switch-Settings

andyblac, 2021-03-09:
> Yeah its 2.7 will the 2.8 support doorbell? Because i saw it here too: https://github.com/mongoose-os-apps/shelly-homekit/wiki/Input-Switch-Settings

yes, i updated wiki a liitle too soon, we thought 2.8 would not be so long in testing, sorry for confusion.

homekitguy, 2021-03-09:
Okay thank you ☻

closed 2021-03-09 by homekitguy