mongoose-os-apps/shelly-homekit #552: Lusacelli Plaque es

homekitguy, 2021-03-10:


timoschilling, 2021-03-10:
What does you concrete want to control with the Shelly?

homekitguy, 2021-03-11:
> What does you concrete want to control with the Shelly?

I want to control an alarm bell and have an icon like this 
[image] in Home. I would only another icon.

homekitguy, 2021-03-11:
[image]

homekitguy, 2021-03-11:
[image]

homekitguy, 2021-03-12:
[image]

AG-dingoant, 2021-03-19:
Hi all,
I am interested as well to have the security option in the shelly. I have a bentel alarm system and I can get signals from it. I will connect the signal to the switch of a shelly module.
It will allow me to know if the system is activated or not. If it is activated, than I would like to have all the lights switched off. 
If the alarm rings, than all the lights have to be switched on.
I can set the shelly as an switch but in homekit it will not be displayed as an alarm system.

Chpedro93, 2021-05-04:
Good morning,
Any news regarding this? It will be implemented in the future updates? 
Thanks.

5xPa, 2022-05-03:
Thanks for this 
I am looking for the security system is “armed” meaning the system is monitoring the house. Which as described above would then switch all the lights and heating off. I have a uni set as a contact to accomplish this. When I return home and disarm the system, heating on.

Another uni is used when the system is triggered and bell ringing sends a push notification, this is also set as a contact.

Has the updates been added to a beta I would be delighted to test the update. 
I have installed the beta 2.11.0-beta3 for 35 days and expierenced  no problems.
Thank you for the excellent upgrades

iMakit, 2022-06-26:
Hi all, I'm trying to use Shelly 1 connected to my "old" alarm hub to set alarm ON and OFF, at the moment I set in the hub one key to be used as dry contact to arm and disarm, I'm setting hap service type as switch and toggle, any suggestion to be used as alarm type?
something like abode system also if at the moment I can use only one
[image]

I just saw the new firmware is compatible with Shelly UNI, tomorrow I'll get one for testing if it can manage 2 states as On and away mode, any suggestion on how it may be configured? 

And thanks to all of you for amazing job and upgrades

iMakit, 2022-06-27:
Hi tried with Shelly UNI but it's like to have 2 separated switch, is it possible to have something like the alarm to cutaway and nigh mode for example? Something like Abode system above 
I saw the trigger documentation [here](https://developer.apple.com/documentation/homekit/hmcharacteristictypesecuritysystemalarmtype) but not sure how it can be used to arm security alarm. Did someone tried to connect alarm hub that may help? Thank :)

iMakit, 2022-10-14:
I made a small improvement to the alarm system, I configured a shelly UNI as 2 switches and momentary impulse and one entry of the central station as key point, that means that with a shelly one impulse set the alarm ON and one again will set it OFF.
I also added an automation in HomeKit put the alarm ON when the last person goes away and set OFF when the first person come back. Would be great if it can be managed in security instead of switch with possibility to set TOTAl PARTIAL and OFF alarm.