mongoose-os-apps/shelly-homekit #553: Any plans to support Shelly Motion and Shelly H&T?

mcackay, 2021-03-11:
Brand new to the Shelly ecosystem -- and starting to experiment with it.

Wondered if it was likely that this project will try to provide Homekit firmware for the following devices?
- Shelly Motion
- Shelly H&T (Humidity & Temperature)

Or if there's another project that anyone can direct me to?

timoschilling, 2021-03-11:
For the H&T is already a request there: https://github.com/mongoose-os-apps/shelly-homekit/issues/186

There are two problems with the Motion:
1. It's a battery device, this firmware uses the homekit ip protocol, which isn't made for battery devices, it requires a permanent active IP connection. This will result in a high battery drain
2. It's processor is currently not supported by mongoose os, which is the base of this firmware.

mcackay, 2021-03-11:
Thanks Timo, that makes sense.
Appreciate the prompt reply.

closed 2021-03-11 by mcackay