mongoose-os-apps/shelly-homekit #556: 2.8.1 with Shelly 2.5 in GDO mode with 24V not working

Pixel-Chris, 2021-03-11:
After fw Upgrade to 2.8.0 multiple reboots without any action of the motor.
Back to 2.7.3.

Now on 2.8.1
More reboots
GUI says only „Loading...“
And restarts

Pixel-Chris, 2021-03-11:
„Normally” the device restarts only when the motor starts 
(With 24V)

Pixel-Chris, 2021-03-11:
How to get back to 2.7.3 with url?

andyblac, 2021-03-11:
> How to get back to 2.7.3 with url?
> Maybe that will work

use my script

`./flash-shelly.py -V=2.7.3 hostname`

rojer, 2021-03-11:
@Pixel-Chris hm. logs please?

Pixel-Chris, 2021-03-11:
So... back to 2.7.3
Now it’s stable again
(Until I will push the button and the motor starts)

i think it is still a power problem
But with 2.8.1 not usable

i will need a power supply with 25 or 26V

Pixel-Chris, 2021-03-11:
Sorry no logs
One short log, but with no problem, the only think was that the current state shows open, target also open
But the door was closed

Will try it again, when I have more time (maybe this weekend)

rojer, 2021-03-11:
thing is, nothing really changed that would've impacted power consumption. wireless SDK is the same, TCP stack version changed but that shoudn't have a direct effect.

Pixel-Chris, 2021-03-11:
> @Pixel-Chris hm. logs please?
> 
> re: going to a specific version - binaries are attached to github releases and an archive is also available on my server - http://rojer.me/files/shelly/

For the next time: is there a way to use OTA? I have tested it with:

Http://*IP*/ota?url=http://rojer.me/files/shelly/2.7.3/shelly-homekit-Shelly25.zip

But it says “not found”

rojer, 2021-03-11:
HK doesn't support `/ota?url=` method (and can't due to mos licensing), but you can download the file you want and upload it using the form. but we could in theory support `/ota` as well.

Pixel-Chris, 2021-03-11:
The form was not really accessible (sometimes only a short period) the fifth attempt was working and the upload and firmware downgrade was working

maybe ota would be a nice feature!?

closed 2021-05-01 by github-actions[bot]