mongoose-os-apps/shelly-homekit #557: 2.8.1 GUI inconsistent capitalisation

jsiegenthaler, 2021-03-11:
[Eagle Eye Mode On]

The following GUI text is sometimes in ProperCase and sometimes a mix of ProperCase and lowercase

HAP Service Type:   ProperCase x 3
Input Mode:    ProperCase x 2
Inverted Input:   ProperCase x 2
Initial state:    ProperCase and lowercase
Inverted output:  ProperCase and lowercase
Auto off:   ProperCase and lowercase
Auto off delay:  ProperCase and lowercase x 2

Choose either one or the other, don't mix styles in the same GUI
Personally, I think first letter of first word capitalised, and then subsequent words lower case looks better on this GUI.

andyblac, 2021-03-13:
@rojer any preferance, i dont mind doing this one.

partytrik, 2021-03-15:
@timoschilling don’t want to open a separate request for such a minor thing, but the new svg logo seems to be distorted

timoschilling, 2021-03-15:
@partytrik That's a known issue, as in the introduction PR already mentioned it's compromise between size and having a logo. It's hand coded by me, so it's not the best, Rojer was fine with it, but improvement PRs are welcome.

closed 2021-03-15 by rojer

rojer, 2021-03-15:
fixed, thanks!