mongoose-os-apps/shelly-homekit #558: Shelly2.5 pitch connector

louis49, 2021-03-12:
Hello, to connect the Shelly 2.5, which Pitch connector do you recommend? Are we at 1mm pitch? I did not find a 6 pin 1mm male connector. Do you have a link to an existing product please?

louis49, 2021-03-15:
Finally found : [B07TS2KPW7](https://www.amazon.fr/shelly-2-5-Adpter/dp/B07TS2KPW7 )
I ordered it, I will update the Wiki ([Serial-connection](https://github.com/mongoose-os-apps/shelly-homekit/wiki/Serial-connection)) when I have verified that it works well

andyblac, 2021-03-15:
i tried that did not work for me, others say it does.

timoschilling, 2021-03-15:
Has worked for me, but depending on your device you need to cut off or move one pin.

louis49, 2021-03-19:
[B07TS2KPW7](https://www.amazon.fr/shelly-2-5-Adpter/dp/B07TS2KPW7) Works for me

louis49, 2021-03-19:
Wiki updated

closed 2021-03-19 by louis49