mongoose-os-apps/shelly-homekit #561: Shelly 1 No HomeKit Connection Update 2.81

SteffenHuettner, 2021-03-12:
All Shelly 1 lost HomeKit Connection after flashing to 1.82. All Shelly 2.5 still work fine. The Shellys are still running and reachable by ip.

SteffenHuettner, 2021-03-12:
Debug log.

10893291539 shelly_rpc_service.:226 SetConfig ok, 0
10893319140 mgos_sys_config.c:232   Loading conf2.json
10893342890 mgos_sys_config.c:232   Loading conf3.json
10893469729 mgos_sys_config.c:174   Saved to conf9.json
10893495816 shelly_main.cpp:554     Up 10893.48, HAP 0/6/12 ns 6, RAM: 19504/38568; st 0; 0.1: st:0 in_st:1 inm:2 ininv:0
10893553770 HAPPlatformTCPStrea:121 0x3fff5844 192.168.2.189:64247 Incoming HAP connection, ns 0/6/12
10893684852 HAPPlatformTCPStrea:121 0x3fff648c 192.168.2.13:50400 Incoming HAP connection, ns 1/6/12
10893727793 HAPPlatformTCPStrea:121 0x3fff6644 192.168.2.49:57572 Incoming HAP connection, ns 2/6/12
10894372640 HAPPlatformTCPStrea:276 0x3fff5844 192.168.2.189:64247 Accepted HAP connection, ns 2/7/12 ts 0x3fff5c8c
10895678239 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.30:60371
10895703538 HAPPlatformTCPStrea:105 0x3fff5844 192.168.2.189:64247 HAP connection closed, ns 2/7/12 i 2
10895715373 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff5c8c nc 0x0 2/6/12
10896545774 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.30:60371
10897060937 HAPPlatformTCPStrea:121 0x3fff6f7c 192.168.2.78:58749 Incoming HAP connection, ns 2/6/12
10897477102 HAPPlatformTCPStrea:276 0x3fff648c 192.168.2.13:50400 Accepted HAP connection, ns 2/7/12 ts 0x3fff5c8c
10898586783 HAPPlatformTCPStrea:105 0x3fff648c 192.168.2.13:50400 HAP connection closed, ns 2/7/12 i 2
10898598632 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff5c8c nc 0x0 2/6/12
10898629434 HAPPlatformTCPStrea:121 0x3fff6a44 192.168.2.78:58750 Incoming HAP connection, ns 2/6/12
10899974612 esp_main.c:137          SDK: LmacRxBlk:1
10900586732 HAPPlatformTCPStrea:276 0x3fff6644 192.168.2.49:57572 Accepted HAP connection, ns 2/7/12 ts 0x3fff5c8c
10901700006 HAPPlatformTCPStrea:105 0x3fff6644 192.168.2.49:57572 HAP connection closed, ns 2/7/12 i 2
10901711831 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff5c8c nc 0x0 2/6/12
10902588054 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.30:60371
10902892218 HAPPlatformTCPStrea:121 0x3fff648c 192.168.2.13:50411 Incoming HAP connection, ns 2/6/12
10903688959 HAPPlatformTCPStrea:276 0x3fff6f7c 192.168.2.78:58749 Accepted HAP connection, ns 2/7/12 ts 0x3fff5c8c
10905788987 shelly_main.cpp:554     Up 10905.78, HAP 2/7/12 ns 7, RAM: 18008/38568; st 0; 0.1: st:0 in_st:1 inm:2 ininv:0
10905902833 HAPPlatformTCPStrea:121 0x3fff668c 192.168.2.227:49975 Incoming HAP connection, ns 2/7/12
10906730708 HAPPlatformTCPStrea:276 0x3fff6a44 192.168.2.78:58750 Accepted HAP connection, ns 2/8/12 ts 0x3fff7544
10907838780 mgos_sntp.c:98          SNTP query to time.google.com
10907854095 HAPPlatformTCPStrea:105 0x3fff6a44 192.168.2.78:58750 HAP connection closed, ns 2/8/12 i 2
10907866857 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff7544 nc 0x0 2/7/12
10907941308 HAPPlatformTCPStrea:121 0x3fff6a44 192.168.2.49:57590 Incoming HAP connection, ns 2/7/12
10908932138 esp_main.c:137          SDK: LmacRxBlk:1
10909885440 HAPPlatformTCPStrea:276 0x3fff648c 192.168.2.13:50411 Accepted HAP connection, ns 2/8/12 ts 0x3fff76e4
10911026918 HAPPlatformTCPStrea:105 0x3fff648c 192.168.2.13:50411 HAP connection closed, ns 2/8/12 i 2
10911038795 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff76e4 nc 0x0 2/7/12
10913026629 HAPPlatformTCPStrea:276 0x3fff668c 192.168.2.227:49975 Accepted HAP connection, ns 1/8/12 ts 0x3fff5c5c
10915553189 HAPPlatformTCPStrea:105 0x3fff668c 192.168.2.227:49975 HAP connection closed, ns 1/8/12 i 2
10915565988 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff5c5c nc 0x0 1/7/12
10916122983 HAPPlatformTCPStrea:276 0x3fff6a44 192.168.2.49:57590 Accepted HAP connection, ns 0/8/12 ts 0x3fff5c5c
10917262713 mongoose.c:12067        Failed to resolve 'time.google.com', server 8.8.8.8
10917281100 HAPPlatformTCPStrea:121 0x3fff5844 192.168.2.49:57602 Incoming HAP connection, ns 0/8/12
10917310536 HAPPlatformTCPStrea:105 0x3fff6a44 192.168.2.49:57590 HAP connection closed, ns 1/8/12 i 2
10917323335 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff5c5c nc 0x0 1/7/12
10917341516 HAPPlatformTCPStrea:121 0x3fff6a44 192.168.2.78:58762 Incoming HAP connection, ns 1/7/12
10918266764 shelly_main.cpp:554     Up 10918.26, HAP 2/7/12 ns 7, RAM: 19688/38568; st 0; 0.1: st:0 in_st:1 inm:2 ininv:0
10919195913 HAPPlatformTCPStrea:276 0x3fff5844 192.168.2.49:57602 Accepted HAP connection, ns 1/8/12 ts 0x3fff6b4c
10920320116 HAPPlatformTCPStrea:105 0x3fff5844 192.168.2.49:57602 HAP connection closed, ns 1/8/12 i 2
10920332687 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff6b4c nc 0x0 1/7/12
10922064774 HAPPlatformTCPStrea:121 0x3fff63d4 192.168.2.227:49990 Incoming HAP connection, ns 1/7/12
10922083275 HAPPlatformTCPStrea:121 0x3fff65c4 192.168.2.227:49989 Incoming HAP connection, ns 2/7/12
10922266506 HAPPlatformTCPStrea:121 0x3fff7804 192.168.2.9:60341 Incoming HAP connection, ns 3/7/12
10922280660 HAPPlatformTCPStrea:276 0x3fff6a44 192.168.2.78:58762 Accepted HAP connection, ns 3/8/12 ts 0x3fff6b4c
10923415709 HAPPlatformTCPStrea:105 0x3fff6a44 192.168.2.78:58762 HAP connection closed, ns 3/8/12 i 2
10923427611 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff6b4c nc 0x0 3/7/12
10923442011 HAPPlatformTCPStrea:121 0x3fff78ec 192.168.2.49:57605 Incoming HAP connection, ns 3/7/12
10924404307 esp_main.c:137          SDK: LmacRxBlk:1
[... cut ...]
10959230162 shelly_main.cpp:554     Up 10959.22, HAP 3/8/12 ns 8, RAM: 15352/38568; st 0; 0.1: st:0 in_st:1 inm:2 ininv:0
10960115072 esp_main.c:137          SDK: LmacRxBlk:1
10960130441 HAPPlatformTCPStrea:276 0x3fff6a44 192.168.2.13:50433 Accepted HAP connection, ns 2/9/12 ts 0x3fff650c
10961310456 HAPPlatformTCPStrea:105 0x3fff6a44 192.168.2.13:50433 HAP connection closed, ns 2/9/12 i 2
10961322883 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff650c nc 0x0 2/8/12
10961494051 HAPPlatformTCPStrea:105 0x3fff6614 192.168.2.49:57628 HAP connection closed, ns 2/8/12 i 2

chstoll, 2021-03-12:
Same here, except in my case some of my Shelly Plug S are unresponsive in Homekit. I've tried to pair them again (reset HAP and delete from Homekit) without any luck.

rojer, 2021-03-12:
hm, `LmacRxBlk:1` does not look good...

iperniaf, 2021-03-12:
Same problem here

tojorocha, 2021-03-12:
The same here. How can I go back to 2.7.3?

andyblac, 2021-03-12:
> The same here. How can I go back to 2.7.3?

download file http://rojer.me/files/shelly/2.7.3/shelly-homekit-Shelly1.zip upload it via webui.

Dag0d, 2021-03-13:
same here, downgrading to 2.7.3 isn't possible, it will fail and disconnects the shelly1 from the network

rojer, 2021-03-13:
just tried downgrading - it works. so it's something to do with this networking issue, not a general problem. i will look into this LmacRxBlk problem ASAP.

partytrik, 2021-03-13:
Hi, my problems weren't quite as bad, but unstable homekit connection. 
To be honest, my impression was that it already got less stable with 2.7.3.
If it helps: I have all shellies connected with a static IP address and I am using a Fritzbox Router. Maybe that helps identifying the problem.
Reverting back to 2.7.3 was no problem for me.

rojer, 2021-03-13:
ok, i believe i understand what the issue is. it's happening on networks with a lot of traffic, the device runs out of network buffers and starts dropping packets on the floor. there are only 5 (five) buffers available, for both TCP and UDP, and with a lot of activity they can get held up for long enough for system to run out.

@SteffenHuettner @partytrik @Dag0d @frankheit @tojorocha can you please try this firmware http://rojer.me/files/shelly/2.8.1_pbuffix/ and let me know if it behaves better

Dag0d, 2021-03-14:
tested with only 2 devices yet, they respond again in HomeKit but way slower than before, at least for me. I've also seen "No Response"

rojer, 2021-03-14:
@Dag0d please post debug log

Dag0d, 2021-03-14:
seems to me like it's dropping some connections

1006577881 shelly_rpc_service.:226 SetConfig ok, 0
1006609716 mgos_sys_config.c:232   Loading conf2.json
1006633476 mgos_sys_config.c:232   Loading conf3.json
1006644044 mgos_config_util.c:110  Extra key: [relays]
1006652686 mgos_config_util.c:110  Extra key: [relays.r0]
1006663612 mgos_config_util.c:110  Extra key: [relays.r0.default_state]
1006673457 mgos_config_util.c:110  Extra key: [relays.r0.btn_reverse]
1006682154 mgos_config_util.c:110  Extra key: [relays.r0]
1006690540 mgos_config_util.c:110  Extra key: [relays]
1006819312 mgos_sys_config.c:174   Saved to conf9.json
1008150050 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.131:56990
1008172918 HAPPlatformTCPStrea:276 0x3fff5bb4 192.168.1.131:59565 Accepted HAP connection, ns 0/8/12 ts 0x3fff5aec
1008707234 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.131:56990
1008865811 shelly_main.cpp:554     Up 1008.85, HAP 0/8/12 ns 8, RAM: 20856/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0
1009707280 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.131:56990
1009950608 mgos_http_server.c:180  0x3fff6b54 HTTP connection from 192.168.1.131:59691
1012302535 HAPPlatformTCPStrea:121 0x3fff7324 192.168.1.125:50873 Incoming HAP connection, ns 0/8/12
1012318043 HAPPlatformTCPStrea:276 0x3fff7324 192.168.1.125:50873 Accepted HAP connection, ns 0/9/12 ts 0x3fff5ac4
1013463939 HAPPlatformTCPStrea:121 0x3fff79dc 192.168.1.105:50600 Incoming HAP connection, ns 0/9/12
1013479639 HAPPlatformTCPStrea:105 0x3fff7324 192.168.1.125:50873 HAP connection closed, ns 1/9/12 i 2
1013492814 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff5ac4 nc 0x0 1/8/12
1015367049 HAPPlatformTCPStrea:276 0x3fff79dc 192.168.1.105:50600 Accepted HAP connection, ns 0/9/12 ts 0x3fff5ac4
1016518195 HAPPlatformTCPStrea:105 0x3fff79dc 192.168.1.105:50600 HAP connection closed, ns 0/9/12 i 2
1016530547 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff5ac4 nc 0x0 0/8/12
1016865226 shelly_main.cpp:554     Up 1016.85, HAP 0/8/12 ns 8, RAM: 20956/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0
1017855703 HAPPlatformTCPStrea:121 0x3fff6a6c 192.168.1.130:56250 Incoming HAP connection, ns 0/8/12
1018253065 HAPPlatformTCPStrea:121 0x3fff6cfc 192.168.1.105:50603 Incoming HAP connection, ns 1/8/12
1018275077 HAPPlatformTCPStrea:121 0x3fff6d84 192.168.1.131:59719 Incoming HAP connection, ns 2/8/12
1018288920 HAPPlatformTCPStrea:121 0x3fff709c 192.168.1.125:50877 Incoming HAP connection, ns 3/8/12
1018381284 HAPPlatformTCPStrea:121 0x3fff68d4 192.168.1.130:56251 Incoming HAP connection, ns 4/8/12
1018394317 HAPPlatformTCPStrea:131 0x3fff68d4 192.168.1.130:56251 Too many pending connections, dropping
1018410266 HAPPlatformTCPStrea:276 0x3fff6a6c 192.168.1.130:56250 Accepted HAP connection, ns 3/9/12 ts 0x3fff7394
1019584083 HAPPlatformTCPStrea:105 0x3fff68d4 192.168.1.130:56251 HAP connection closed, ns 3/9/12 i 1
1019631411 HAPPlatformTCPStrea:121 0x3fff715c 192.168.1.130:56252 Incoming HAP connection, ns 3/9/12
1019645364 HAPPlatformTCPStrea:131 0x3fff715c 192.168.1.130:56252 Too many pending connections, dropping
1019665872 HAPPlatformTCPStrea:105 0x3fff715c 192.168.1.130:56252 HAP connection closed, ns 3/9/12 i 1
1019944462 HAPPlatformTCPStrea:105 0x3fff6a6c 192.168.1.130:56250 HAP connection closed, ns 3/9/12 i 2
1019957802 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff7394 nc 0x0 3/8/12
1021517904 HAPPlatformTCPStrea:276 0x3fff6cfc 192.168.1.105:50603 Accepted HAP connection, ns 2/9/12 ts 0x3fff7394
1022689072 HAPPlatformTCPStrea:105 0x3fff6cfc 192.168.1.105:50603 HAP connection closed, ns 2/9/12 i 2
1022701439 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff7394 nc 0x0 2/8/12
1024862696 HAPPlatformTCPStrea:276 0x3fff6d84 192.168.1.131:59719 Accepted HAP connection, ns 1/9/12 ts 0x3fff5ac4
1026026134 shelly_main.cpp:554     Up 1026.02, HAP 1/9/12 ns 9, RAM: 16480/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0
1027287534 HAPPlatformTCPStrea:105 0x3fff6d84 192.168.1.131:59719 HAP connection closed, ns 1/9/12 i 2
1027300832 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff5ac4 nc 0x0 1/8/12
1028024699 HAPPlatformTCPStrea:276 0x3fff709c 192.168.1.125:50877 Accepted HAP connection, ns 0/9/12 ts 0x3fff5ac4
1029160278 HAPPlatformTCPStrea:105 0x3fff709c 192.168.1.125:50877 HAP connection closed, ns 0/9/12 i 2
1029172644 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff5ac4 nc 0x0 0/8/12
1031965478 HAPPlatformTCPStrea:121 0x3fff5d0c 192.168.1.105:50606 Incoming HAP connection, ns 0/8/12
1031981005 HAPPlatformTCPStrea:276 0x3fff5d0c 192.168.1.105:50606 Accepted HAP connection, ns 0/9/12 ts 0x3fff5ac4
1033184296 HAPPlatformTCPStrea:105 0x3fff5d0c 192.168.1.105:50606 HAP connection closed, ns 0/9/12 i 2
1033197574 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff5ac4 nc 0x0 0/8/12
1034047850 HAPPlatformTCPStrea:121 0x3fff5d0c 192.168.1.125:50884 Incoming HAP connection, ns 0/8/12
1035130228 HAPPlatformTCPStrea:276 0x3fff5d0c 192.168.1.125:50884 Accepted HAP connection, ns 0/9/12 ts 0x3fff5ac4
1035148853 shelly_main.cpp:554     Up 1035.14, HAP 0/9/12 ns 9, RAM: 18744/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0
1036389387 HAPPlatformTCPStrea:105 0x3fff5d0c 192.168.1.125:50884 HAP connection closed, ns 0/9/12 i 2
1036402684 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff5ac4 nc 0x0 0/8/12
[... cut ...]
1040409043 HAPPlatformTCPStrea:105 0x3fff5d0c 192.168.1.130:56253 HAP connection closed, ns 0/9/12 i 2
1040421394 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff5ac4 nc 0x0 0/8/12
1043132205 shelly_main.cpp:554     Up 1043.12, HAP 0/8/12 ns 8, RAM: 20956/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0
1051132273 shelly_main.cpp:554     Up 1051.12, HAP 0/8/12 ns 8, RAM: 20916/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0
1059131916 shelly_main.cpp:554     Up 1059.12, HAP 0/8/12 ns 8, RAM: 20956/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0

rojer, 2021-03-14:
> 1018381284 HAPPlatformTCPStrea:121 0x3fff68d4 192.168.1.130:56251 Incoming HAP connection, ns 4/8/12
> 1018394317 HAPPlatformTCPStrea:131 0x3fff68d4 192.168.1.130:56251 Too many pending connections, dropping

that is actually working as expected. we only support 12 connection max, here we receive a new connection while already having 8 established and 4 pending, so this would be the 13th, we therefore drop it.
looks like you have  many iDevices on the network, rebooting a HAP device always causes a bit of a stampede with everyone trying to connect. we rate limit incoming connections because handshake requires extra CPU and memory, if we didn't we'd run out. so seeing a slowdown immediately after reboot is kind of expected, just let it settle a bit.

Dag0d, 2021-03-14:
Here is another device, which turned on and off with HomeKit during the log capture, but at a certain point, it didn't turn off when turning it off in HomeKit, after a tries it eventually worked:

46372183:1615680980882177 shelly_rpc_service.:226 SetConfig ok, 0
46400134 mgos_sys_config.c:232   Loading conf2.json
46423839 mgos_sys_config.c:232   Loading conf3.json
46434372 mgos_config_util.c:110  Extra key: [relays]
46443000 mgos_config_util.c:110  Extra key: [relays.r0]
46453894 mgos_config_util.c:110  Extra key: [relays.r0.default_state]
46463730 mgos_config_util.c:110  Extra key: [relays.r0.btn_reverse]
46472413 mgos_config_util.c:110  Extra key: [relays.r0]
46480785 mgos_config_util.c:110  Extra key: [relays]
46608285 mgos_sys_config.c:174   Saved to conf9.json
47940201 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.131:61009
48350492 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.131:61009
49062753 shelly_debug.cpp:203    Tailing
52308930 shelly_main.cpp:554     Up 52.30, HAP 0/5/12 ns 5, RAM: 22352/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0
59087929 HAPPlatformTCPStrea:121 0x3fff6b04 192.168.1.105:50632 Incoming HAP connection, ns 0/5/12
59102513 HAPPlatformTCPStrea:276 0x3fff6b04 192.168.1.105:50632 Accepted HAP connection, ns 0/6/12 ts 0x3fff692c
60200072 HAPPlatformTCPStrea:105 0x3fff6b04 192.168.1.105:50632 HAP connection closed, ns 0/6/12 i 2
60212353 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff692c nc 0x0 0/5/12
60232644 HAPPlatformTCPStrea:121 0x3fff6b04 192.168.1.127:63000 Incoming HAP connection, ns 0/5/12
60309466 shelly_main.cpp:554     Up 60.30, HAP 1/5/12 ns 5, RAM: 20300/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0
62306734 HAPPlatformTCPStrea:276 0x3fff6b04 192.168.1.127:63000 Accepted HAP connection, ns 0/6/12 ts 0x3fff692c
63389696 HAPPlatformTCPStrea:105 0x3fff6b04 192.168.1.127:63000 HAP connection closed, ns 0/6/12 i 2
63401999 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff692c nc 0x0 0/5/12
66192460 shelly_output.cpp:58    Output 1: off -> on (HAP)
67532888 shelly_output.cpp:58    Output 1: on -> off (HAP)
69399522 mgos_sys_config.c:232   Loading conf2.json
69423770 mgos_sys_config.c:232   Loading conf3.json
69435537 mgos_config_util.c:110  Extra key: [relays]
69444594 mgos_config_util.c:110  Extra key: [relays.r0]
69455029 mgos_config_util.c:110  Extra key: [relays.r0.default_state]
69465583 mgos_config_util.c:110  Extra key: [relays.r0.btn_reverse]
69475655 mgos_config_util.c:110  Extra key: [relays.r0]
69484487 mgos_config_util.c:110  Extra key: [relays]
69612625 mgos_sys_config.c:174   Saved to conf9.json
69629487 shelly_main.cpp:554     Up 69.62, HAP 0/5/12 ns 5, RAM: 21700/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0
69648813 shelly_output.cpp:58    Output 1: off -> on (HAP)
72278489 shelly_output.cpp:58    Output 1: on -> off (HAP)
74966505 shelly_output.cpp:58    Output 1: off -> on (HAP)
77399654 mgos_sys_config.c:232   Loading conf2.json
77424829 mgos_sys_config.c:232   Loading conf3.json
77435678 mgos_config_util.c:110  Extra key: [relays]
77444743 mgos_config_util.c:110  Extra key: [relays.r0]
77455185 mgos_config_util.c:110  Extra key: [relays.r0.default_state]
77466692 mgos_config_util.c:110  Extra key: [relays.r0.btn_reverse]
77475848 mgos_config_util.c:110  Extra key: [relays.r0]
77484693 mgos_config_util.c:110  Extra key: [relays]
77612977 mgos_sys_config.c:174   Saved to conf9.json
77629849 shelly_main.cpp:554     Up 77.62, HAP 0/5/12 ns 5, RAM: 21700/38568; st 0; 0.1: st:1 in_st:0 inm:2 ininv:0
80463450 HAPPlatformTCPStrea:105 0x3fff6b94 192.168.1.60:49991 HAP connection closed, ns 0/5/12 i 2
80475753 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff6a24 nc 0x0 0/4/12
85381981 shelly_main.cpp:554     Up 85.37, HAP 0/4/12 ns 4, RAM: 22760/38568; st 0; 0.1: st:1 in_st:0 inm:2 ininv:0
90460718 HAPPlatformTCPStrea:105 0x3fff6974 192.168.1.125:50913 HAP connection closed, ns 0/4/12 i 2
90473013 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff6c4c nc 0x0 0/3/12
90492315 HAPPlatformTCPStrea:105 0x3fff6544 192.168.1.105:50629 HAP connection closed, ns 0/3/12 i 2
90504591 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff69fc nc 0x0 0/2/12
91261726 HAPPlatformTCPStrea:121 0x3fff5e6c 192.168.1.105:50637 Incoming HAP connection, ns 0/2/12
91276353 HAPPlatformTCPStrea:276 0x3fff5e6c 192.168.1.105:50637 Accepted HAP connection, ns 0/3/12 ts 0x3fff692c
92354544 HAPPlatformTCPStrea:121 0x3fff5ff4 192.168.1.127:63006 Incoming HAP connection, ns 0/3/12
92556494 HAPPlatformTCPStrea:105 0x3fff5e6c 192.168.1.105:50637 HAP connection closed, ns 1/3/12 i 2
92568806 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff692c nc 0x0 1/2/12
93381631 shelly_main.cpp:554     Up 93.37, HAP 1/2/12 ns 2, RAM: 23056/38568; st 0; 0.1: st:1 in_st:0 inm:2 ininv:0
94379167 HAPPlatformTCPStrea:276 0x3fff5ff4 192.168.1.127:63006 Accepted HAP connection, ns 0/3/12 ts 0x3fff692c
95446217 HAPPlatformTCPStrea:105 0x3fff5ff4 192.168.1.127:63006 HAP connection closed, ns 0/3/12 i 2
95459464 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff692c nc 0x0 0/2/12
102436714 shelly_main.cpp:554     Up 102.43, HAP 0/2/12 ns 2, RAM: 23580/38568; st 0; 0.1: st:1 in_st:0 inm:2 ininv:0
110436648 shelly_main.cpp:554     Up 110.43, HAP 0/2/12 ns 2, RAM: 23580/38568; st 0; 0.1: st:1 in_st:0 inm:2 ininv:0
116323089 HAPPlatformTCPStrea:121 0x3fff5e6c 192.168.1.125:50927 Incoming HAP connection, ns 0/2/12
[... cut ...]
130907714 mgos_config_util.c:110  Extra key: [relays]
130916843 mgos_config_util.c:110  Extra key: [relays.r0]
130927350 mgos_config_util.c:110  Extra key: [relays.r0.default_state]
130938946 mgos_config_util.c:110  Extra key: [relays.r0.btn_reverse]
130948162 mgos_config_util.c:110  Extra key: [relays.r0]
130957068 mgos_config_util.c:110  Extra key: [relays]
131085875 mgos_sys_config.c:174   Saved to conf9.json
131103810 shelly_main.cpp:554     Up 131.09, HAP 2/3/12 ns 3, RAM: 21428/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0

rojer, 2021-03-14:
@Dag0d ok, try this one - http://rojer.me/files/shelly/2.8.2-beta2/
i reduced the interval at which we accept new connections from 3 seconds to 1.5, this should reduce the time they spend in the queue. i think controller just loses patience and gives up when made to wait for too long.

tojorocha, 2021-03-14:
I have 31 shellys with mongoose. I tried in 3 of them the firmware that you recommend @rojer. They connected to homekit for a short time and then they give a “no response”. I revert the 3 again to 2.7.3 and they stay rock solid in homekit.

rojer, 2021-03-14:
@tojorocha can you try http://rojer.me/files/shelly/2.8.2-beta2/ ? and post debug log

Dag0d, 2021-03-14:
@rojer still pretty much the same, see the log below, before it turned on the first time, I tried at least 4 times. Same applies for the turn off after the second time it turned on.

133768082:1615682570265100 shelly_rpc_service.:226 SetConfig ok, 0
133801092 mgos_sys_config.c:232   Loading conf2.json
133824642 mgos_sys_config.c:232   Loading conf3.json
133835109 mgos_config_util.c:110  Extra key: [relays]
133843716 mgos_config_util.c:110  Extra key: [relays.r0]
133854613 mgos_config_util.c:110  Extra key: [relays.r0.default_state]
133864426 mgos_config_util.c:110  Extra key: [relays.r0.btn_reverse]
133873093 mgos_config_util.c:110  Extra key: [relays.r0]
133881447 mgos_config_util.c:110  Extra key: [relays]
134008518 mgos_sys_config.c:174   Saved to conf9.json
135337990 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.131:50713
136264836 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.131:50713
136954570 mgos_http_server.c:180  0x3fff6304 HTTP connection from 192.168.1.131:50849
137094414 shelly_main.cpp:554     Up 137.08, HAP 0/1/12 ns 1, RAM: 24104/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0
145093997 shelly_main.cpp:554     Up 145.08, HAP 0/1/12 ns 1, RAM: 24104/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0
146322623 HAPPlatformTCPStrea:121 0x3fff5a7c 192.168.1.130:56470 Incoming HAP connection, ns 0/1/12
146337188 HAPPlatformTCPStrea:276 0x3fff5a7c 192.168.1.130:56470 Accepted HAP connection, ns 0/2/12 ts 0x3fff66ac
148163447 HAPPlatformTCPStrea:105 0x3fff5a7c 192.168.1.130:56470 HAP connection closed, ns 0/2/12 i 2
148175722 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff66ac nc 0x0 0/1/12
150569790 HAPPlatformTCPStrea:121 0x3fff598c 192.168.1.130:56473 Incoming HAP connection, ns 0/1/12
150584367 HAPPlatformTCPStrea:276 0x3fff598c 192.168.1.130:56473 Accepted HAP connection, ns 0/2/12 ts 0x3fff5894
151634365 mgos_mongoose.c:66      New heap free LWM: 16328
151991950 HAPPlatformTCPStrea:121 0x3fff6944 192.168.1.60:50167 Incoming HAP connection, ns 0/2/12
152098574 HAPPlatformTCPStrea:276 0x3fff6944 192.168.1.60:50167 Accepted HAP connection, ns 0/3/12 ts 0x3fff62dc
153182798 shelly_main.cpp:554     Up 153.17, HAP 0/3/12 ns 3, RAM: 17412/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0
153193154 mgos_mongoose.c:66      New heap free LWM: 16064
153219410 HAPPlatformTCPStrea:105 0x3fff6944 192.168.1.60:50167 HAP connection closed, ns 0/3/12 i 2
153232599 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff62dc nc 0x0 0/2/12
153576496 HAPPlatformTCPStrea:105 0x3fff598c 192.168.1.130:56473 HAP connection closed, ns 0/2/12 i 2
153588756 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff5894 nc 0x0 0/1/12
161094024 shelly_main.cpp:554     Up 161.08, HAP 0/1/12 ns 1, RAM: 24104/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0
169094255 shelly_main.cpp:554     Up 169.08, HAP 0/1/12 ns 1, RAM: 24104/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0
169940468 HAPPlatformTCPStrea:121 0x3fff677c 192.168.1.130:56481 Incoming HAP connection, ns 0/1/12
169955056 HAPPlatformTCPStrea:276 0x3fff677c 192.168.1.130:56481 Accepted HAP connection, ns 0/2/12 ts 0x3fff62dc
171006527 HAPPlatformTCPStrea:121 0x3fff654c 192.168.1.105:51061 Incoming HAP connection, ns 0/2/12
171350796 HAPPlatformTCPStrea:105 0x3fff677c 192.168.1.130:56481 HAP connection closed, ns 1/2/12 i 2
171363057 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff62dc nc 0x0 1/1/12
171472801 HAPPlatformTCPStrea:276 0x3fff654c 192.168.1.105:51061 Accepted HAP connection, ns 0/2/12 ts 0x3fff62dc
172588186 HAPPlatformTCPStrea:105 0x3fff654c 192.168.1.105:51061 HAP connection closed, ns 0/2/12 i 2
172600478 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff62dc nc 0x0 0/1/12
177094219 shelly_main.cpp:554     Up 177.08, HAP 0/1/12 ns 1, RAM: 24104/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0
177786387 shelly_output.cpp:58    Output 1: off -> on (HAP)
185112158 mgos_sys_config.c:232   Loading conf2.json
185136291 mgos_sys_config.c:232   Loading conf3.json
185147120 mgos_config_util.c:110  Extra key: [relays]
185156197 mgos_config_util.c:110  Extra key: [relays.r0]
185166648 mgos_config_util.c:110  Extra key: [relays.r0.default_state]
185178171 mgos_config_util.c:110  Extra key: [relays.r0.btn_reverse]
185187332 mgos_config_util.c:110  Extra key: [relays.r0]
185196182 mgos_config_util.c:110  Extra key: [relays]
185324246 mgos_sys_config.c:174   Saved to conf9.json
185342118 shelly_main.cpp:554     Up 185.33, HAP 0/1/12 ns 1, RAM: 23448/38568; st 0; 0.1: st:1 in_st:0 inm:2 ininv:0
188497382 shelly_output.cpp:58    Output 1: on -> off (HAP)
193111837 mgos_sys_config.c:232   Loading conf2.json
193136925 mgos_sys_config.c:232   Loading conf3.json
193147761 mgos_config_util.c:110  Extra key: [relays]
193156837 mgos_config_util.c:110  Extra key: [relays.r0]
193167295 mgos_config_util.c:110  Extra key: [relays.r0.default_state]
193177900 mgos_config_util.c:110  Extra key: [relays.r0.btn_reverse]
193188008 mgos_config_util.c:110  Extra key: [relays.r0]
193196865 mgos_config_util.c:110  Extra key: [relays]
193325008 mgos_sys_config.c:174   Saved to conf9.json
193341942 shelly_main.cpp:554     Up 193.33, HAP 0/1/12 ns 1, RAM: 23448/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0
193361197 shelly_output.cpp:58    Output 1: off -> on (HAP)
194258538 HAPPlatformTCPStrea:121 0x3fff5a7c 192.168.1.105:51069 Incoming HAP connection, ns 0/1/12
[... cut ...]
221009650 HAPPlatformTCPStrea:121 0x3fff651c 192.168.1.60:50173 Incoming HAP connection, ns 0/2/12
221025126 HAPPlatformTCPStrea:276 0x3fff651c 192.168.1.60:50173 Accepted HAP connection, ns 0/3/12 ts 0x3fff69fc
222080685 mgos_mongoose.c:66      New heap free LWM: 15616
222100879 mgos_mongoose.c:66      New heap free LWM: 13584
222498180 HAPPlatformTCPStrea:105 0x3fff638c 192.168.1.130:56486 HAP connection closed, ns 0/3/12 i 2
222510532 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff62dc nc 0x0 0/2/12

tojorocha, 2021-03-14:
@rojer exactly the same

91487260 HAPPlatformTCPStrea:276 0x3fff5fbc 192.168.2.123:51215 Accepted HAP connection, ns 0/4/12 ts 0x3fff710c
92544666 HAPPlatformTCPStrea:105 0x3fff5fbc 192.168.2.123:51215 HAP connection closed, ns 0/4/12 i 2
92556462 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff710c nc 0x0 0/3/12
97377890 HAPPlatformTCPStrea:121 0x3fff5fbc 192.168.2.106:58271 Incoming HAP connection, ns 0/3/12
97392030 HAPPlatformTCPStrea:276 0x3fff5fbc 192.168.2.106:58271 Accepted HAP connection, ns 0/4/12 ts 0x3fff710c
98504168 HAPPlatformTCPStrea:105 0x3fff5fbc 192.168.2.106:58271 HAP connection closed, ns 0/4/12 i 2
98515972 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff710c nc 0x0 0/3/12
98708247 shelly_main.cpp:554     Up 98.70, HAP 0/3/12 ns 3, RAM: 23224/38568; st 0; 0.1: st:0 in_st:0 inm:0 ininv:0
99952038 mgos_http_server.c:187  0x3fff48bc GET /
100004352 mgos_http_server.c:180  0x3fff5fbc HTTP connection from 192.168.2.123:51216
100036218 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.123:51216
104842064 HAPPlatformTCPStrea:121 0x3fff5fbc 192.168.2.193:60951 Incoming HAP connection, ns 0/3/12
104856217 HAPPlatformTCPStrea:276 0x3fff5fbc 192.168.2.193:60951 Accepted HAP connection, ns 0/4/12 ts 0x3fff710c
105981785 HAPPlatformTCPStrea:105 0x3fff5fbc 192.168.2.193:60951 HAP connection closed, ns 0/4/12 i 2
105994526 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff710c nc 0x0 0/3/12
106708378 shelly_main.cpp:554     Up 106.70, HAP 0/3/12 ns 3, RAM: 23224/38568; st 0; 0.1: st:0 in_st:0 inm:0 ininv:0
108267323 HAPPlatformTCPStrea:121 0x3fff5fbc 192.168.2.123:51218 Incoming HAP connection, ns 0/3/12
108281479 HAPPlatformTCPStrea:276 0x3fff5fbc 192.168.2.123:51218 Accepted HAP connection, ns 0/4/12 ts 0x3fff710c
109404657 HAPPlatformTCPStrea:105 0x3fff5fbc 192.168.2.123:51218 HAP connection closed, ns 0/4/12 i 2
109416465 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff710c nc 0x0 0/3/12
110447114 HAPPlatformTCPStrea:121 0x3fff5fbc 192.168.2.13:58789 Incoming HAP connection, ns 0/3/12
110462109 HAPPlatformTCPStrea:276 0x3fff5fbc 192.168.2.13:58789 Accepted HAP connection, ns 0/4/12 ts 0x3fff710c
111542453 HAPPlatformTCPStrea:121 0x3fff7054 192.168.2.123:51219 Incoming HAP connection, ns 0/4/12
111592949 HAPPlatformTCPStrea:105 0x3fff5fbc 192.168.2.13:58789 HAP connection closed, ns 1/4/12 i 2
111604767 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff710c nc 0x0 1/3/12
111870063 HAPPlatformTCPStrea:121 0x3fff5fbc 192.168.2.47:58075 Incoming HAP connection, ns 1/3/12
112692924 HAPPlatformTCPStrea:276 0x3fff7054 192.168.2.123:51219 Accepted HAP connection, ns 1/4/12 ts 0x3fff5974
113753481 HAPPlatformTCPStrea:105 0x3fff7054 192.168.2.123:51219 HAP connection closed, ns 1/4/12 i 2
113766242 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff5974 nc 0x0 1/3/12
114459702 HAPPlatformTCPStrea:276 0x3fff5fbc 192.168.2.47:58075 Accepted HAP connection, ns 0/4/12 ts 0x3fff5974
115604598 HAPPlatformTCPStrea:105 0x3fff5fbc 192.168.2.47:58075 HAP connection closed, ns 0/4/12 i 2
115617367 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff5974 nc 0x0 0/3/12
115755136 shelly_main.cpp:554     Up 115.74, HAP 0/3/12 ns 3, RAM: 23224/38568; st 0; 0.1: st:0 in_st:0 inm:0 ininv:0
117790161 HAPPlatformTCPStrea:121 0x3fff5fbc 192.168.2.106:58275 Incoming HAP connection, ns 0/3/12
117804342 HAPPlatformTCPStrea:276 0x3fff5fbc 192.168.2.106:58275 Accepted HAP connection, ns 0/4/12 ts 0x3fff710c
118889540 HAPPlatformTCPStrea:105 0x3fff5fbc 192.168.2.106:58275 HAP connection closed, ns 0/4/12 i 2
118902301 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff710c nc 0x0 0/3/12
121996504 HAPPlatformTCPStrea:121 0x3fff5fbc 192.168.2.123:51224 Incoming HAP connection, ns 0/3/12
122011608 HAPPlatformTCPStrea:276 0x3fff5fbc 192.168.2.123:51224 Accepted HAP connection, ns 0/4/12 ts 0x3fff710c
123514301 HAPPlatformTCPStrea:105 0x3fff5fbc 192.168.2.123:51224 HAP connection closed, ns 0/4/12 i 2
123526767 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff710c nc 0x0 0/3/12
123755267 shelly_main.cpp:554     Up 123.74, HAP 0/3/12 ns 3, RAM: 23224/38568; st 0; 0.1: st:0 in_st:0 inm:0 ininv:0
124805850 HAPPlatformTCPStrea:121 0x3fff5fbc 192.168.2.193:60955 Incoming HAP connection, ns 0/3/12
124820954 HAPPlatformTCPStrea:276 0x3fff5fbc 192.168.2.193:60955 Accepted HAP connection, ns 0/4/12 ts 0x3fff710c
125859309 HAPPlatformTCPStrea:105 0x3fff5fbc 192.168.2.193:60955 HAP connection closed, ns 0/4/12 i 2
125872084 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff710c nc 0x0 0/3/12
129348181 HAPPlatformTCPStrea:121 0x3fff5fbc 192.168.2.193:60960 Incoming HAP connection, ns 0/3/12
129362369 HAPPlatformTCPStrea:276 0x3fff5fbc 192.168.2.193:60960 Accepted HAP connection, ns 0/4/12 ts 0x3fff710c
130454571 HAPPlatformTCPStrea:105 0x3fff5fbc 192.168.2.193:60960 HAP connection closed, ns 0/4/12 i 2
130466420 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff710c nc 0x0 0/3/12
131754903 shelly_main.cpp:554     Up 131.74, HAP 0/3/12 ns 3, RAM: 23716/38568; st 0; 0.1: st:0 in_st:0 inm:0 ininv:0
139407229 HAPPlatformTCPStrea:121 0x3fff5fbc 192.168.2.47:58081 Incoming HAP connection, ns 0/3/12
139421327 HAPPlatformTCPStrea:276 0x3fff5fbc 192.168.2.47:58081 Accepted HAP connection, ns 0/4/12 ts 0x3fff5974
140463813 HAPPlatformTCPStrea:121 0x3fff6f24 192.168.2.106:58276 Incoming HAP connection, ns 0/4/12
[... cut ...]
255562616 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff42c4 nc 0x0 1/2/12
256173880 HAPPlatformTCPStrea:276 0x3fff488c 192.168.2.5:52363 Accepted HAP connection, ns 0/3/12 ts 0x3fff42c4
257275804 HAPPlatformTCPStrea:105 0x3fff488c 192.168.2.5:52363 HAP connection closed, ns 0/3/12 i 2
257288165 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff42c4 nc 0x0 0/2/12
262866025 shelly_main.cpp:554     Up 262.86, HAP 0/2/12 ns 2, RAM: 23648/38568; st 0; 0.1: st:0 in_st:0 inm:0 ininv:0

rojer, 2021-03-14:
all this frantic activity by controllers makes no sense to me... they open multiple connections, then drop them immediately - all HAPPlatformTCPStreamClose are caused by remote side (controller), not us.

tojorocha, 2021-03-14:
Put them again on 2.7.3 and everything flawless

Dag0d, 2021-03-14:
I have 2 apple TVs, 1 HomePod, 3 iPhones, 2 apple watches, 2 MacBooks and 1 Mac mini running here, this wasn't an issue with 2.7.3.
My Shelly 1PM works flawless with 2.8.1 it's only the Shelly 1

Dag0d, 2021-03-14:
@rojer the immediate close always happens when HomeKit says "No response"

tojorocha, 2021-03-14:
My Shelly 2.5 works on 2.8.1 but not so good like 2.7.3. Sometimes they stay in “no response” too. That doesn’t happened in 2.7.3. The Shelly 1 don’t work with 2.8.1

rojer, 2021-03-14:
ok, i think i reproduced it locally, investigating.

tojorocha, 2021-03-14:
Thanks @rojer

rojer, 2021-03-14:
@tojorocha @Dag0d ok, i think i may have nabbed the bug. please update to [beta3](http://rojer.me/files/shelly/2.8.2-beta3/) (updtae button should work too) and let me know.

iperniaf, 2021-03-14:
I have updated to beta 3 and I think it’s working ok now

Dag0d, 2021-03-14:
for me it's still bugy, I've seen no response sometimes if it looks normal in homeKit an. I try too change the state it won't do anything. but strangely if I change the state using the shelly web interface HomeKit shows the state accordingly after ~7 seconds.
here's a log when I tried several times to turn it on in HomeKit but it didn't do nothing:

110426340 shelly_debug.cpp:227    No log file, sending new entries
111224391 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.131:52035
112220740 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.131:52035
113218287 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.131:52035
114221522 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.131:52035
115218896 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.131:52035
116096393 HAPPlatformTCPStrea:121 0x3fff5f2c 192.168.1.127:64608 Incoming HAP connection, ns 0/2/12
116106358 HAPPlatformTCPStrea:278 0x3fff5f2c 192.168.1.127:64608 Accepted HAP connection, ns 0/3/12 ts 0x3fff5ac4
117201194 HAPPlatformTCPStrea:121 0x3fff67fc 192.168.1.125:54327 Incoming HAP connection, ns 0/3/12
117210964 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.131:52035
117229756 HAPPlatformTCPStrea:105 0x3fff5f2c 192.168.1.127:64608 HAP connection closed, ns 1/3/12 i 2
117237461 HAPPlatformTCPStrea:305 HAPPlatformTCPStreamClose ts 0x3fff5ac4 nc 0x0 1/2/12
117282844 shelly_main.cpp:554     Up 117.27, HAP 1/2/12 ns 2, RAM: 20388/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0
117295855 HAPPlatformTCPStrea:121 0x3fff5e74 192.168.1.127:64609 Incoming HAP connection, ns 1/2/12
117311414 HAPPlatformTCPStrea:121 0x3fff66b4 192.168.1.60:50594 Incoming HAP connection, ns 2/2/12
117828826 HAPPlatformTCPStrea:278 0x3fff67fc 192.168.1.125:54327 Accepted HAP connection, ns 2/3/12 ts 0x3fff5ac4
118936249 HAPPlatformTCPStrea:105 0x3fff67fc 192.168.1.125:54327 HAP connection closed, ns 2/3/12 i 2
118944195 HAPPlatformTCPStrea:305 HAPPlatformTCPStreamClose ts 0x3fff5ac4 nc 0x0 2/2/12
118992411 HAPPlatformTCPStrea:121 0x3fff600c 192.168.1.60:50596 Incoming HAP connection, ns 2/2/12
119355346 HAPPlatformTCPStrea:278 0x3fff5e74 192.168.1.127:64609 Accepted HAP connection, ns 2/3/12 ts 0x3fff5ac4
120480372 HAPPlatformTCPStrea:105 0x3fff5e74 192.168.1.127:64609 HAP connection closed, ns 2/3/12 i 2
120488317 HAPPlatformTCPStrea:305 HAPPlatformTCPStreamClose ts 0x3fff5ac4 nc 0x0 2/2/12
120567132 HAPPlatformTCPStrea:121 0x3fff67b4 192.168.1.125:54328 Incoming HAP connection, ns 2/2/12
120817674 HAPPlatformTCPStrea:121 0x3fff729c 192.168.1.125:54332 Incoming HAP connection, ns 3/2/12
120892178 HAPPlatformTCPStrea:278 0x3fff66b4 192.168.1.60:50594 Accepted HAP connection, ns 3/3/12 ts 0x3fff5ac4
122001747 HAPPlatformTCPStrea:105 0x3fff66b4 192.168.1.60:50594 HAP connection closed, ns 3/3/12 i 2
122009599 HAPPlatformTCPStrea:305 HAPPlatformTCPStreamClose ts 0x3fff5ac4 nc 0x0 3/2/12
122431392 HAPPlatformTCPStrea:278 0x3fff600c 192.168.1.60:50596 Accepted HAP connection, ns 2/3/12 ts 0x3fff402c
123948719 HAPPlatformTCPStrea:278 0x3fff67b4 192.168.1.125:54328 Accepted HAP connection, ns 1/4/12 ts 0x3fff5ac4
125282821 shelly_main.cpp:554     Up 125.27, HAP 1/4/12 ns 4, RAM: 19788/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0
125460012 HAPPlatformTCPStrea:278 0x3fff729c 192.168.1.125:54332 Accepted HAP connection, ns 0/5/12 ts 0x3fff6944
126565672 HAPPlatformTCPStrea:105 0x3fff729c 192.168.1.125:54332 HAP connection closed, ns 0/5/12 i 2
126573624 HAPPlatformTCPStrea:305 HAPPlatformTCPStreamClose ts 0x3fff6944 nc 0x0 0/4/12
126597797 HAPPlatformTCPStrea:105 0x3fff67b4 192.168.1.125:54328 HAP connection closed, ns 0/4/12 i 2
126605606 HAPPlatformTCPStrea:305 HAPPlatformTCPStreamClose ts 0x3fff5ac4 nc 0x0 0/3/12
128145587 HAPPlatformTCPStrea:105 0x3fff600c 192.168.1.60:50596 HAP connection closed, ns 0/3/12 i 2
128154090 HAPPlatformTCPStrea:305 HAPPlatformTCPStreamClose ts 0x3fff402c nc 0x0 0/2/12
132249039 HAPPlatformTCPStrea:121 0x3fff5604 192.168.1.127:64621 Incoming HAP connection, ns 0/2/12
132259012 HAPPlatformTCPStrea:278 0x3fff5604 192.168.1.127:64621 Accepted HAP connection, ns 0/3/12 ts 0x3fff402c
133340284 HAPPlatformTCPStrea:105 0x3fff5604 192.168.1.127:64621 HAP connection closed, ns 0/3/12 i 2
133348230 HAPPlatformTCPStrea:305 HAPPlatformTCPStreamClose ts 0x3fff402c nc 0x0 0/2/12
134320886 shelly_main.cpp:554     Up 134.31, HAP 0/2/12 ns 2, RAM: 24724/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0
135611856 HAPPlatformTCPStrea:121 0x3fff5604 192.168.1.60:50602 Incoming HAP connection, ns 0/2/12
135621733 HAPPlatformTCPStrea:278 0x3fff5604 192.168.1.60:50602 Accepted HAP connection, ns 0/3/12 ts 0x3fff5ac4
136671449 HAPPlatformTCPStrea:105 0x3fff5604 192.168.1.60:50602 HAP connection closed, ns 0/3/12 i 2
136679368 HAPPlatformTCPStrea:305 HAPPlatformTCPStreamClose ts 0x3fff5ac4 nc 0x0 0/2/12
142320920 shelly_main.cpp:554     Up 142.31, HAP 0/2/12 ns 2, RAM: 24724/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0
144483292 HAPPlatformTCPStrea:121 0x3fff5e74 192.168.1.60:50603 Incoming HAP connection, ns 0/2/12
144493163 HAPPlatformTCPStrea:278 0x3fff5e74 192.168.1.60:50603 Accepted HAP connection, ns 0/3/12 ts 0x3fff402c
146143483 HAPPlatformTCPStrea:121 0x3fff6754 192.168.1.125:54340 Incoming HAP connection, ns 0/3/12
146153480 HAPPlatformTCPStrea:278 0x3fff6754 192.168.1.125:54340 Accepted HAP connection, ns 0/4/12 ts 0x3fff5be4
[... cut ...]
183136779 HAPPlatformTCPStrea:305 HAPPlatformTCPStreamClose ts 0x3fff5bb4 nc 0x0 1/4/12
183560246 HAPPlatformTCPStrea:278 0x3fff67fc 192.168.1.130:57351 Accepted HAP connection, ns 0/5/12 ts 0x3fff5bb4
184282151 HAPPlatformTCPStrea:121 0x3fff5efc 192.168.1.125:54349 Incoming HAP connection, ns 0/5/12
185605708 HAPPlatformTCPStrea:278 0x3fff5efc 192.168.1.125:54349 Accepted HAP connection, ns 0/6/12 ts 0x3fff5bdc
186684132 shelly_main.cpp:554     Up 186.67, HAP 0/6/12 ns 6, RAM: 21132/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0

chstoll, 2021-03-14:
I can confirm, still not working 100% again like with 2.7.3. 2 out of 5 times the Plug doesn't change state if I toggle it in Home-App. It becomes unresponsive for a short period and then comes back again. Here's a log, but I've noticed that there is no entry generated if toggling didn't work. 

272050014 shelly_debug.cpp:227    No log file, sending new entries
272715541 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.111.1:52077
273715085 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.111.1:52077
274121878 shelly_main.cpp:554     Up 274.11, HAP 0/3/12 ns 3, RAM: 23024/38544; st 32; 0.1: st:1 in_st:-1 inm:-1 ininv:0
274713846 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.111.1:52077
275713943 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.111.1:52077
276717641 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.111.1:52077
277718501 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.111.1:52077
278718846 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.111.1:52077
279715502 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.111.1:52077
280715026 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.111.1:52077
281719729 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.111.1:52077
282122010 shelly_main.cpp:554     Up 282.11, HAP 0/3/12 ns 3, RAM: 23024/38544; st 32; 0.1: st:1 in_st:-1 inm:-1 ininv:0
290121726 shelly_main.cpp:554     Up 290.11, HAP 0/3/12 ns 3, RAM: 23612/38544; st 32; 0.1: st:1 in_st:-1 inm:-1 ininv:0
298122266 shelly_main.cpp:554     Up 298.11, HAP 0/3/12 ns 3, RAM: 23612/38544; st 32; 0.1: st:1 in_st:-1 inm:-1 ininv:0
306121826 shelly_main.cpp:554     Up 306.11, HAP 0/3/12 ns 3, RAM: 23612/38544; st 32; 0.1: st:1 in_st:-1 inm:-1 ininv:0
314122105 shelly_main.cpp:554     Up 314.11, HAP 0/3/12 ns 3, RAM: 23612/38544; st 32; 0.1: st:1 in_st:-1 inm:-1 ininv:0
316014751 HAPPlatformTCPStrea:121 0x3fff57d4 192.168.111.157:54132 Incoming HAP connection, ns 0/3/12
316025138 HAPPlatformTCPStrea:278 0x3fff57d4 192.168.111.157:54132 Accepted HAP connection, ns 0/4/12 ts 0x3fff6cc4
322121752 shelly_main.cpp:554     Up 322.11, HAP 0/4/12 ns 4, RAM: 22940/38544; st 32; 0.1: st:1 in_st:-1 inm:-1 ininv:0
325267027 shelly_output.cpp:58    Output 1: on -> off (HAP)
325272482 shelly_output.cpp:58    Output 99: on -> off (HAP)
329433376 shelly_output.cpp:58    Output 1: off -> on (HAP)
329438806 shelly_output.cpp:58    Output 99: off -> on (HAP)
330139221 mgos_sys_config.c:232   Loading conf2.json
330276348 mgos_sys_config.c:174   Saved to conf9.json
330289543 shelly_main.cpp:554     Up 330.28, HAP 0/4/12 ns 4, RAM: 22624/38544; st 32; 0.1: st:1 in_st:-1 inm:-1 ininv:0
337302710 shelly_output.cpp:58    Output 1: on -> off (HAP)
337308134 shelly_output.cpp:58    Output 99: on -> off (HAP)
338139554 mgos_sys_config.c:232   Loading conf2.json
338276426 mgos_sys_config.c:174   Saved to conf9.json
338289621 shelly_main.cpp:554     Up 338.28, HAP 0/4/12 ns 4, RAM: 22624/38544; st 32; 0.1: st:0 in_st:-1 inm:-1 ininv:0
341501589 shelly_output.cpp:58    Output 1: off -> on (HAP)
341506999 shelly_output.cpp:58    Output 99: off -> on (HAP)
346139428 mgos_sys_config.c:232   Loading conf2.json
346276466 mgos_sys_config.c:174   Saved to conf9.json
346289664 shelly_main.cpp:554     Up 346.28, HAP 0/4/12 ns 4, RAM: 22624/38544; st 32; 0.1: st:1 in_st:-1 inm:-1 ininv:0
354121537 shelly_main.cpp:554     Up 354.11, HAP 0/4/12 ns 4, RAM: 22940/38544; st 32; 0.1: st:1 in_st:-1 inm:-1 ininv:0
356728184 shelly_output.cpp:58    Output 1: on -> off (HAP)
356733608 shelly_output.cpp:58    Output 99: on -> off (HAP)
362138972 mgos_sys_config.c:232   Loading conf2.json
362275956 mgos_sys_config.c:174   Saved to conf9.json
362289152 shelly_main.cpp:554     Up 362.28, HAP 0/4/12 ns 4, RAM: 22624/38544; st 32; 0.1: st:0 in_st:-1 inm:-1 ininv:0
370121711 shelly_main.cpp:554     Up 370.11, HAP 0/4/12 ns 4, RAM: 22940/38544; st 32; 0.1: st:0 in_st:-1 inm:-1 ininv:0
378121549 shelly_main.cpp:554     Up 378.11, HAP 0/4/12 ns 4, RAM: 22940/38544; st 32; 0.1: st:0 in_st:-1 inm:-1 ininv:0
386121673 shelly_main.cpp:554     Up 386.11, HAP 0/4/12 ns 4, RAM: 22940/38544; st 32; 0.1: st:0 in_st:-1 inm:-1 ininv:0
394121620 shelly_main.cpp:554     Up 394.11, HAP 0/4/12 ns 4, RAM: 22940/38544; st 32; 0.1: st:0 in_st:-1 inm:-1 ininv:0
402121602 shelly_main.cpp:554     Up 402.11, HAP 0/4/12 ns 4, RAM: 22940/38544; st 32; 0.1: st:0 in_st:-1 inm:-1 ininv:0
410121649 shelly_main.cpp:554     Up 410.11, HAP 0/4/12 ns 4, RAM: 22940/38544; st 32; 0.1: st:0 in_st:-1 inm:-1 ininv:0
418121720 shelly_main.cpp:554     Up 418.11, HAP 0/4/12 ns 4, RAM: 22940/38544; st 32; 0.1: st:0 in_st:-1 inm:-1 ininv:0
426121663 shelly_main.cpp:554     Up 426.11, HAP 0/4/12 ns 4, RAM: 22940/38544; st 32; 0.1: st:0 in_st:-1 inm:-1 ininv:0
434121534 shelly_main.cpp:554     Up 434.11, HAP 0/4/12 ns 4, RAM: 22940/38544; st 32; 0.1: st:0 in_st:-1 inm:-1 ininv:0
442121586 shelly_main.cpp:554     Up 442.11, HAP 0/4/12 ns 4, RAM: 22940/38544; st 32; 0.1: st:0 in_st:-1 inm:-1 ininv:0
450121673 shelly_main.cpp:554     Up 450.11, HAP 0/4/12 ns 4, RAM: 22940/38544; st 32; 0.1: st:0 in_st:-1 inm:-1 ininv:0
458121826 shelly_main.cpp:554     Up 458.11, HAP 0/4/12 ns 4, RAM: 22940/38544; st 32; 0.1: st:0 in_st:-1 inm:-1 ininv:0
466121659 shelly_main.cpp:554     Up 466.11, HAP 0/4/12 ns 4, RAM: 22940/38544; st 32; 0.1: st:0 in_st:-1 inm:-1 ininv:0
[... cut ...]
561908339 HAPPlatformTCPStrea:121 0x3fff57d4 192.168.111.157:54144 Incoming HAP connection, ns 0/3/12
561918698 HAPPlatformTCPStrea:278 0x3fff57d4 192.168.111.157:54144 Accepted HAP connection, ns 0/4/12 ts 0x3fff6cc4
563001693 shelly_main.cpp:554     Up 562.99, HAP 0/4/12 ns 4, RAM: 19912/38544; st 32; 0.1: st:1 in_st:-1 inm:-1 ininv:0
570121860 shelly_main.cpp:554     Up 570.11, HAP 0/4/12 ns 4, RAM: 22940/38544; st 32; 0.1: st:1 in_st:-1 inm:-1 ininv:0
578121641 shelly_main.cpp:554     Up 578.11, HAP 0/4/12 ns 4, RAM: 22940/38544; st 32; 0.1: st:1 in_st:-1 inm:-1 ininv:0

rojer, 2021-03-14:
pushed beta4, please update. it's not expected to fix the problem but i added some debug output that may help.
please update and post logs

iperniaf, 2021-03-14:
I have updated to beta4 one of my 15 Shelly 1
For me, It’s working 
47610090 shelly_debug.cpp:227    No log file, sending new entries
48309639 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.193:61564
49035395 shelly_main.cpp:554     Up 49.02, HAP 0/6/12 ns 6, RAM: 21016/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0
49309087 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.193:61564
50033024 mongoose.c:12067        Failed to resolve 'time.google.com', server 8.8.8.8
50311724 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.193:61564
51315574 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.193:61564
57035682 shelly_main.cpp:554     Up 57.03, HAP 0/6/12 ns 6, RAM: 23252/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0
65035904 shelly_main.cpp:554     Up 65.03, HAP 0/6/12 ns 6, RAM: 23252/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0
68186718 mgos_sntp.c:98          SNTP query to time.google.com
73035495 shelly_main.cpp:554     Up 73.02, HAP 0/6/12 ns 6, RAM: 21788/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0
78033468 mongoose.c:12067        Failed to resolve 'time.google.com', server 8.8.8.8
81035447 shelly_main.cpp:554     Up 81.02, HAP 0/6/12 ns 6, RAM: 23252/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0
89035644 shelly_main.cpp:554     Up 89.03, HAP 0/6/12 ns 6, RAM: 23252/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0
92215701 HAPPlatformTCPStrea:211 0x3fff68cc Recv 128
92220446 HAPPlatformTCPStrea:386 0x3fff68cc Read 128
92225120 HAPPlatformTCPStrea:211 0x3fff68cc Recv 66
92229321 HAPPlatformTCPStrea:386 0x3fff68cc Read 66
92238631 HAPPlatformTCPStrea:408 0x3fff68cc Write 45
92425136 HAPPlatformTCPStrea:211 0x3fff68cc Recv 96
92429525 HAPPlatformTCPStrea:386 0x3fff68cc Read 96
92439543 HAPPlatformTCPStrea:408 0x3fff68cc Write 128
92444443 HAPPlatformTCPStrea:408 0x3fff68cc Write 16
95123536 HAPPlatformTCPStrea:211 0x3fff68cc Recv 128
95128244 HAPPlatformTCPStrea:386 0x3fff68cc Read 128
95133740 HAPPlatformTCPStrea:211 0x3fff68cc Recv 66
95138040 HAPPlatformTCPStrea:386 0x3fff68cc Read 66
95145869 shelly_output.cpp:58    Output 1: off -> on (HAP)
95153188 HAPPlatformTCPStrea:408 0x3fff68cc Write 45
95171832 HAPPlatformTCPStrea:408 0x3fff65b4 Write 128
95176990 HAPPlatformTCPStrea:408 0x3fff65b4 Write 17
95182236 HAPPlatformTCPStrea:408 0x3fff5c14 Write 128
95186888 HAPPlatformTCPStrea:408 0x3fff5c14 Write 17
95502733 mgos_sntp.c:98          SNTP query to time.google.com
97048474 mgos_sys_config.c:232   Loading conf2.json
97184535 mgos_sys_config.c:174   Saved to conf9.json
97197240 shelly_main.cpp:554     Up 97.19, HAP 0/6/12 ns 6, RAM: 21448/38568; st 0; 0.1: st:1 in_st:0 inm:2 ininv:0
104521493 HAPPlatformTCPStrea:211 0x3fff65b4 Recv 128
104526237 HAPPlatformTCPStrea:386 0x3fff65b4 Read 128
104530945 HAPPlatformTCPStrea:211 0x3fff65b4 Recv 66
104535135 HAPPlatformTCPStrea:386 0x3fff65b4 Read 66
104544409 HAPPlatformTCPStrea:408 0x3fff65b4 Write 45
105037073 shelly_main.cpp:554     Up 105.03, HAP 0/6/12 ns 6, RAM: 21608/38568; st 0; 0.1: st:1 in_st:0 inm:2 ininv:0
105046679 mongoose.c:12067        Failed to resolve 'time.google.com', server 8.8.8.8
113035564 shelly_main.cpp:554     Up 113.02, HAP 0/6/12 ns 6, RAM: 23224/38568; st 0; 0.1: st:1 in_st:0 inm:2 ininv:0
116454929 HAPPlatformTCPStrea:211 0x3fff68cc Recv 96
116459568 HAPPlatformTCPStrea:386 0x3fff68cc Read 96
116470075 HAPPlatformTCPStrea:408 0x3fff68cc Write 128
116474946 HAPPlatformTCPStrea:408 0x3fff68cc Write 16
117240449 HAPPlatformTCPStrea:211 0x3fff68cc Recv 128
117245243 HAPPlatformTCPStrea:386 0x3fff68cc Read 128
117249913 HAPPlatformTCPStrea:211 0x3fff68cc Recv 66
117254108 HAPPlatformTCPStrea:386 0x3fff68cc Read 66
117261950 shelly_output.cpp:58    Output 1: on -> off (HAP)
117269325 HAPPlatformTCPStrea:408 0x3fff68cc Write 45
117283957 HAPPlatformTCPStrea:408 0x3fff65b4 Write 128
117289086 HAPPlatformTCPStrea:408 0x3fff65b4 Write 17
117294293 HAPPlatformTCPStrea:408 0x3fff5c14 Write 128
117298942 HAPPlatformTCPStrea:408 0x3fff5c14 Write 17
121053570 mgos_sys_config.c:232   Loading conf2.json
121189161 mgos_sys_config.c:174   Saved to conf9.json
121201938 shelly_main.cpp:554     Up 121.19, HAP 0/6/12 ns 6, RAM: 22908/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0
126362524 mgos_sntp.c:98          SNTP query to time.google.com
129036340 shelly_main.cpp:554     Up 129.03, HAP 0/6/12 ns 6, RAM: 21760/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0
136033789 mongoose.c:12067        Failed to resolve 'time.google.com', server 8.8.8.8
137035785 shelly_main.cpp:554     Up 137.03, HAP 0/6/12 ns 6, RAM: 23224/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0
145035607 shelly_main.cpp:554     Up 145.03, HAP 0/6/12 ns 6, RAM: 23224/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0
153035871 shelly_main.cpp:554     Up 153.03, HAP 0/6/12 ns 6, RAM: 23224/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0
158825700 mgos_sntp.c:98          SNTP query to time.google.com
161035774 shelly_main.cpp:554     Up 161.03, HAP 0/6/12 ns 6, RAM: 21760/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0

tojorocha, 2021-03-14:
Where can i get beta4?

rojer, 2021-03-14:
@tojorocha if running earlier beta, just use the button in the ui. or download from http://rojer.me/files/shelly/2.8.2-beta4/

tojorocha, 2021-03-14:
Unstable. Some “no response”
255913302 shelly_debug.cpp:227    No log file, sending new entries
256814316 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.193:49516
257611363 shelly_main.cpp:554     Up 257.60, HAP 0/1/12 ns 1, RAM: 24816/38568; st 0; 0.1: st:0 in_st:0 inm:0 ininv:0
257810574 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.193:49516
258815587 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.193:49516
259809940 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.193:49516
260818672 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.193:49516
263073653 HAPPlatformTCPStrea:121 0x3fff59fc 192.168.2.106:60206 Incoming HAP connection, ns 0/1/12
263083627 HAPPlatformTCPStrea:278 0x3fff59fc 192.168.2.106:60206 Accepted HAP connection, ns 0/2/12 ts 0x3fff5efc
263088246 HAPPlatformTCPStrea:211 0x3fff59fc Recv 128
263092515 HAPPlatformTCPStrea:386 0x3fff59fc Read 128
263097638 HAPPlatformTCPStrea:211 0x3fff59fc Recv 35
263101911 HAPPlatformTCPStrea:386 0x3fff59fc Read 35
264102411 HAPPlatformTCPStrea:408 0x3fff59fc Write 128
264107375 HAPPlatformTCPStrea:408 0x3fff59fc Write 92
264396332 HAPPlatformTCPStrea:211 0x3fff59fc Recv 128
264401089 HAPPlatformTCPStrea:386 0x3fff59fc Read 128
264406389 HAPPlatformTCPStrea:211 0x3fff59fc Recv 124
264410724 HAPPlatformTCPStrea:386 0x3fff59fc Read 124
264576546 HAPPlatformTCPStrea:408 0x3fff59fc Write 81
264670891 HAPPlatformTCPStrea:211 0x3fff59fc Recv 83
264675480 HAPPlatformTCPStrea:386 0x3fff59fc Read 83
264684094 HAPPlatformTCPStrea:408 0x3fff59fc Write 128
264689092 HAPPlatformTCPStrea:408 0x3fff59fc Write 61
264706992 HAPPlatformTCPStrea:408 0x3fff59fc Write 128
264718241 HAPPlatformTCPStrea:408 0x3fff59fc Write 16
264733857 HAPPlatformTCPStrea:408 0x3fff59fc Write 128
264745175 HAPPlatformTCPStrea:408 0x3fff59fc Write 16
264759305 HAPPlatformTCPStrea:408 0x3fff59fc Write 128
264770460 HAPPlatformTCPStrea:408 0x3fff59fc Write 17
264789812 HAPPlatformTCPStrea:408 0x3fff59fc Write 128
264805840 HAPPlatformTCPStrea:408 0x3fff59fc Write 16
264821853 HAPPlatformTCPStrea:408 0x3fff59fc Write 128
264835802 HAPPlatformTCPStrea:408 0x3fff59fc Write 18
264853461 HAPPlatformTCPStrea:408 0x3fff59fc Write 128
264866669 HAPPlatformTCPStrea:408 0x3fff59fc Write 14
264884516 HAPPlatformTCPStrea:408 0x3fff59fc Write 128
264921548 HAPPlatformTCPStrea:408 0x3fff59fc Write 14
264942984 HAPPlatformTCPStrea:408 0x3fff59fc Write 128
264959029 HAPPlatformTCPStrea:408 0x3fff59fc Write 17
264980753 HAPPlatformTCPStrea:408 0x3fff59fc Write 128
265001573 HAPPlatformTCPStrea:408 0x3fff59fc Write 14
265018520 HAPPlatformTCPStrea:408 0x3fff59fc Write 128
265611334 shelly_main.cpp:554     Up 265.60, HAP 0/2/12 ns 2, RAM: 24916/38568; st 0; 0.1: st:0 in_st:0 inm:0 ininv:0
266293330 HAPPlatformTCPStrea:121 0x3fff5bc4 192.168.2.47:58436 Incoming HAP connection, ns 0/2/12
266303208 HAPPlatformTCPStrea:278 0x3fff5bc4 192.168.2.47:58436 Accepted HAP connection, ns 0/3/12 ts 0x3fff54b4
266310514 HAPPlatformTCPStrea:211 0x3fff5bc4 Recv 128
266315103 HAPPlatformTCPStrea:386 0x3fff5bc4 Read 128
266320285 HAPPlatformTCPStrea:211 0x3fff5bc4 Recv 35
266324532 HAPPlatformTCPStrea:386 0x3fff5bc4 Read 35
267355244 HAPPlatformTCPStrea:408 0x3fff5bc4 Write 128
267360293 HAPPlatformTCPStrea:408 0x3fff5bc4 Write 92
267409642 HAPPlatformTCPStrea:105 0x3fff5bc4 192.168.2.47:58436 HAP connection closed, ns 0/3/12 i 2
267417552 HAPPlatformTCPStrea:305 HAPPlatformTCPStreamClose ts 0x3fff54b4 nc 0x0 0/2/12
273611230 shelly_main.cpp:554     Up 273.60, HAP 0/2/12 ns 2, RAM: 24916/38568; st 0; 0.1: st:0 in_st:0 inm:0 ininv:0
281610882 shelly_main.cpp:554     Up 281.60, HAP 0/2/12 ns 2, RAM: 24916/38568; st 0; 0.1: st:0 in_st:0 inm:0 ininv:0
287260912 HAPPlatformTCPStrea:121 0x3fff5bc4 192.168.2.193:49518 Incoming HAP connection, ns 0/2/12
287270903 HAPPlatformTCPStrea:278 0x3fff5bc4 192.168.2.193:49518 Accepted HAP connection, ns 0/3/12 ts 0x3fff6174
287275527 HAPPlatformTCPStrea:211 0x3fff5bc4 Recv 128
287279778 HAPPlatformTCPStrea:386 0x3fff5bc4 Read 128
287284907 HAPPlatformTCPStrea:211 0x3fff5bc4 Recv 35
287289180 HAPPlatformTCPStrea:386 0x3fff5bc4 Read 35
288301765 HAPPlatformTCPStrea:408 0x3fff5bc4 Write 128
288306746 HAPPlatformTCPStrea:408 0x3fff5bc4 Write 92
288358701 HAPPlatformTCPStrea:211 0x3fff5bc4 Recv 128
288363473 HAPPlatformTCPStrea:386 0x3fff5bc4 Read 128
288368722 HAPPlatformTCPStrea:211 0x3fff5bc4 Recv 124
288373030 HAPPlatformTCPStrea:386 0x3fff5bc4 Read 124
288538812 HAPPlatformTCPStrea:408 0x3fff5bc4 Write 81
[... 88 lines cut ...]
382889920 HAPPlatformTCPStrea:211 0x3fff5c4c Recv 35
382894209 HAPPlatformTCPStrea:386 0x3fff5c4c Read 35
383922486 HAPPlatformTCPStrea:408 0x3fff5c4c Write 128
383927464 HAPPlatformTCPStrea:408 0x3fff5c4c Write 92
383963080 HAPPlatformTCPStrea:105 0x3fff5c4c 192.168.2.5:52567 HAP connection closed, ns 0/4/12 i 2
383971019 HAPPlatformTCPStrea:305 HAPPlatformTCPStreamClose ts 0x3fff549c nc 0x0 0/3/12
383984534 HAPPlatformTCPStrea:121 0x3fff53ac 192.168.2.13:59038 Incoming HAP connection, ns 0/3/12
384016026 HAPPlatformTCPStrea:121 0x3fff6404 192.168.2.5:52568 Incoming HAP connection, ns 1/3/12
384577742 HAPPlatformTCPStrea:278 0x3fff53ac 192.168.2.13:59038 Accepted HAP connection, ns 1/4/12 ts 0x3fff6514
384582585 HAPPlatformTCPStrea:386 0x3fff53ac Read 128
384587599 HAPPlatformTCPStrea:211 0x3fff53ac Recv 35
384591752 HAPPlatformTCPStrea:386 0x3fff53ac Read 35
[... cut ...]
386669275 shelly_main.cpp:554     Up 386.66, HAP 0/4/12 ns 4, RAM: 22124/38568; st 0; 0.1: st:0 in_st:0 inm:0 ininv:0
386674573 HAPPlatformTCPStrea:386 0x3fff6404 Read 128
386679765 HAPPlatformTCPStrea:211 0x3fff6404 Recv 35
386684025 HAPPlatformTCPStrea:386 0x3fff6404 Read 35
387724980 HAPPlatformTCPStrea:408 0x3fff6404 Write 128
387730026 HAPPlatformTCPStrea:408 0x3fff6404 Write 92
388102594 HAPPlatformTCPStrea:105 0x3fff6404 192.168.2.5:52568 HAP connection closed, ns 0/4/12 i 2
388110525 HAPPlatformTCPStrea:305 HAPPlatformTCPStreamClose ts 0x3fff549c nc 0x0 0/3/12

tojorocha, 2021-03-14:
Other Shelly

743037100 shelly_debug.cpp:227    No log file, sending new entries
743061507 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.193:49519
744067715 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.193:49519
744466594 shelly_main.cpp:554     Up 744.46, HAP 0/2/12 ns 2, RAM: 23980/38568; st 0; 0.1: st:0 in_st:0 inm:0 ininv:0
745067408 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.193:49519
746062487 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.193:49519
747137678 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.193:49519
747870977 HAPPlatformTCPStrea:121 0x3fff5a6c 192.168.2.123:54663 Incoming HAP connection, ns 0/2/12
747880712 HAPPlatformTCPStrea:278 0x3fff5a6c 192.168.2.123:54663 Accepted HAP connection, ns 0/3/12 ts 0x3fff59f4
747886598 HAPPlatformTCPStrea:211 0x3fff5a6c Recv 128
747890952 HAPPlatformTCPStrea:386 0x3fff5a6c Read 128
747895693 HAPPlatformTCPStrea:211 0x3fff5a6c Recv 35
747899803 HAPPlatformTCPStrea:386 0x3fff5a6c Read 35
748913662 HAPPlatformTCPStrea:408 0x3fff5a6c Write 128
748918374 HAPPlatformTCPStrea:408 0x3fff5a6c Write 92
749497132 HAPPlatformTCPStrea:105 0x3fff5a6c 192.168.2.123:54663 HAP connection closed, ns 0/3/12 i 2
749504752 HAPPlatformTCPStrea:305 HAPPlatformTCPStreamClose ts 0x3fff59f4 nc 0x0 0/2/12
749517326 HAPPlatformTCPStrea:121 0x3fff59f4 192.168.2.47:58440 Incoming HAP connection, ns 0/2/12
749563225 HAPPlatformTCPStrea:278 0x3fff59f4 192.168.2.47:58440 Accepted HAP connection, ns 0/3/12 ts 0x3fff6864
749568280 HAPPlatformTCPStrea:386 0x3fff59f4 Read 128
749572924 HAPPlatformTCPStrea:211 0x3fff59f4 Recv 35
749576989 HAPPlatformTCPStrea:386 0x3fff59f4 Read 35
750599435 HAPPlatformTCPStrea:408 0x3fff59f4 Write 128
750604124 HAPPlatformTCPStrea:408 0x3fff59f4 Write 92
750616844 HAPPlatformTCPStrea:105 0x3fff59f4 192.168.2.47:58440 HAP connection closed, ns 0/3/12 i 2
750624412 HAPPlatformTCPStrea:305 HAPPlatformTCPStreamClose ts 0x3fff6864 nc 0x0 0/2/12
752466848 shelly_main.cpp:554     Up 752.46, HAP 0/2/12 ns 2, RAM: 24564/38568; st 0; 0.1: st:0 in_st:0 inm:0 ininv:0
760466578 shelly_main.cpp:554     Up 760.46, HAP 0/2/12 ns 2, RAM: 24564/38568; st 0; 0.1: st:0 in_st:0 inm:0 ininv:0
768466649 shelly_main.cpp:554     Up 768.46, HAP 0/2/12 ns 2, RAM: 24564/38568; st 0; 0.1: st:0 in_st:0 inm:0 ininv:0
776466681 shelly_main.cpp:554     Up 776.46, HAP 0/2/12 ns 2, RAM: 24564/38568; st 0; 0.1: st:0 in_st:0 inm:0 ininv:0
784466708 shelly_main.cpp:554     Up 784.46, HAP 0/2/12 ns 2, RAM: 24564/38568; st 0; 0.1: st:0 in_st:0 inm:0 ininv:0
792466676 shelly_main.cpp:554     Up 792.46, HAP 0/2/12 ns 2, RAM: 24564/38568; st 0; 0.1: st:0 in_st:0 inm:0 ininv:0
799180319 HAPPlatformTCPStrea:121 0x3fff5a6c 192.168.2.123:54666 Incoming HAP connection, ns 0/2/12
799190055 HAPPlatformTCPStrea:278 0x3fff5a6c 192.168.2.123:54666 Accepted HAP connection, ns 0/3/12 ts 0x3fff59f4
799194457 HAPPlatformTCPStrea:211 0x3fff5a6c Recv 128
799198593 HAPPlatformTCPStrea:386 0x3fff5a6c Read 128
799203292 HAPPlatformTCPStrea:211 0x3fff5a6c Recv 35
799207424 HAPPlatformTCPStrea:386 0x3fff5a6c Read 35
800219235 HAPPlatformTCPStrea:408 0x3fff5a6c Write 128
800223854 HAPPlatformTCPStrea:408 0x3fff5a6c Write 92
800316418 HAPPlatformTCPStrea:105 0x3fff5a6c 192.168.2.123:54666 HAP connection closed, ns 0/3/12 i 2
800324037 HAPPlatformTCPStrea:305 HAPPlatformTCPStreamClose ts 0x3fff59f4 nc 0x0 0/2/12
800466957 shelly_main.cpp:554     Up 800.46, HAP 0/2/12 ns 2, RAM: 24564/38568; st 0; 0.1: st:0 in_st:0 inm:0 ininv:0
803588305 HAPPlatformTCPStrea:121 0x3fff5a6c 192.168.2.47:58444 Incoming HAP connection, ns 0/2/12
803597951 HAPPlatformTCPStrea:278 0x3fff5a6c 192.168.2.47:58444 Accepted HAP connection, ns 0/3/12 ts 0x3fff59f4
803604162 HAPPlatformTCPStrea:211 0x3fff5a6c Recv 128
803608527 HAPPlatformTCPStrea:386 0x3fff5a6c Read 128
803613234 HAPPlatformTCPStrea:211 0x3fff5a6c Recv 35
803617359 HAPPlatformTCPStrea:386 0x3fff5a6c Read 35
804624053 HAPPlatformTCPStrea:408 0x3fff5a6c Write 128
804628772 HAPPlatformTCPStrea:408 0x3fff5a6c Write 92
804678318 HAPPlatformTCPStrea:105 0x3fff5a6c 192.168.2.47:58444 HAP connection closed, ns 0/3/12 i 2
804685875 HAPPlatformTCPStrea:305 HAPPlatformTCPStreamClose ts 0x3fff59f4 nc 0x0 0/2/12
806776484 HAPPlatformTCPStrea:121 0x3fff5a6c 192.168.2.5:52570 Incoming HAP connection, ns 0/2/12
806786023 HAPPlatformTCPStrea:278 0x3fff5a6c 192.168.2.5:52570 Accepted HAP connection, ns 0/3/12 ts 0x3fff59f4
806791918 HAPPlatformTCPStrea:211 0x3fff5a6c Recv 128
806796273 HAPPlatformTCPStrea:386 0x3fff5a6c Read 128
806801012 HAPPlatformTCPStrea:211 0x3fff5a6c Recv 35
806805135 HAPPlatformTCPStrea:386 0x3fff5a6c Read 35
807813474 HAPPlatformTCPStrea:408 0x3fff5a6c Write 128
807818182 HAPPlatformTCPStrea:408 0x3fff5a6c Write 92
807861823 HAPPlatformTCPStrea:105 0x3fff5a6c 192.168.2.5:52570 HAP connection closed, ns 0/3/12 i 2
807869387 HAPPlatformTCPStrea:305 HAPPlatformTCPStreamClose ts 0x3fff59f4 nc 0x0 0/2/12
808466715 shelly_main.cpp:554     Up 808.46, HAP 0/2/12 ns 2, RAM: 24564/38568; st 0; 0.1: st:0 in_st:0 inm:0 ininv:0
816466594 shelly_main.cpp:554     Up 816.46, HAP 0/2/12 ns 2, RAM: 24564/38568; st 0; 0.1: st:0 in_st:0 inm:0 ininv:0
818814337 HAPPlatformTCPStrea:121 0x3fff5a6c 192.168.2.123:54667 Incoming HAP connection, ns 0/2/12
818824067 HAPPlatformTCPStrea:278 0x3fff5a6c 192.168.2.123:54667 Accepted HAP connection, ns 0/3/12 ts 0x3fff59f4
818828455 HAPPlatformTCPStrea:211 0x3fff5a6c Recv 128
[... cut ...]
964627717 shelly_main.cpp:554     Up 964.62, HAP 0/4/12 ns 4, RAM: 23688/38568; st 0; 0.1: st:0 in_st:0 inm:0 ininv:0
972627621 shelly_main.cpp:554     Up 972.62, HAP 0/4/12 ns 4, RAM: 23688/38568; st 0; 0.1: st:0 in_st:0 inm:0 ininv:0
980627417 shelly_main.cpp:554     Up 980.62, HAP 0/4/12 ns 4, RAM: 23688/38568; st 0; 0.1: st:0 in_st:0 inm:0 ininv:0
988627572 shelly_main.cpp:554     Up 988.62, HAP 0/4/12 ns 4, RAM: 23688/38568; st 0; 0.1: st:0 in_st:0 inm:0 ininv:0
996627606 shelly_main.cpp:554     Up 996.62, HAP 0/4/12 ns 4, RAM: 23688/38568; st 0; 0.1: st:0 in_st:0 inm:0 ininv:0

SteffenHuettner, 2021-03-14:
Wow what a great community. Just went back from a weekend seminar. I need to recover and than I try beta 4 as well.

SteffenHuettner, 2021-03-14:
Ok, Tryed Beta 4 no success:

25082915 shelly_debug.cpp:227    No log file, sending new entries
25697102 shelly_main.cpp:554     Up 25.69, HAP 0/0/12 ns 0, RAM: 25100/38568; st 0; 0.1: st:1 in_st:1 inm:1 ininv:0
25709212 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.30:64703
26441322 HAPPlatformTCPStrea:121 0x3fff5f54 192.168.2.30:64713 Incoming HAP connection, ns 0/0/12
26451215 HAPPlatformTCPStrea:278 0x3fff5f54 192.168.2.30:64713 Accepted HAP connection, ns 0/1/12 ts 0x3fff5cbc
26455900 HAPPlatformTCPStrea:211 0x3fff5f54 Recv 128
26460148 HAPPlatformTCPStrea:386 0x3fff5f54 Read 128
26465228 HAPPlatformTCPStrea:211 0x3fff5f54 Recv 35
26469502 HAPPlatformTCPStrea:386 0x3fff5f54 Read 35
27478833 HAPPlatformTCPStrea:408 0x3fff5f54 Write 128
27483821 HAPPlatformTCPStrea:408 0x3fff5f54 Write 92
27500184 mgos_mongoose.c:66      New heap free LWM: 19048
27528473 HAPPlatformTCPStrea:211 0x3fff5f54 Recv 128
27533113 HAPPlatformTCPStrea:386 0x3fff5f54 Read 128
27538373 HAPPlatformTCPStrea:211 0x3fff5f54 Recv 124
27542701 HAPPlatformTCPStrea:386 0x3fff5f54 Read 124
27708403 HAPPlatformTCPStrea:408 0x3fff5f54 Write 81
27725419 HAPPlatformTCPStrea:211 0x3fff5f54 Recv 83
27730015 HAPPlatformTCPStrea:386 0x3fff5f54 Read 83
27738610 HAPPlatformTCPStrea:408 0x3fff5f54 Write 128
27743598 HAPPlatformTCPStrea:408 0x3fff5f54 Write 61
27760014 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.30:64703
27775259 mgos_mongoose.c:66      New heap free LWM: 17560
27792149 HAPPlatformTCPStrea:408 0x3fff5f54 Write 128
27806271 HAPPlatformTCPStrea:408 0x3fff5f54 Write 16
27823258 HAPPlatformTCPStrea:408 0x3fff5f54 Write 128
27837384 HAPPlatformTCPStrea:408 0x3fff5f54 Write 16
27856815 HAPPlatformTCPStrea:408 0x3fff5f54 Write 128
27869568 HAPPlatformTCPStrea:408 0x3fff5f54 Write 17
27908143 HAPPlatformTCPStrea:408 0x3fff5f54 Write 128
27929907 HAPPlatformTCPStrea:408 0x3fff5f54 Write 16
27951915 HAPPlatformTCPStrea:408 0x3fff5f54 Write 128
27964152 HAPPlatformTCPStrea:408 0x3fff5f54 Write 18
27980912 HAPPlatformTCPStrea:408 0x3fff5f54 Write 128
27993504 HAPPlatformTCPStrea:408 0x3fff5f54 Write 14
28011131 HAPPlatformTCPStrea:408 0x3fff5f54 Write 128
28025821 HAPPlatformTCPStrea:408 0x3fff5f54 Write 14
28042517 HAPPlatformTCPStrea:408 0x3fff5f54 Write 128
28056272 HAPPlatformTCPStrea:408 0x3fff5f54 Write 17
28078169 HAPPlatformTCPStrea:408 0x3fff5f54 Write 128
28093333 HAPPlatformTCPStrea:408 0x3fff5f54 Write 14
28109927 HAPPlatformTCPStrea:408 0x3fff5f54 Write 128
28175556 HAPPlatformTCPStrea:211 0x3fff5f54 Recv 94
28180088 HAPPlatformTCPStrea:386 0x3fff5f54 Read 94
28189118 HAPPlatformTCPStrea:408 0x3fff5f54 Write 128
28194051 HAPPlatformTCPStrea:408 0x3fff5f54 Write 20
28211659 HAPPlatformTCPStrea:105 0x3fff5f54 192.168.2.30:64713 HAP connection closed, ns 0/1/12 i 2
28219531 HAPPlatformTCPStrea:305 HAPPlatformTCPStreamClose ts 0x3fff5cbc nc 0x0 0/0/12
29027545 HAPPlatformTCPStrea:121 0x3fff5cbc 192.168.2.189:65311 Incoming HAP connection, ns 0/0/12
29037510 HAPPlatformTCPStrea:278 0x3fff5cbc 192.168.2.189:65311 Accepted HAP connection, ns 0/1/12 ts 0x3fff563c
29042195 HAPPlatformTCPStrea:211 0x3fff5cbc Recv 128
29046450 HAPPlatformTCPStrea:386 0x3fff5cbc Read 128
29051522 HAPPlatformTCPStrea:211 0x3fff5cbc Recv 35
29055852 HAPPlatformTCPStrea:386 0x3fff5cbc Read 35
30045701 HAPPlatformTCPStrea:408 0x3fff5cbc Write 128
30050651 HAPPlatformTCPStrea:408 0x3fff5cbc Write 92
30089330 HAPPlatformTCPStrea:105 0x3fff5cbc 192.168.2.189:65311 HAP connection closed, ns 0/1/12 i 2
30097229 HAPPlatformTCPStrea:305 HAPPlatformTCPStreamClose ts 0x3fff563c nc 0x0 0/0/12
31828697 HAPPlatformTCPStrea:121 0x3fff5cbc 192.168.2.9:64082 Incoming HAP connection, ns 0/0/12
31838458 HAPPlatformTCPStrea:278 0x3fff5cbc 192.168.2.9:64082 Accepted HAP connection, ns 0/1/12 ts 0x3fff563c
31847098 HAPPlatformTCPStrea:211 0x3fff5cbc Recv 128
31851746 HAPPlatformTCPStrea:386 0x3fff5cbc Read 128
31856889 HAPPlatformTCPStrea:211 0x3fff5cbc Recv 35
31861160 HAPPlatformTCPStrea:386 0x3fff5cbc Read 35
32855373 HAPPlatformTCPStrea:408 0x3fff5cbc Write 128
32860412 HAPPlatformTCPStrea:408 0x3fff5cbc Write 92
32880993 HAPPlatformTCPStrea:105 0x3fff5cbc 192.168.2.9:64082 HAP connection closed, ns 0/1/12 i 2
32888859 HAPPlatformTCPStrea:305 HAPPlatformTCPStreamClose ts 0x3fff563c nc 0x0 0/0/12
[... 275 lines cut ...]
74744564 HAPPlatformTCPStrea:408 0x3fff5cbc Write 128
74749527 HAPPlatformTCPStrea:408 0x3fff5cbc Write 61
74763463 HAPPlatformTCPStrea:408 0x3fff5cbc Write 128
74774513 HAPPlatformTCPStrea:408 0x3fff5cbc Write 16
74789461 HAPPlatformTCPStrea:408 0x3fff5cbc Write 128
74800541 HAPPlatformTCPStrea:408 0x3fff5cbc Write 16
74817268 HAPPlatformTCPStrea:408 0x3fff5cbc Write 128
74830710 HAPPlatformTCPStrea:408 0x3fff5cbc Write 17
74845326 HAPPlatformTCPStrea:408 0x3fff5cbc Write 128
74858197 HAPPlatformTCPStrea:408 0x3fff5cbc Write 16
74874972 HAPPlatformTCPStrea:408 0x3fff5cbc Write 128
74909417 HAPPlatformTCPStrea:408 0x3fff5cbc Write 18
74926404 HAPPlatformTCPStrea:408 0x3fff5cbc Write 128
74938126 HAPPlatformTCPStrea:408 0x3fff5cbc Write 14
74954146 HAPPlatformTCPStrea:408 0x3fff5cbc Write 128
74967041 HAPPlatformTCPStrea:408 0x3fff5cbc Write 14
74983311 HAPPlatformTCPStrea:408 0x3fff5cbc Write 128
74998397 HAPPlatformTCPStrea:408 0x3fff5cbc Write 17
75013692 HAPPlatformTCPStrea:408 0x3fff5cbc Write 128
75040857 HAPPlatformTCPStrea:408 0x3fff5cbc Write 14
75056440 HAPPlatformTCPStrea:408 0x3fff5cbc Write 128
75094199 HAPPlatformTCPStrea:105 0x3fff5cbc 192.168.2.9:64123 HAP connection closed, ns 0/1/12 i 2
75102007 HAPPlatformTCPStrea:305 HAPPlatformTCPStreamClose ts 0x3fff3474 nc 0x0 0/0/12
76786900 shelly_main.cpp:554     Up 76.78, HAP 0/0/12 ns 0, RAM: 25884/38568; st 0; 0.1: st:1 in_st:1 inm:1 ininv:0
76972550 HAPPlatformTCPStrea:121 0x

SteffenHuettner, 2021-03-14:
Just for fun, one of my Shelly 1 is working with 1.8.1 it the garage door opener.

Dag0d, 2021-03-14:
for me it feels like beta4 is even worse, I can't even get a real log, whole device is unresponsive when checking the log. if I close the log window it seems to work better

Dag0d, 2021-03-14:
a third shelly one was able to give me a log file. in the beginning it didn't worked at all when controlling in the home app it just showed no response. Then out of the sudden it turned on and off without any issues, after turning it on the second time, I wasn't able to turn it off for approx. 7 tries, but then it turned off.
here the log file which captured this event

418391769 shelly_debug.cpp:227    No log file, sending new entries
419004885 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.131:56857
420012271 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.131:56857
421006126 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.131:56857
421466895 shelly_main.cpp:554     Up 421.46, HAP 0/5/12 ns 5, RAM: 21896/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0
422007161 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.131:56857
423008381 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.131:56857
424008265 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.131:56857
427210148 HAPPlatformTCPStrea:211 0x3fff5604 Recv 128
427214926 HAPPlatformTCPStrea:386 0x3fff5604 Read 128
427219669 HAPPlatformTCPStrea:211 0x3fff5604 Recv 66
427223883 HAPPlatformTCPStrea:386 0x3fff5604 Read 66
427231769 shelly_output.cpp:58    Output 1: off -> on (HAP)
427239262 HAPPlatformTCPStrea:408 0x3fff5604 Write 45
427255283 HAPPlatformTCPStrea:408 0x3fff6e84 Write 128
427260322 HAPPlatformTCPStrea:408 0x3fff6e84 Write 17
427264987 HAPPlatformTCPStrea:408 0x3fff74a4 Write 128
427269538 HAPPlatformTCPStrea:408 0x3fff74a4 Write 17
427274246 HAPPlatformTCPStrea:408 0x3fff6cbc Write 128
427278784 HAPPlatformTCPStrea:408 0x3fff6cbc Write 17
427283980 mgos_mongoose.c:66      New heap free LWM: 13824
429479450 mgos_sys_config.c:232   Loading conf2.json
429499369 mgos_sys_config.c:232   Loading conf3.json
429505567 mgos_config_util.c:110  Extra key: [relays]
429510242 mgos_config_util.c:110  Extra key: [relays.r0]
429515941 mgos_config_util.c:110  Extra key: [relays.r0.default_state]
429521548 mgos_config_util.c:110  Extra key: [relays.r0.btn_reverse]
429525956 mgos_config_util.c:110  Extra key: [relays.r0]
429530091 mgos_config_util.c:110  Extra key: [relays]
429650906 mgos_sys_config.c:174   Saved to conf9.json
429663612 shelly_main.cpp:554     Up 429.65, HAP 0/5/12 ns 5, RAM: 22004/38568; st 0; 0.1: st:1 in_st:0 inm:2 ininv:0
432072069 HAPPlatformTCPStrea:211 0x3fff5604 Recv 128
432076881 HAPPlatformTCPStrea:386 0x3fff5604 Read 128
432081644 HAPPlatformTCPStrea:211 0x3fff5604 Recv 66
432085859 HAPPlatformTCPStrea:386 0x3fff5604 Read 66
432093816 shelly_output.cpp:58    Output 1: on -> off (HAP)
432101260 HAPPlatformTCPStrea:408 0x3fff5604 Write 45
432119210 HAPPlatformTCPStrea:408 0x3fff6e84 Write 128
432124374 HAPPlatformTCPStrea:408 0x3fff6e84 Write 17
432129000 HAPPlatformTCPStrea:408 0x3fff74a4 Write 128
432133564 HAPPlatformTCPStrea:408 0x3fff74a4 Write 17
432138257 HAPPlatformTCPStrea:408 0x3fff6cbc Write 128
432142818 HAPPlatformTCPStrea:408 0x3fff6cbc Write 17
435916411 HAPPlatformTCPStrea:211 0x3fff5604 Recv 128
435921181 HAPPlatformTCPStrea:386 0x3fff5604 Read 128
435925920 HAPPlatformTCPStrea:211 0x3fff5604 Recv 66
435930127 HAPPlatformTCPStrea:386 0x3fff5604 Read 66
435937986 shelly_output.cpp:58    Output 1: off -> on (HAP)
435945465 HAPPlatformTCPStrea:408 0x3fff5604 Write 45
435961484 HAPPlatformTCPStrea:408 0x3fff6e84 Write 128
435966516 HAPPlatformTCPStrea:408 0x3fff6e84 Write 17
435971186 HAPPlatformTCPStrea:408 0x3fff74a4 Write 128
435975732 HAPPlatformTCPStrea:408 0x3fff74a4 Write 17
435980445 HAPPlatformTCPStrea:408 0x3fff6cbc Write 128
435984979 HAPPlatformTCPStrea:408 0x3fff6cbc Write 17
435995947 mgos_mongoose.c:66      New heap free LWM: 11688
437484994 mgos_sys_config.c:232   Loading conf2.json
437504699 mgos_sys_config.c:232   Loading conf3.json
437511092 mgos_config_util.c:110  Extra key: [relays]
437515638 mgos_config_util.c:110  Extra key: [relays.r0]
437521329 mgos_config_util.c:110  Extra key: [relays.r0.default_state]
437526991 mgos_config_util.c:110  Extra key: [relays.r0.btn_reverse]
437531395 mgos_config_util.c:110  Extra key: [relays.r0]
437535530 mgos_config_util.c:110  Extra key: [relays]
437656236 mgos_sys_config.c:174   Saved to conf9.json
437668949 shelly_main.cpp:554     Up 437.66, HAP 0/5/12 ns 5, RAM: 18340/38568; st 0; 0.1: st:1 in_st:0 inm:2 ininv:0
447908159 HAPPlatformTCPStrea:105 0x3fff616c 192.168.1.127:51514 HAP connection closed, ns 1/6/12 i 2
[... 6 lines cut ...]
449105435 HAPPlatformTCPStrea:408 0x3fff62fc Write 92
449171791 HAPPlatformTCPStrea:105 0x3fff62fc 192.168.1.125:62537 HAP connection closed, ns 0/6/12 i 2
449179653 HAPPlatformTCPStrea:305 HAPPlatformTCPStreamClose ts 0x3fff571c nc 0x0 0/5/12
451942271 HAPPlatformTCPStrea:211 0x3fff6e84 Recv 128
451947070 HAPPlatformTCPStrea:386 0x3fff6e84 Read 128
451951829 HAPPlatformTCPStrea:211 0x3fff6e84 Recv 66
451956036 HAPPlatformTCPStrea:386 0x3fff6e84 Read 66
451963961 shelly_output.cpp:58    Output 1: on -> off (HAP)
451971443 HAPPlatformTCPStrea:408 0x3fff6e84 Write 45
451987681 HAPPlatformTCPStrea:408 0x3fff5604 Write 128
451992730 HAPPlatformTCPStrea:408 0x3fff5604 Write 17
451997947 HAPPlatformTCPStrea:408 0x3fff74a4 Write 128
452002612 HAPPlatformTCPStrea:408 0x3fff74a4 Write 17
452007360 HAPPlatformTCPStrea:408 0x3fff6cbc Write 128
452011898 HAPPlatformTCPStrea:408 0x3fff6cbc Write 17
454617710 mgos_sys_config.c:232   Loading conf2.json
[... cut ...]
454643806 mgos_config_util.c:110  Extra key: [relays]
454648515 mgos_config_util.c:110  Extra key: [relays.r0]
454654213 mgos_config_util.c:110  Extra key: [relays.r0.default_state]
454659818 mgos_config_util.c:110  Extra key: [relays.r0.btn_reverse]
454664226 mgos_config_util.c:110  Extra key: [relays.r0]
454668361 mgos_config_util.c:110  Extra key: [relays]
454789106 mgos_sys_config.c:174   Saved to conf9.json
454801828 shelly_main.cpp:554     Up 454.79, HAP 0/5/12 ns 5, RAM: 18800/38568; st 0; 0.1: st:0 in_st:0 inm:2 ininv:0

partytrik, 2021-03-14:
As said before, my Shelly 1 all still worked, but unstable. Now put beta 4 only on one, seems to be quite stable now. Here is the log:

158883781 shelly_debug.cpp:227    No log file, sending new entries
164450159 shelly_main.cpp:554     Up 164.44, HAP 0/9/12 ns 9, RAM: 20888/38568; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
167921570 HAPPlatformTCPStrea:211 0x3fff7624 Recv 110
167926366 HAPPlatformTCPStrea:386 0x3fff7624 Read 110
167937078 HAPPlatformTCPStrea:408 0x3fff7624 Write 128
167942001 HAPPlatformTCPStrea:408 0x3fff7624 Write 16
169455746 HAPPlatformTCPStrea:211 0x3fff7624 Recv 128
169460598 HAPPlatformTCPStrea:386 0x3fff7624 Read 128
169465305 HAPPlatformTCPStrea:211 0x3fff7624 Recv 80
169469568 HAPPlatformTCPStrea:386 0x3fff7624 Read 80
169477468 shelly_output.cpp:58    Output 1: off -> on (HAP)
169485141 HAPPlatformTCPStrea:408 0x3fff7624 Write 45
169504745 HAPPlatformTCPStrea:408 0x3fff60b4 Write 128
169509850 HAPPlatformTCPStrea:408 0x3fff60b4 Write 17
169514509 HAPPlatformTCPStrea:408 0x3fff67f4 Write 128
169519098 HAPPlatformTCPStrea:408 0x3fff67f4 Write 17
169523790 HAPPlatformTCPStrea:408 0x3fff738c Write 128
169528372 HAPPlatformTCPStrea:408 0x3fff738c Write 17
169533055 HAPPlatformTCPStrea:408 0x3fff5c9c Write 128
169537636 HAPPlatformTCPStrea:408 0x3fff5c9c Write 17
172467736 mgos_sys_config.c:232   Loading conf2.json
172487753 mgos_sys_config.c:232   Loading conf3.json
172612737 mgos_sys_config.c:174   Saved to conf9.json
172625622 shelly_main.cpp:554     Up 172.62, HAP 0/9/12 ns 9, RAM: 20516/38568; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
172636098 HAPPlatformTCPStrea:211 0x3fff7624 Recv 128
172640711 HAPPlatformTCPStrea:386 0x3fff7624 Read 128
172645394 HAPPlatformTCPStrea:211 0x3fff7624 Recv 80
172649631 HAPPlatformTCPStrea:386 0x3fff7624 Read 80
172657592 shelly_output.cpp:58    Output 1: on -> off (HAP)
172666554 HAPPlatformTCPStrea:408 0x3fff7624 Write 45
172685499 HAPPlatformTCPStrea:408 0x3fff60b4 Write 128
172690551 HAPPlatformTCPStrea:408 0x3fff60b4 Write 17
172695222 HAPPlatformTCPStrea:408 0x3fff67f4 Write 128
172699791 HAPPlatformTCPStrea:408 0x3fff67f4 Write 17
172705084 HAPPlatformTCPStrea:408 0x3fff738c Write 128
172709762 HAPPlatformTCPStrea:408 0x3fff738c Write 17
172714504 HAPPlatformTCPStrea:408 0x3fff5c9c Write 128
172719069 HAPPlatformTCPStrea:408 0x3fff5c9c Write 17
180467631 mgos_sys_config.c:232   Loading conf2.json
180487637 mgos_sys_config.c:232   Loading conf3.json
180612633 mgos_sys_config.c:174   Saved to conf9.json
180625521 shelly_main.cpp:554     Up 180.62, HAP 0/9/12 ns 9, RAM: 20552/38568; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
188449722 shelly_main.cpp:554     Up 188.44, HAP 0/9/12 ns 9, RAM: 20888/38568; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
196449634 shelly_main.cpp:554     Up 196.44, HAP 0/9/12 ns 9, RAM: 20888/38568; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
204449416 shelly_main.cpp:554     Up 204.44, HAP 0/9/12 ns 9, RAM: 20888/38568; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
212449702 shelly_main.cpp:554     Up 212.44, HAP 0/9/12 ns 9, RAM: 20888/38568; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
220449623 shelly_main.cpp:554     Up 220.44, HAP 0/9/12 ns 9, RAM: 20888/38568; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
228449573 shelly_main.cpp:554     Up 228.44, HAP 0/9/12 ns 9, RAM: 20888/38568; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
236449623 shelly_main.cpp:554     Up 236.44, HAP 0/9/12 ns 9, RAM: 20888/38568; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
238837805 HAPPlatformTCPStrea:211 0x3fff7624 Recv 128
238842655 HAPPlatformTCPStrea:386 0x3fff7624 Read 128
238847391 HAPPlatformTCPStrea:211 0x3fff7624 Recv 81
238851658 HAPPlatformTCPStrea:386 0x3fff7624 Read 81
238861210 HAPPlatformTCPStrea:408 0x3fff7624 Write 45
244449797 shelly_main.cpp:554     Up 244.44, HAP 0/9/12 ns 9, RAM: 20924/38568; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
250806837 HAPPlatformTCPStrea:211 0x3fff7624 Recv 128
250811668 HAPPlatformTCPStrea:386 0x3fff7624 Read 128
250816430 HAPPlatformTCPStrea:211 0x3fff7624 Recv 80
250820665 HAPPlatformTCPStrea:386 0x3fff7624 Read 80
250830214 HAPPlatformTCPStrea:408 0x3fff7624 Write 45
251791537 HAPPlatformTCPStrea:211 0x3fff7624 Recv 110
251796291 HAPPlatformTCPStrea:386 0x3fff7624 Read 110
251806416 HAPPlatformTCPStrea:408 0x3fff7624 Write 128
251811430 HAPPlatformTCPStrea:408 0x3fff7624 Write 16
252450065 shelly_main.cpp:554     Up 252.44, HAP 0/9/12 ns 9, RAM: 20888/38568; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
257947423 HAPPlatformTCPStrea:211 0x3fff7624 Recv 110
257952199 HAPPlatformTCPStrea:386 0x3fff7624 Read 110
257962849 HAPPlatformTCPStrea:408 0x3fff7624 Write 128
[... 24 lines cut ...]
262301520 HAPPlatformTCPStrea:408 0x3fff67f4 Write 128
262306109 HAPPlatformTCPStrea:408 0x3fff67f4 Write 17
262310815 HAPPlatformTCPStrea:408 0x3fff738c Write 128
262315393 HAPPlatformTCPStrea:408 0x3fff738c Write 17
262320073 HAPPlatformTCPStrea:408 0x3fff5c9c Write 128
262324648 HAPPlatformTCPStrea:408 0x3fff5c9c Write 17
264707511 HAPPlatformTCPStrea:211 0x3fff7624 Recv 110
264712283 HAPPlatformTCPStrea:386 0x3fff7624 Read 110
264722947 HAPPlatformTCPStrea:408 0x3fff7624 Write 128
264727885 HAPPlatformTCPStrea:408 0x3fff7624 Write 16
266199845 HAPPlatformTCPStrea:211 0x3fff7624 Recv 110
266204627 HAPPlatformTCPStrea:386 0x3fff7624 Read 110
266215274 HAPPlatformTCPStrea:408 0x3fff7624 Write 128
266220224 HAPPlatformTCPStrea:408 0x3fff7624 Write 16
268297573 HAPPlatformTCPStrea:211 0x3fff738c Recv 128
268302375 HAPPlatformTCPStrea:386 0x3fff738c Read 128
[... cut ...]
268467757 mgos_sys_config.c:232   Loading conf2.json
268487823 mgos_sys_config.c:232   Loading conf3.json
268612835 mgos_sys_config.c:174   Saved to conf9.json
268625752 shelly_main.cpp:554     Up 268.62, HAP 0/9/12 ns 9, RAM: 20548/38568; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
276449717 shelly_main.cpp:554     Up 276.44, HAP 0/9/12 ns 9, RAM: 20888/38568; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
284449642 shelly_main.cpp:554     Up 284.44, HAP 0/9/12 ns 9, RAM: 20888/38568; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0

rojer, 2021-03-15:
ok, gentlemen, thank you for your patience, i believe i have the bug pinned for real. working on a fix now.

rojer, 2021-03-15:
please update to beta5 (http://rojer.me/files/shelly/2.8.2-beta5/) , it should solve the problem.

Dag0d, 2021-03-15:
looks like this has done the trick! devices are responsive again. will continue testing

rojer, 2021-03-15:
alright, pushed 2.8.2

SteffenHuettner, 2021-03-15:
Solved for me, many thx.

partytrik, 2021-03-16:
Dear @rojer, after two days on 2.8.2. with 25 Shelly 1, we still have very bad homekit connection problems. That was not the case with 2.7.3. Whenever you open a room in the app all shellys start to "update" and sometimes all shellys in a room are unreachable while those in other rooms are normal. Very strange. Here is the log of one of them. At some point the log didn't continue writing. Also when the device restarts working it is shown in the Home App with a circled ! and the status text (on/off) in red. Also, would it be possible to allow more than 12 connections (is that referring to every home app device? we have 15 iPhones, iPads, Macbooks in the house plus 3 home Pods and 1 apple tv).

Best patrik

110849898445 shelly_debug.cpp:227    No log file, sending new entries
110849903442 E:M 1572 (197 blocks)
110849908302 E:M 1572 (197 blocks)
110849911953 E:M 1572 (197 blocks)
110849915457 E:M 424 (54 blocks)
)
110849919161 E:M 459 (58 blocks)
)
110849922814 E:M 494 (63 blocks)
)
110849926474 110849930189 110850149276 E:M 555 (70 blocks)
)
110850154603 E:M 590 (75 blocks)
)
110850159300 110850163098 110850910048 E:M 1460 (183 blocks)
110851188866 E:M 600 (76 blocks)
110851366615 E:M 202 (26 blocks)
110851438846 110851543602 110851641802 HAPPlatformTCPStrea:276 0x3fff581c 192.168.1.40:50138 Accepted HAP connection, ns 0/11/12 ts 0x3fff7d34
110852974402 E:M 2080 (261 blocks)
110852979849 E:M 1572 (197 blocks)
110852984596 E:M 1572 (197 blocks)
110852988247 E:M 1572 (197 blocks)
110852991588 E:M 1572 (197 blocks)
110852994817 E:M 1572 (197 blocks)
110852997871 E:M 1572 (197 blocks)
110853000922 E:M 1572 (197 blocks)
110853003976 E:M 443 (56 blocks)
)
110853007352 110853145053 E:M 491 (62 blocks)
)
110853148991 E:M 526 (67 blocks)
)
110853152697 E:M 561 (71 blocks)
)
110853156403 E:M 596 (75 blocks)
)
110853177849 HAPPlatformTCPStrea:105 0x3fff87e4 192.168.1.4:53653 HAP connection closed, ns 0/11/12 i 2
110853185858 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fffbe94 nc 0x0 0/10/12
110853189129 E:M 1572 (197 blocks)
110853198626 E:M 1572 (197 blocks)
110853203521 E:M 1572 (197 blocks)
110853206911 E:M 1572 (197 blocks)
110853210343 E:M 1572 (197 blocks)
110853213541 E:M 1572 (197 blocks)
110853216614 E:M 1572 (197 blocks)
110853219686 E:M 1572 (197 blocks)
110853224447 E:M 1572 (197 blocks)
110853227664 E:M 1572 (197 blocks)
110853230731 E:M 1572 (197 blocks)
110855372439 HAPPlatformTCPStrea:105 0x3fff581c 192.168.1.40:50138 HAP connection closed, ns 0/10/12 i 2
110855380494 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff7d34 nc 0x0 0/9/12
110857373300 HAPPlatformTCPStrea:105 0x3fff80d4 192.168.1.19:56404 HAP connection closed, ns 0/9/12 i 2
110857381326 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff723c nc 0x0 0/8/12
110857390700 HAPPlatformTCPStrea:105 0x3fff804c 192.168.1.8:54191 HAP connection closed, ns 0/8/12 i 2
110857398305 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff6f3c nc 0x0 0/7/12
110857993173 shelly_main.cpp:560     Up 110857.98, HAP 0/7/12 ns 7, RAM: 12160/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
110859219138 HAPPlatformTCPStrea:121 0x3fff581c 192.168.1.8:54200 Incoming HAP connection, ns 0/7/12
110859229069 HAPPlatformTCPStrea:276 0x3fff581c 192.168.1.8:54200 Accepted HAP connection, ns 0/8/12 ts 0x3fff77cc
110860721462 HAPPlatformTCPStrea:105 0x3fff949c 192.168.1.94:50963 HAP connection closed, ns 0/8/12 i 2
110860729492 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff34cc nc 0x0 0/7/12
110862813210 HAPPlatformTCPStrea:105 0x3fff581c 192.168.1.8:54200 HAP connection closed, ns 0/7/12 i 2
110862821257 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff77cc nc 0x0 0/6/12
110862913061 HAPPlatformTCPStrea:121 0x3fff5fa4 192.168.1.70:57776 Incoming HAP connection, ns 0/6/12
110862923033 HAPPlatformTCPStrea:276 0x3fff5fa4 192.168.1.70:57776 Accepted HAP connection, ns 0/7/12 ts 0x3fff957c
110866465613 HAPPlatformTCPStrea:121 0x3fff7a14 192.168.1.13:60830 Incoming HAP connection, ns 0/7/12
110866475369 HAPPlatformTCPStrea:276 0x3fff7a14 192.168.1.13:60830 Accepted HAP connection, ns 0/8/12 ts 0x3fff7114
[... 9 lines cut ...]
110891921319 HAPPlatformTCPStrea:276 0x3fffbe24 192.168.1.70:57779 Accepted HAP connection, ns 0/11/12 ts 0x3fff67a4
110893220489 shelly_main.cpp:560     Up 110893.21, HAP 0/11/12 ns 11, RAM: 5132/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
110895051413 HAPPlatformTCPStrea:105 0x3fff5fa4 192.168.1.70:57776 HAP connection closed, ns 0/11/12 i 2
110895059560 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff957c nc 0x0 0/10/12
110901220973 shelly_main.cpp:560     Up 110901.21, HAP 0/10/12 ns 10, RAM: 12560/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
110906771551 HAPPlatformTCPStrea:105 0x3fff7c6c 192.168.1.21:60957 HAP connection closed, ns 0/10/12 i 2
110906779583 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff677c nc 0x0 0/9/12
110909220885 shelly_main.cpp:560     Up 110909.21, HAP 0/9/12 ns 9, RAM: 9312/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
110909919008 HAPPlatformTCPStrea:121 0x3fff6e8c 192.168.1.4:53660 Incoming HAP connection, ns 0/9/12
110909928980 HAPPlatformTCPStrea:276 0x3fff6e8c 192.168.1.4:53660 Accepted HAP connection, ns 0/10/12 ts 0x3fff74f4
110911207960 HAPPlatformTCPStrea:121 0x3fff629c 192.168.1.94:50982 Incoming HAP connection, ns 0/10/12
[... cut ...]
110913201881 HAPPlatformTCPStrea:276 0x3fff87e4 192.168.1.95:54570 Accepted HAP connection, ns 0/12/12 ts 0x3fff77fc
110915882347 HAPPlatformTCPStrea:105 0x3fff629c 192.168.1.94:50982 HAP connection closed, ns 0/12/12 i 2
110915890247 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff677c nc 0x0 0/11/12
110915893418 E:M 1460 (183 blocks)
110915923267 E:M 1460 (183 blocks)
110915932423 E:M 1460 (183 blocks)
110915937591 E:M 1460 (183 blocks)
110915940725 E:M 1460 (183 blocks)
110915944089 E:M 650 (82 blocks)
110915947188 E:M 683 (86 blocks)
110915950411 E:M 716 (90 blocks)
110915953584

rojer, 2021-03-17:
@partytrik the log shows a device under memory pressure, unable to allocate memory (`E:M nnn` messages). there's very little free RAM, less than i would expect with 10 active connections... i see it's been up for a while, can you reboot it and start log earlier?

rojer, 2021-03-17:
as for allowing more connections - as you can see, we can barely handle 12.

partytrik, 2021-03-17:
> as for allowing more connections - as you can see, we can barely handle 12.

OK, I see. Maybe less does the trick then? Although in 2.7.3 with also already 12 I did not have these problems.

partytrik, 2021-03-17:
> @partytrik the log shows a device under memory pressure, unable to allocate memory (`E:M nnn` messages). there's very little free RAM (under 10K), less than i would expect with 10 active connections... i see it's been up for a while, can you reboot it and start log earlier?

Well, it's not even been a day, but here is another one. I keep closing and launching the home app on the iPhone and keep changing rooms to force a reload of the respective shelly and I keep pressing the respective shelly, but it doesn't show any reaction, neither in the app, nor in the log. In the app it just says "no response". It doesn't seem to connect, even though there are not even 12 connections active. Changing the status with the home app on the computer, also didn't force a connection. So actually it just shows as unresponsve without even trying to update the respective light in the home app. Maybe there is a problem in there? 

And!! what also happens is that at some point the shellys do show connected, but I caught this device yesterday showing an "off status" (without a ! in the corner or red text), when the light was actually turned on.

I do have this same problem with half of the shellys, sometimes. not always the same. Usually it's an entire room that is gone (oddly). And then when they do get responsive again, the status is shown in red (off/on) with the same ! in the upper right corner that you get with the "no response" message.

Here's the log:

234519398 shelly_debug.cpp:227    No log file, sending new entries
235271524 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.4:52153
236523356 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.4:52153
236992072 shelly_main.cpp:560     Up 236.98, HAP 0/12/12 ns 12, RAM: 19228/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
237271997 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.4:52153
238274401 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.4:52153
239273883 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.4:52153
244991665 shelly_main.cpp:560     Up 244.98, HAP 0/12/12 ns 12, RAM: 19836/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
252992210 shelly_main.cpp:560     Up 252.98, HAP 0/12/12 ns 12, RAM: 19836/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
260991806 shelly_main.cpp:560     Up 260.98, HAP 0/12/12 ns 12, RAM: 19836/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
268991609 shelly_main.cpp:560     Up 268.98, HAP 0/12/12 ns 12, RAM: 19836/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
276991548 shelly_main.cpp:560     Up 276.98, HAP 0/12/12 ns 12, RAM: 19836/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
284991481 shelly_main.cpp:560     Up 284.98, HAP 0/12/12 ns 12, RAM: 19836/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
292991636 shelly_main.cpp:560     Up 292.98, HAP 0/12/12 ns 12, RAM: 19836/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
300991540 shelly_main.cpp:560     Up 300.98, HAP 0/12/12 ns 12, RAM: 19668/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
308991572 shelly_main.cpp:560     Up 308.98, HAP 0/12/12 ns 12, RAM: 19668/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
316991465 shelly_main.cpp:560     Up 316.98, HAP 0/12/12 ns 12, RAM: 19668/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
324993170 shelly_main.cpp:560     Up 324.98, HAP 0/12/12 ns 12, RAM: 19668/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
332991603 shelly_main.cpp:560     Up 332.98, HAP 0/12/12 ns 12, RAM: 19668/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
340991918 shelly_main.cpp:560     Up 340.98, HAP 0/12/12 ns 12, RAM: 19668/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
348991522 shelly_main.cpp:560     Up 348.98, HAP 0/12/12 ns 12, RAM: 19668/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
356991499 shelly_main.cpp:560     Up 356.98, HAP 0/12/12 ns 12, RAM: 19668/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
360383752 HAPPlatformTCPStrea:105 0x3fff6d14 192.168.1.39:49585 HAP connection closed, ns 0/12/12 i 2
360391817 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff614c nc 0x0 0/11/12
364991673 shelly_main.cpp:560     Up 364.98, HAP 0/11/12 ns 11, RAM: 20068/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
372991525 shelly_main.cpp:560     Up 372.98, HAP 0/11/12 ns 11, RAM: 20068/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
380991630 shelly_main.cpp:560     Up 380.98, HAP 0/11/12 ns 11, RAM: 20068/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
388991989 shelly_main.cpp:560     Up 388.98, HAP 0/11/12 ns 11, RAM: 20068/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
396991662 shelly_main.cpp:560     Up 396.98, HAP 0/11/12 ns 11, RAM: 20068/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
404991628 shelly_main.cpp:560     Up 404.98, HAP 0/11/12 ns 11, RAM: 20068/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
412991792 shelly_main.cpp:560     Up 412.98, HAP 0/11/12 ns 11, RAM: 20068/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
420991839 shelly_main.cpp:560     Up 420.98, HAP 0/11/12 ns 11, RAM: 20068/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
428991892 shelly_main.cpp:560     Up 428.98, HAP 0/11/12 ns 11, RAM: 20068/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
436991716 shelly_main.cpp:560     Up 436.98, HAP 0/11/12 ns 11, RAM: 20068/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
444991987 shelly_main.cpp:560     Up 444.98, HAP 0/11/12 ns 11, RAM: 20068/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
452991665 shelly_main.cpp:560     Up 452.98, HAP 0/11/12 ns 11, RAM: 20068/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
[... cut ...]
812991755 shelly_main.cpp:560     Up 812.98, HAP 0/11/12 ns 11, RAM: 20300/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
820991600 shelly_main.cpp:560     Up 820.98, HAP 0/11/12 ns 11, RAM: 20300/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
828991621 shelly_main.cpp:560     Up 828.98, HAP 0/11/12 ns 11, RAM: 20300/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
836991681 shelly_main.cpp:560     Up 836.98, HAP 0/11/12 ns 11, RAM: 20300/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0
844991652 shelly_main.cpp:560     Up 844.98, HAP 0/11/12 ns 11, RAM: 20300/38552; st 0; 0.1: st:1 in_st:0 inm:1 ininv:0

seb74600, 2021-03-18:
https://github.com/mongoose-os-apps/shelly-homekit/issues/561

chstoll, 2021-03-18:
> Dear @rojer, after two days on 2.8.2. with 25 Shelly 1, we still have very bad homekit connection problems. That was not the case with 2.7.3. Whenever you open a room in the app all shellys start to "update" and sometimes all shellys in a room are unreachable while those in other rooms are normal. 

Thats really strange. I had this issue with 2.7.3 (devices allways show update, had to switch rooms to show the status). With 2.8.2 for the first time all shellys are responsive and show their state directly by opening the room. This is the best version so far for me. I have 47 shellys (39 with rojers firmware - Shelly Plug S, Shelly 1, Shelly 2.5, the others are shelly dimmers with orignal firmware which work via homebridge). Actually I'm on iOS 14.5 beta 4 (Apple TV has tvOS 14.5 beta also).

partytrik, 2021-03-18:
> Thats really strange. I had this issue with 2.7.3 (devices allways show update, had to switch rooms to show the status). With 2.8.2 for the first time all shellys are responsive and show their state [...]

Wenn my experience with 2.7.3 was that it happens sometimes and then in all the rooms and you could force devices to update by changing rooms or closing the app. Now, for me it’s occasionally ok, and most often one of the rooms doesn’t connect the shellies, while those in other rooms work. And then it’s partly not possible for 3min to connect to them, not forcing an “update”, then they do work, but show text in red and once I witnessed that one showed ok and as off, while the connected light was actually on.

partytrik, 2021-03-22:
@rojer, any idea regarding the above? What’s really new now is that sometimes the status of a Shelly is not displayed / updated correctly in the home app. So according to the app a light seems to be off, when in reality it’s on. I did never have that with any of the versions since 2.0

rojer, 2021-03-22:
i need some more time to think about this and reproduce. there indeed have been changes between 2.7.3 and 2.8 that may have changed the behavior under extreme network pressure, need to think about it.

partytrik, 2021-03-23:
> i need some more time to think about this and reproduce. there indeed have been changes between 2.7.3 and 2.8 that may have changed the behavior under extreme network pressure, need to think about it.

OK, my brother and friends left now, so only but still 8 iPhones/iPads/macs + 4 home pods + 1 Apple TV left in the house. I will monitor it a little more, if something improves., before going back to 2.7.3.

Thank you very much

Dag0d, 2021-03-23:
After 8 days of testing I can tell you from my side, that it works without any flaws in a high density and busy network with many Apple devices. Status is updated quick, response time is fast and no network connections are lost so far from any of my shellies.

[4 more replies]

closed 2021-05-21 by github-actions[bot]