mongoose-os-apps/shelly-homekit #571: Shelly1 "garage door opener" - chatter of close sensor

ingski, 2021-03-16:
I'm using a Shelly1 in garage door opener mode with the "Close Sensor" set to "Normally Closed". 
When I'm closing the garage door the close sensor reports twice, i.e. HomeKit receives a sequence of
- closed
- opened
- closed

instead of just a single "closed".
I assume that this is eventually caused by a mechanical issue with a chatter of the sensor itself.

It would be great if the firmware could be modified so that after a status change of the close sensor  has been discovered it would re-check 500ms (or better a configurable time) later and only send out the change notification if the status change still persists.

-
Ingo

Whiterabbit64, 2021-03-16:
Check the sensor’s magnet position. I had the same kind of issue and solved it by repositioning the magnet.

ingski, 2021-03-18:
The magnet is located directly opposite of the sensor.

joddye2, 2021-03-24:
We talk about this problem in the [448](https://github.com/mongoose-os-apps/shelly-homekit/issues/448) issue. But there is no solution yet

andyblac, 2021-03-24:
@joddye2 thx I'll close as duplicate.

closed 2021-03-24 by andyblac