mongoose-os-apps/shelly-homekit #572: Add button with state for detached mode

wisesimpson, 2021-03-17:
First thanks to all the contributors to this project. It's really useful. Here is a thought after I setup my Homekit with Shelly 1.

### Problem to solve
When using shelly switch with small bulb, an ideal scenario is that the bulb is always powered, the wall switch can still control it. The current solution is:
1. Set shelly to detached mode.
2. There will be a new stateless button (let's call it button2) shown up in Homekit.
3. Set Homekit automation for clicking button2. Because it's a stateless button, to know it should turn on or off the light you must check the status of the bulb first.

This implementation has led to a noticeable delay. I think having two roundtrips to the bulb is one major reason. 

### How
To minimize the delay, my proposal is to add an option of button with state for detached mode. If we have a button with state (let's call it button3), we can implement this way:
1. Set shelly to detached mode.
2. Set the new button (button3) with state instead of stateless.
3. The physical switch acts like the "edge" mode, every click changes the state of button3.
4. Set an automation "when button3 is on turn on the light"
5. Set an automation "when button3 is off turn off the light"
6. Set an automation "when the light is off set button3 status to off"

This way we just have one roundtrip to the bulb which should cut the delay significantly.

timoschilling, 2021-03-17:
Prove me wrong, but I expect that this will not improve the performance. The reason is that it looks like the event which the shelly emits to homekit contains only the characteristic, but not the value. That results in a read request of the shelly on characteristic, right after the change event is send to homekit. That means in the end you have the same amount of requests in both cases (stateless / stateful).

Maybe I'm wrong, but thats what the logs look like if you place a log statement in the read characteristic functions.

And your setup will have two trade offs:
1. if for some reasons `when the light is off set button3 status to off` will not run, your state will come out of sync.
2. You can end up in a endless loop with that, thats not possible with stateless actions.

wisesimpson, 2021-03-17:
Hi @timoschilling
Thanks for your reply. I totally agree with the trade offs you mentioned. About the "characteristic" it's beyond my knowledge. In your opinion is there a way to solve the delay issue?

jsiegenthaler, 2021-03-17:
You do not need a stateless switch

You need the following
1. Shelly switch in normal stateful mode: when ON, turn the bulb ON. When OFF, turn the bulb OFF
2. An automation to set the switch state: When bulb is ON, check state of switch, if switch is OFF, set switch ON. And vice versa.

This way the switch follows the state of the bulb, if the bulb is ever activated by any means other that the switch.
You need the IF condition to prevent causing a race condition (although it will be a slow race).

The automations themselves are the slow part: I do things similar in my house: my hallway light is controlled by Zigbee direct to a Hue bulb, and also by a Shelly1 controlling 3 x dumb LED lamps.

An automation ties them together.
A Philips Hue motion sensor controls the lights.

Zigbee is always very fast. If I walk into the hall, the Zigbee bulb comes on immediately. And then the automation fires, and anything from 0 to 2 seconds after the Shelly1 turns on and the other lamps come on. This is for me a very useful automation response time indicator.

Some days automation is faster, some days slower. I use an Apple TV 4K as a hub, and it is connected via Ethernet to the same network as the Hue bridge.

People say Three is much much faster and more reliable, but I still say the slowest part is the automation

wisesimpson, 2021-03-17:
Hi @jsiegenthaler 
Did you mean I don't need the detached mode? Then if the physical switch is turned off I can't control the smart bulb by other means.

I need the bulb to always be powered.

timoschilling, 2021-03-17:
@wisesimpson I think what @jsiegenthaler means is don’t wire the smart bulb to the Shelly, wire only the switch to the Shelly and let the Shelly output unused and give the smart bulb permanent power. 

What kind of smart bulb is it? Maybe using a dimmer and a not smart dimmable led bulb is the better option for you.

jsiegenthaler, 2021-03-17:
@wisesimpson Wire the bulb to be always powered. Use the physical switch to control the SW input of the Shelly. Use automations / webcalls etc to control the light, and set the SW state. Ignore the Shelly1 relay output, you don't need it.

wisesimpson, 2021-03-17:
I'm using different brands including Nanoleaf Essential, Philips Hue and Ikea TRÅDFRI.

Wiring bulbs with permanent power sounds risky to me. Still thanks for the suggestion.

I'm curious about how to improve the delay when using detached mode, or it's unavoidable?

timoschilling, 2021-03-17:
The concept behind a smart device at all is that it is permanently powered and turned on / off by network protocol (IP, HomeKit, Bluetooth, ZigBee, ...). I think you're “holding it wrong”, if you turn on the device by the shelly you will always have a delay, because of your device need to be booted and connected first or even if it's on by default there will be a delay.

If you don't want to have a smart bulb 100% on, replace your Shelly 1 with a Dimmer2 and your smart bulb with a dimmable LED. 

You need to make a decision between that both options, otherwise you always have delay and/or trouble.

andyblac, 2021-03-17:
@timo is correct, smart bulbs are for those areas where you can not replace the switch with a smart one, and designed to powered all the time, also cutting the power to a hk bulb will also result in the informus "No Responce" in homekit.

wisesimpson, 2021-03-17:
I agree a smart bulb should always be powered. I just prefer using "detached mode" over wiring the power permanently. 

If that's "holding it wrong", may I know what is the right use of "detached mode"?

jsiegenthaler, 2021-03-17:
Detached mode means the SW input and the Relay output on the Shelly1 operate independently

andyblac, 2021-03-17:
sounds like this issue https://github.com/mongoose-os-apps/shelly-homekit/issues/194, anyway, i'll close this one as it not a shelly issue. IMO, if you think i'm incorrect re-open and ask rojer advice.

closed 2021-03-17 by andyblac