mongoose-os-apps/shelly-homekit #574: Serial Connection / FT232 chip

louis49, 2021-03-18:
Hi all,
As indicated in the [wiki](https://github.com/mongoose-os-apps/shelly-homekit/wiki/Serial-connection) I acquired a USB to serial converter with an FT232 chip ([B08DXRKFZR](https://www.amazon.fr/gp/product/B08DXRKFZR)) but it appears that this one indicates connections that I do not understand: TXD+/TXD-/RXD+/RXD-/GND
I am a beginner in electronics and home automation and despite my research I could not understand how to do the conversion with the connections indicated in the wiki: RX / TX / 3V3 / GND / GPIO0
Can anyone tell me what connections to make or tell me where / what to look for?
Once I find this information I can update the wiki

timoschilling, 2021-03-18:
I'm unsure how to wire the txd +/- things. But I would recommend to use a different flasher! Your flasher didn't provide a 3.3V output, that means you need a different power source for your shelly during the flash. Using a separate power source together with your current flasher can kill your shelly or your PC.

You should luck for flashers like this: https://www.amazon.fr/Solution-Connector-Supports-Android-3-3V-5V/dp/B01I720HTG/ref=sr_1_1?__mk_fr_FR=ÅMÅŽÕÑ&dchild=1&keywords=FT232&qid=1616072521&sr=8-1
This is not a recommendation, I would only recommend flashers which I have tested by my own, it's just an example.

louis49, 2021-03-19:
Thank you very much.
Before your answer I ordered this one : https://www.amazon.fr/gp/product/B01N9RZK6I

louis49, 2021-03-19:
It works with https://www.amazon.fr/gp/product/B01N9RZK6I

closed 2021-03-19 by louis49