mongoose-os-apps/shelly-homekit #575: Connection issue

paolodesalvatore, 2021-03-18:
I experience continuous disconnections with the Shelly 1, and the only way to reconnect it is to disconnect and reconnect the power.  The Wi-Fi connection is good, and with the original firmware I don't have this problem.

MattAydin, 2021-04-17:
I have the same. Also to add, I have a multiple story house. Any ideas why this happens?

rojer, 2021-04-18:
does any of you use UniFi access points?

trolli-ch, 2021-05-02:
Yes, i also have connectivity issues. And I have UI Access Points.
On which build is 2.9.0 based ? 1.10.3 ?

Thanks !

timoschilling, 2021-05-02:
> On which build is 2.9.0 based ? 1.10.3 ?

Shelly HomeKit uses the same base OS (MongooseOS) as the stock firmware, but is not based on stock.

trolli-ch, 2021-05-02:
I had real strange issues with my network (UI AP) and the shellies in my house. After moving to stock 10.3 it seems to be better now.

closed 2021-06-10 by github-actions[bot]