mongoose-os-apps/shelly-homekit #578: Shelly 1PM not working with Shelly Homekit

bmpalmeida, 2021-03-19:
Hello,

I have a deployment of +30 Shelly, including 1PM 2.5PM, and i migrate all of them to shelly HomeKit, but one of them, one 1PM, doesn't work with Shelly HomeKit.

I'm using version 2.8.2, and the device gets completely unresponsive, i have a momentary switch connected to it, and not even the switch work. it connect to the wifi from time to time, and it is on 8 seconds more a less, and disconnect.

Even after reboot not every time i can get it to work.

I already remove it from the wall, and try to put it in different locations but the behaviour is always the same.

I try to get in fail-safe mode, and it is also not working. If i manage to update to stock, it works reliable without any problem.

Any suggestions?

bmpalmeida, 2021-03-19:
[image]

I've manage to get a print of the log

timoschilling, 2021-03-19:
@rojer looks like a issue for you

ma-jkl, 2021-04-04:
Hi,
I have tried everything. The last thing I did was remove the home in HomeKit and added a new one. It solved all the connection problems.

andyblac, 2021-04-04:
> Hi,
> I have tried everything. The last thing I did was remove the home in HomeKit and added a new one. It solved all the connection problems.

thx for letting us know, i'll close this one, please feel free to reopen if needed.

closed 2021-04-04 by andyblac

bmpalmeida, 2021-04-07:
Hi, actually i already tried all that options and i still have the problem.

But the problem is a little bit trickier, even if i not connect it to wifi or HomeKit, from time to time i loose the device. Even the physical switch stop working, and i have to turn off the circuit breaker to reboot it.

Adittionaly i have other device that was working ok, but it started to have the same behaviour.

I have a Asus wifi network, with 4 routers, and they are not connected to the same router.

Could be a hardware problem? even that with the original firmware i don't have problems?