mongoose-os-apps/shelly-homekit #580: Configure Roller shutter manually (no powerconsumption)

rwi1993, 2021-03-19:
I have swapped my existing switch with the Shelly 2.5 an it works for every roller shutter except one that has been installed later. It has some complicated wiring hidden behind a wall that has recently been covered so I know it’s there but can’t access it. 

However the wiring seems to be correct because technically it’s working (moving up and down) but only for around a second in each direction. This is because the Shelly can’t read any power consumption and when trying to calibrate it just flicks between on and off. 

Is there any way to manually calibrate the up and down moving time? Or any other way to trick It so I can set up the time manually?

BlancoDeMallloca, 2021-03-21:
This seems to be quite the same like issue #404. May you want to add support to this issue

rwi1993, 2021-03-21:
Thank you 🙏 that seems to be the solution. Will try on Monday.

closed 2021-03-21 by rwi1993