mongoose-os-apps/shelly-homekit #582: Shutter shows closed in homekit, when shutter is open

bybymax, 2021-03-20:
Hi there !

I installed a shelly 2.5 and updated it to last homekit firmware. Everything seems to be fine except for the fact that the shutter in the home app appears closed whereas it is open.
What can I do to reverse that ?
Thanks in advance and thank you so much for all the work done .

Bye !

Bybymax

BlancoDeMallloca, 2021-03-21:
May the setting „Swap Outputs“ on the web interface will solve your problem.

bybymax, 2021-03-21:
Thx 😊 I’ll try that again but I did it and nothing changed… Maybe the setting did not get to homekit properly ;)
If you have other possibilities I can explore ;) 

Have a great day guys !

jadaka23, 2021-03-27:
You have to recalibrate to solve it, same issue here

pivale, 2021-03-29:
Where are you seeing the mismatched status? webui or home app?
I have seen that from time to time, but force closing home app and reopening usually does the trick for me

jadaka23, 2021-03-29:
Hey, 

thanks for your reply. The missmatched status is in Home App as well as in the web ui. Even in the web ui it Is not possible to open or close it. just a new calibration is the workaround that works for me and solves the issue

Mit freundlichem Gruß

Abbas Banaha
- - - - - - - - - - -
Abbas Banaha
Wilfried-Ismar-Str. 4
D-50171 Kerpen
Telefon: +49 (0)2237/5066896
Mobil:    +49 (0)174/8876375
E-Mail:   ***@***.***

bybymax, 2021-03-29:
Hi thx so much for your replies. The mismatch is on the web and on the home app. I'll try to calibrate again and keep you posted. Thx again ! It works, but in the reverse mode 😝

jadaka23, 2021-03-30:
Thanks a lot, Great work!

Mit freundlichem Gruß

Abbas Banaha
- - - - - - - - - - -
Abbas Banaha
Wilfried-Ismar-Str. 4
D-50171 Kerpen
Telefon: +49 (0)2237/5066896
Mobil:    +49 (0)174/8876375
E-Mail:   ***@***.***

pivale, 2021-04-24:
Hey @andyblac does it make sense to close?
Recalibration is a workaround it didn't solve the problem.

I commented here about the same issue
https://github.com/mongoose-os-apps/shelly-homekit/issues/374#issuecomment-824601873

Let's at least reference/link them #374

andyblac, 2021-04-24:
> Hey @andyblac does it make sense to close?
> Recalibration is a workaround it didn't solve the problem.
> 
> I commented here about the same issue
> [#374 (comment)](https://github.com/mongoose-os-apps/shelly-homekit/issues/374#issuecomment-824601873)
> 
> Let's at least reference/link them #374

i was asked to close by @timoschilling I'll reopen if you feel is still required.

pivale, 2021-04-24:
Maybe @timoschilling thought about it as duplicate? I'm ok if we consider it will be handled in another issue

andyblac, 2021-04-24:
@timoschilling can you coment ?

timoschilling, 2021-04-24:
Keep it open for now

bybymax, 2021-04-24:
Hi guys ! I was pretty sure I had left a comment but it seems not. I apologize for that. 
my problem was solved thanks to your help. 
i had to swap out going, calibrate and save the configuration. 
as far as i m concerned, every thing has been working fine ever since. I even got another Shelly, and it works like a charm. 
thank you all for your help. 

Bye. 

Cyrille

timoschilling, 2021-04-24:
As anybody here still this problem? Otherwise this should be closed

pivale, 2021-04-24:
It happened to me a few days ago with 1.8.2., but the reverse situation: shutter was closed but home.app and website stated open. I couldn't do anything and had to recalibrate. 

I left a comment and log file here 
https://github.com/mongoose-os-apps/shelly-homekit/issues/374#issuecomment-824601873

timoschilling, 2021-04-24:
Ok then I think we can close it if @pivale infos are in #374

closed 2021-04-24 by andyblac