mongoose-os-apps/shelly-homekit #584: External Dimmer Switch Control

robbinsmisc, 2021-03-21:
Here's the situation, my wife likes dimmers with a mechanical slider rather than a digital button (or an app).  Consequently, smart on/off with a manual dimmer is the goal.  I came across this project this weekend for a potential lighting solution, and I'm looking forward to tinkering with the Shelly 1 whenever it arrives this week.

I realize that the Shelly devices are intended either for a standard switch or button.  Having never seen or used a Shelly device, I would imagine that phase controlled dimming likely produces a noisy SW signal for the Shelly compared to a strictly closed line signal.  My question is can a dimmer trigger the GPIO pin for the SW input?  Or is it going to be too intermittent?  I'm not familiar with how often the adc samples or the circuit connected to the GPIO pin.

I don't mind coding my own solution if a smart on/off triggered by a dimmer switch is not currently working and figured I'd ask people much more familiar than myself.  At first glance, it appears the noisy_input_pin is meant for a button, but a similar strategy applied to a dimmer switch.

timoschilling, 2021-03-22:
I'm not 100% sure what you try to build, can you provide a small sketch of your planed setup?

robbinsmisc, 2021-03-22:
Never mind.  What I'm really after is a SPDT relay for the circuit, and I don't see a shelly solution for that at the moment.

closed 2021-03-22 by robbinsmisc