mongoose-os-apps/shelly-homekit #586: "Roaming" feature

mrmaximas, 2021-03-22:
How about "roaming" feature like in latest stock firmware? May be for backup APN too?
Thanks!

timoschilling, 2021-04-23:
That feature is already there, it's a function of the mongoose os it self, you can find the documentation [here](https://mongoose-os.com/docs/mongoose-os/api/net/wifi.md#configuration)

mrmaximas, 2021-04-23:
i don't this features in the web interface.

rojer, 2021-04-23:
we need a UI for multiple APs, yes.

timoschilling, 2021-04-23:
This feature from the stock firmware, is it from shelly or from mongoose?
[image]
If it's from mongoose, which config value is it?

rojer, 2021-04-23:
no, actually. this is specific to stock fw.

rojer, 2021-04-23:
it makes sense as a feature of the wifi lib.

timoschilling, 2021-04-23:
ok, then I think this issue is a double of #501

rojer, 2021-04-23:
no, it's a different one. #501 is about alternate configuration, this one is about seeking a better AP.

rojer, 2021-12-17:
AP roaming is implemented, roaming threshold is set at -75. i think -65 stock setting is too aggressive.

closed 2021-12-17 by rojer