mongoose-os-apps/shelly-homekit #589: regular expressions strings

juanfal, 2021-03-23:
Only a minor aesthetic, yet perhaps can be more

    hostcheck = re.search(r"\d{1,3}\.\d{1,3}\.\d{1,3}\.\d{1,3}", host)

shouldn't regexp string be prefixed by **r** so the backslashes are preserved?

Another:

     logger.trace(f"\n…

inserts a newline before the string, doesn't it?

andyblac, 2021-03-23:
> Only a minor aesthetic, yet perhaps can be more
> 
> ```
> hostcheck = re.search(r"\d{1,3}\.\d{1,3}\.\d{1,3}\.\d{1,3}", host)
> ```
> 
> shouldn't regexp string be prefixed by **r** so the backslashes are preserved?
> 
it does make any differance to result:
```
andrews-mbp-wifi:tools andrewblackburn$ ./flash-shelly.py 192.168.3.45
hostcheck: <re.Match object; span=(0, 12), match='192.168.3.45'>

Host: http://192.168.3.45
Device Name: Hall Spots
Model: ShellyDimmer2
Firmware: Stock 1.10.0 ➜ HomeKit Not available

Devices found: 1 Upgradeable: 0 Flashed: 0 Failed: 0
andrews-mbp-wifi:tools andrewblackburn$ ./flash-shelly.py 192.168.3.45
hostcheck: <re.Match object; span=(0, 12), match='192.168.3.45'>

Host: http://192.168.3.45
Device Name: Hall Spots
Model: ShellyDimmer2
Firmware: Stock 1.10.0 ➜ HomeKit Not available

Devices found: 1 Upgradeable: 0 Flashed: 0 Failed: 0
```

> Another:
> 
> ```
>  logger.trace(f"\n…
> ```
> 
> inserts a newline before the string, doesn't it?

yes, it would, but it then messes with line numbers and trace logging, it easier to read logs if new lines have own code line.

andyblac, 2021-03-23:
as `r` will only help, i'll add it to next update.

andyblac, 2021-03-23:
done, `r'xxx'` will be in next update, thx for sugestion.

andyblac, 2021-03-23:
https://github.com/mongoose-os-apps/shelly-homekit/pull/590

juanfal, 2021-03-23:
Thank you! I am only just trying to dig enough into the firmware (!) to make Shelly 1 read not a switch on/off but a led from an Air Conditioner wall control panel… uf!!

Thanks

andyblac, 2021-03-23:
np, keep em coming 😄

PR #590 (2021-04-18): [Flashscript] Version 2.7.0
- move run block into Main class, move config, device_scan, manual_hosts into own functions, better handling of webserver shutdown.
- add support for RGBW2 (both stock color modes).
- add support for ShellyMotion (list info only, must use manual IP's and use --timeout 240, as motion needs longer to wake and be detected)
- add -dn | --device-name filter to limit scan to include term.
- add '-f | --flash', '--app-version', '--timeout'
- add battery info for those devices that support it.
- python formatting and cleanup, more optimisations
- add regex literal to re.sea [... cut]
closed 2021-04-11 by andyblac