mongoose-os-apps/shelly-homekit #592: Shelly-1 won't update anymore ("Another update is in progress.")

g3ntleman, 2021-03-23:
I currently have two Shelly-1 installing behind light switches. The one giving me problems has line-of-sight with the Access-Point.

It's available in HomeKit and currently works (I had to switch off power today). Since I first installed shelly-homekit, I experienced both shellys becoming unresponsive for network requests. One Shelly has an off-timer and that always worked, even if the Shelly was not accessible via network. They shows up in the router connection status, though.
Usually, I need to power-cycle them for updates. The last one was 2.7.3 which I installed via the Update-Buttons on the Shelly-Web-Interfaces.
Now those show a later version being available (2.8.2 as of writing). I again hit "Update" after reading the release notes. Worked as expected for one shelly-1. The other one hung with the status "Uploading...".
After some minutes, I reloaded the web-interface. It looked like before:

Status: RSSI: -50 Host: shelly1-kueche.local IP: 192.168.178.28
Firmware: Version: 2.7.3 (build 20210124-130526/2.7.3-g883914a-master)

 Revert to stock: Uploading 847168 bytes... 
 Update failed to start

Upload of the firmware zip fails immediately with Firefox (Mac). It takes longer to fail using Safari.
Reloading the page /update revealed:

HTTP/1.1 409 OK
Server: Mongoose/6.18
Content-Type: text/plain
Connection: close

Another update is in progress.

Reboot does not help.

Anyone any idea?

Various snippets from the debug-log:

2072276307 mgos_ota_core.c:587     begin: waiting for user
2072288426 mgos_ota_core.c:587     begin: waiting for user
2072304521 mgos_ota_core.c:587     begin: waiting for user
2072319144 mgos_ota_core.c:587     begin: waiting for user
2072336893 HAPPlatformTCPStrea:121 0x3fff6554 192.168.178.62:52617 Incoming HAP connection, ns 0/1/12
2072351742 HAPPlatformTCPStrea:276 0x3fff6554 192.168.178.62:52617 Accepted HAP connection, ns 0/2/12 ts 0x3fff5e7c
2073731654 mgos_ota_core.c:587     begin: waiting for user
2073788983 mgos_ota_core.c:587     begin: waiting for user
2073800922 mgos_ota_core.c:587     begin: waiting for user
2073815229 HAPPlatformTCPStrea:121 0x3fff6e3c 192.168.178.23:61008 Incoming HAP connection, ns 0/2/12
2074003675 mgos_ota_core.c:587     begin: waiting for user
...
2074773494 mgos_ota_core.c:587     begin: waiting for user
2074794319 shelly_main.cpp:543     Up 2074.78, HAP 1/2/12 ns 2, RAM: 11864/37912; st 0; 0.1: st:1 in_st:0 inm:2 ininv:0
2074808789 mgos_ota_core.c:587     begin: waiting for user
2074825040 mgos_ota_core.c:587     begin: waiting for user
...
2075329322 mgos_ota_core.c:587     begin: waiting for user
2075345434 mgos_ota_core.c:587     begin: waiting for user
2075399280 HAPPlatformTCPStrea:276 0x3fff6e3c 192.168.178.23:61008 Accepted HAP connection, ns 0/3/12 ts 0x3fff6244
2076790221 mgos_ota_core.c:587     begin: waiting for user
2076804876 mgos_ota_core.c:587     begin: waiting for user
....
2077236072 mgos_ota_core.c:587     begin: waiting for user
2077252848 HAPPlatformTCPStrea:105 0x3fff6e3c 192.168.178.23:61008 HAP connection closed, ns 0/3/12 i 2
2077266292 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff6244 nc 0x0 0/2/12

Trying the update...

Once I saw:
2264670601 mgos_ota_core.c:206     Update already in progress

Then I rebooted and tried to revert...

2344878743 mgos_ota_core.c:587     begin: waiting for user
2344894721 mgos_ota_core.c:587     begin: waiting for user
2344912359 HAPPlatformTCPStrea:121 0x3fff7e9c 192.168.178.22:57670 Incoming HAP connection, ns 0/2/12
2344928151 HAPPlatformTCPStrea:276 0x3fff7e9c 192.168.178.22:57670 Accepted HAP connection, ns 0/3/12 ts 0x3fff5e7c
2346467536 mgos_ota_core.c:587     begin: waiting for user
2346488761 shelly_main.cpp:543     Up 2346.48, HAP 0/3/12 ns 3, RAM: 10836/37912; st 0; 0.1: st:1 in_st:0 inm:2 ininv:0
2346503176 mgos_ota_core.c:178     Update timed out
2346514018 mgos_ota_core.c:811     Update finished, result -5 (Update timed out)
2354489918 shelly_main.cpp:543     Up 2354.48, HAP 0/3/12 ns 3, RAM: 19636/37912; st 0; 0.1: st:1 in_st:0 inm:2 ininv:0
2359507135 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.178.46:64825
2360508878 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.178.46:64825
2362490213 shelly_main.cpp:543     Up 2362.48, HAP 0/3/12 ns 3, RAM: 19636/37912; st 0; 0.1: st:1 in_st:0 inm:2 ininv:0
2370488822 shelly_main.cpp:543     Up 2370.48, HAP 0/3/12 ns 3, RAM: 20304/37912; st 0; 0.1: st:1 in_st:0 inm:2 ininv:0
2372163283 HAPPlatformTCPStrea:121 0x3fff623c 192.168.178.62:52677 Incoming HAP connection, ns 0/3/12
2372179054 HAPPlatformTCPStrea:276 0x3fff623c 192.168.178.62:52677 Accepted HAP connection, ns 0/4/12 ts 0x3fff6084
2378489142 shelly_main.cpp:543     Up 2378.48, HAP 0/4/12 ns 4, RAM: 19628/37912; st 0; 0.1: st:1 in_st:0 inm:2 ininv:0
2381331964 HAPPlatformTCPStrea:105 0x3fff6fa4 192.168.178.23:61104 HAP connection closed, ns 0/4/12 i 2
2381344386 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff95a4 nc 0x0 0/3/12
2386490466 shelly_main.cpp:543     Up 2386.48, HAP 0/3/12 ns 3, RAM: 19648/37912; st 0; 0.1: st:1 in_st:0 inm:2 ininv:0
2390775442 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.178.46:65123
2391773046 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.178.46:65123
2392775646 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.178.46:65123
2393424497 mg_rpc.c:305            Sys.Reboot via WS_in 192.168.178.46:65123
2393433632 mgos_system.c:58        Rebooting in 500 ms
2393451683 shelly_main.cpp:365     HAP server state: 2
2393469369 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff5e7c nc 0x3fff7e9c 0/2/12
2393482547 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff6084 nc 0x3fff623c 0/1/12
2393496776 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff5094 nc 0x3fff7d04 0/0/12
2393516839 HAPPlatformTCPStrea:105 0x3fff8434 0.0.0.0:9001 HAP connection closed, ns 0/0/12 i 1
2393529338 shelly_main.cpp:365     HAP server state: 0

That update got stuck again (no error status).

andyblac, 2021-03-23:
thats a know issue in 2.7.3, HAP gets stuck and refuses to stop, so update can start. try rebooting device wait few mins for homekit connections to settle, then try doing firmware update again.

timoschilling, 2021-03-23:
`2344878743 mgos_ota_core.c:587 begin: waiting for user`
Is it possible that you have an running auto off timer? If yes disable auto off and retry it.

rojer, 2021-03-24:
> That update got stuck again (no error status).

can you show the log? as Andy said, there is a known issue in 2.7 (that we are trying to track down, it only happens after 20+ days online), but you should be able to update after a reboot (and 2.8 has a temporary workaround).

rojer, 2021-04-24:
this is fixed in 2.9, caused by http server getting stuck.

closed 2021-04-24 by rojer