mongoose-os-apps/shelly-homekit #594: Update Issue

Shadokay, 2021-03-26:
Hello,
I have a problem with my Shelly 1 v3.
I plugged it in and connected it to my WiFi.
Now I have tried to integrate it into my Homekit system and therefore wanted to update the version. Unfortunately, after entering "http://192.168.2.194/ota?url=https://rojer.me/files/shelly/shelly-homekit-Shelly1.zip" I only get {"status": "updating", " has_update ": false," new_version ":" 20210323-104500 / v1.10.1-gf276b51 "," old_version ":" 20210323-104500 / v1.10.1-gf276b51 "} as result. Apparently this is not the right version or at least not the Homekit compatible because I don't see the possibility to set it up. Hopefully you can help me with my problem.

MustafaYildirim3107, 2021-03-26:
I always setup my shelly, connect it to wifi, update my shelly and then go to the url.
After the message go to the ip adress you should see the new interface.

So after seeing this message:
`{"status": "updating", " has_update ": false," new_version ":" 20210323-104500 / v1.10.1-gf276b51 "," old_version ":" 20210323-104500 / v1.10.1-gf276b51 "}` 

Go to http://192.168.2.194

ScottLeMagicien, 2021-04-13:
None of my Shelly 1's will update past the 20191217... version.  Does anyone know why this would be, or have a tip on how to make it? work

I've tried with every browser I have, and their own iPhone app.  Nothing works :-( 

Thanks for any ideas!

andyblac, 2021-04-13:
what have your tried?, rebooting etc. please post log of device trying to do an update.

ScottLeMagicien, 2021-04-13:
No, straight out of the box from Shelly.  I tried DHCP & Static.  I can add it to my.Shelly.Cloud "Home" and it keeps telling me there's an update for it, but it never works...  Rebooted, powered down and back up, Factory Reset... Nothing helps.

I see a notices saying:  "Device is currently updating its firmware."  "This could take up to 5 minutes."  And "Loading" for a while, and then it's still the same firmware as before...

andyblac, 2021-04-13:
> No, I'm trying to update it from straight out of the box from Shelly. I've tried DHCP & Static. I can add it to my.Shelly.Cloud "Home" and it keeps telling me there's an update for it, but it never works... Rebooted, powered down and back up, Factory Reset... Nothing helps.
> 
> I see a notices saying: "Device is currently updating its firmware." "This could take up to 5 minutes." And "Loading" for a while, and then it's still the same firmware as before... I don't know how to show you a log from the shelly website/app. Sorry.

have your tried going to the device web user interface and update from there, you need to be on lastest stock firmware before you can load this firmware.

ScottLeMagicien, 2021-04-14:
Sorry, I'm not sure I know what you mean.  I cannot update the STOCK Shelly firmware on these devices, and thus I cannot install the HomeKit firmware.   I've gone to the device's IP address on our local network with every web browser I have and tried to update it using its web interface.  Nothing has worked.

cjekelund, 2021-04-15:
Hi guys, I have the same problem. I have tried with the WebGUI and the python-script, both fail. 

I have also looked at the script and it seems like the script only updates stock firmware until version 1.10.0 and the stock firmware is now 1.10.2...

cjekelund, 2021-04-15:
Device Name: None
Model: ShellyDimmer2
Firmware: Stock 1.10.2 ➜ HomeKit Not available

Devices found: 1 Upgradeable: 0 Flashed: 0 Failed: 0

andyblac, 2021-04-15:
> Device Name: None
> Model: ShellyDimmer2
> Firmware: Stock 1.10.2 ➜ HomeKit Not available
> 
> Devices found: 1 Upgradeable: 0 Flashed: 0 Failed: 0

dimmer2 is not supported in this firmware.

andyblac, 2021-04-15:
> I have also looked at the script and it seems like the script only updates stock firmware until version 1.10.0 and the stock firmware is now 1.10.2...

script supports all stock versions.

cjekelund, 2021-04-15:
> > Device Name: None
> > Model: ShellyDimmer2
> > Firmware: Stock 1.10.2 ➜ HomeKit Not available
> > Devices found: 1 Upgradeable: 0 Flashed: 0 Failed: 0
> 
> dimmer2 is not supported in this firmware.

Thank you for prompt replies!

Do you know when dimmer2 will be supported?

Br

/cj

ScottLeMagicien, 2021-04-15:
Ok, I contacted Shelly and I was finally able to update the stock firmware.

To do it I had to set up a Personal Hotspot with my iPhone.

Then I factory reset the device and then connected to its WiFi signal on my computer.  

I went to its Web UI (192.168.33.1) in Safari and then under Internet & Security, WIFI MODE - CLIENT I set it to connect it to my iPhone's Hotspot WiFi.

I then had to download the Network Analyzer app.   I selected the LAN tab at the bottom of the screen and tapping the Scan button it told me the IP address for all the devices connected to it. 

I noted the IP address for the Shelly 1 and typed that into Safari on my iPhone to get its Web UI.  Then I went to Settings and then FIRMWARE UPDATE.  Finally it worked!   My Shelly 1 now has firmware version  20210413-155228/v1.10.2-gb89901a 

What a pain in @$$ though...

ScottLeMagicien, 2021-04-15:
So now I'm trying to install the HomKit firmware and it's not working.

I'm using this URL:  http://192.168.1.132/ota?url=http://rojer.me/files/shelly/shelly-homekit-Shelly1.zip

I see this:  {"status":"updating","has_update":false,"new_version":"20210413-155228/v1.10.2-gb89901a","old_version":"20210413-155228/v1.10.2-gb89901a"}

But no matter how long I wait, when I go to its IP (192.168.1.132) I still see the Shelly Web UI...

andyblac, 2021-04-15:
> > > Device Name: None
> > > Model: ShellyDimmer2
> > > Firmware: Stock 1.10.2 ➜ HomeKit Not available
> > > Devices found: 1 Upgradeable: 0 Flashed: 0 Failed: 0
> > 
> > 
> > dimmer2 is not supported in this firmware.
> 
> Thank you for prompt replies!
> 
> Do you know when dimmer2 will be supported?
> 
> Br
> 
> /cj

add your support here https://github.com/mongoose-os-apps/shelly-homekit/issues/14, but dimmer is a tricky device to support, as it can overheat / melt etc if you use wong light type in wrong modes.

timoschilling, 2021-04-15:
can you post the output of  http://192.168.1.132/settings please

ScottLeMagicien, 2021-04-15:
Sure:

{"device":{"type":"SHSW-1","mac":"8CAAB5480898","hostname":"shelly1-8CAAB5480898","num_outputs":1},"wifi_ap":{"enabled":false,"ssid":"shelly1-8CAAB5480898","key":""},"wifi_sta":{"enabled":true,"ssid":"2_nomap","ipv4_method":"dhcp","ip":null,"gw":null,"mask":null,"dns":null},"wifi_sta1":{"enabled":false,"ssid":null,"ipv4_method":"dhcp","ip":null,"gw":null,"mask":null,"dns":null},"ap_roaming":{"enabled":false,"threshold":-70},"mqtt": {"enable":false,"server":"192.168.33.3:1883","user":"","id":"shelly1-8CAAB5480898","reconnect_timeout_max":60.000000,"reconnect_timeout_min":2.000000,"clean_session":true,"keep_alive":60,"max_qos":0,"retain":false,"update_period":30},"coiot": {"enabled":true,"update_period":15,"peer":""},"sntp":{"server":"time.google.com","enabled":true},"login":{"enabled":false,"unprotected":false,"username":"admin"},"pin_code":"","name":null,"fw":"20210413-155228/v1.10.2-gb89901a","factory_reset_from_switch":true,"discoverable":true,"build_info":{"build_id":"20210413-155228/v1.10.2-gb89901a","build_timestamp":"2021-04-13T15:52:28Z","build_version":"1.0"},"cloud":{"enabled":false,"connected":false},"timezone":"Europe/Amsterdam","lat":52.068958,"lng":4.324020,"tzautodetect":true,"tz_utc_offset":7200,"tz_dst":false,"tz_dst_auto":true,"time":"18:37","unixtime":1618504621,"debug_enable":false,"allow_cross_origin":false,"ext_switch_enable":false,"ext_switch_reverse":false,"ext_switch":{"0":{"relay_num":-1}},"actions":{"active":false,"names":["btn_on_url","btn_off_url","longpush_url","shortpush_url","out_on_url","out_off_url","lp_on_url","lp_off_url","ext_temp_over_url","ext_temp_under_url","ext_temp_over_url","ext_temp_under_url","ext_temp_over_url","ext_temp_under_url","ext_hum_over_url","ext_hum_under_url"]},"hwinfo":{"hw_revision":"prod-191217", "batch_id":1},"mode" :"relay","longpush_time":800,"relays":[{"name":null,"appliance_type":"General","ison":false,"has_timer":false,"default_state":"switch","btn_type":"toggle","btn_reverse":0,"auto_on":0.00,"auto_off":0.00,"power":0.00,"schedule":false,"schedule_rules":[]}],"ext_sensors":{},"ext_temperature":{},"ext_humidity":{},"wifirecovery_reboot_enabled":true}

andyblac, 2021-04-15:
> Sure: [...]

> {"enabled":false,"ssid":null,

your not connected to your wifii ap.

ScottLeMagicien, 2021-04-15:
> your not connected to your wifii ap.

Sorry, I don't know what that means.

andyblac, 2021-04-15:
> > your not connected to your wifii ap.
> 
> Sorry, I don't know what that means.

see https://github.com/mongoose-os-apps/shelly-homekit/issues/631 here same issue

closed 2021-04-15 by andyblac