mongoose-os-apps/shelly-homekit #600: Short open and close switch..

marienaz, 2021-03-29:
I have a shelly one connected to my apartment building door open button.The problem is that the apartment system is looking for a impulse (contacts shorting for a small period of time) . Currently, I have the shelly added in as a switch using the rest function in the configuration file. I click the switch on and off quickly to simulate the contacts shorting for a small period of time.

My question: Is there anyway to add in a push button that shorts the shelly 1 for a given time? Maybe 1000ms?...

If I set my shelly as a garage door opener it functions correctly with the door... but I want a switch or button that functions in the same manner?

marienaz, 2021-03-29:
Hi Timo...
Yes i know  about the auto off settings, but then your switch is always off or always on?? and u cant see the actual state of the alarm?

timoschilling, 2021-03-29:
Ok, then this is maybe a double of https://github.com/mongoose-os-apps/shelly-homekit/issues/552

andyblac, 2021-03-29:
sounds more like https://github.com/mongoose-os-apps/shelly-homekit/issues/408 which i dont think is possible, as HoeKit would not know the current alarm state, if it is changed outside of homekit.

marienaz, 2021-03-29:
His allready working if I select the garage door function.. so all that needs to be done is copy all settings of the garage door openener but call it a short contact switch??

timoschilling, 2021-03-29:
But how das the Shelly knows the state of your alarm system?

andyblac, 2021-03-29:
> His allready working if I select the garage door function.. so all that needs to be done is copy all settings of the garage door openener but call it a short contact switch??

it not that easy, as i said,atm the relau is hardcoded the to switch so it reports the RELAY state, and even if you detach it, the 'SWITCH' still show the RELAY state, tere is NO WAY atm to have a switch not showing the relay state.

marienaz, 2021-03-29:
if I only use homekit then to activate and deactivate in stead of a remote, then it stays in sync... same as my garage doors, on the garage doors I can put a magnet but its the same type of switch..

andyblac, 2021-03-29:
> if I only use homekit then to activate and deactivate in stead of a remote, then it stays in sync... same as my garage doors, on the garage doors I can put a magnet but its the same type of switch..

not in code, 'switch' is just that, yes it maybe posible infuture but no atm. but as this is a unique request it will be low priorty.

timoschilling, 2021-03-29:
> if I only use homekit then to activate and deactivate in stead of a remote, then it stays in sync

In theory yes, in practice not. The Shelly can be become out of sync during a reboot or an update, same counts for the alarm system. It only stays in sync if it works to 100%. Else you will be out of sync.
 That's why a garage door opener requires a close sensor, because then the shelly can always read that contact and knows it's closed or not.

andyblac, 2021-03-29:
@marienaz have a read of this issue https://github.com/mongoose-os-apps/shelly-homekit/issues/146 is sound verry similar to what you want ?

andyblac, 2021-03-29:
ok, i'm closing this issue as it is a duplicate of https://github.com/mongoose-os-apps/shelly-homekit/issues/146

closed 2021-03-29 by andyblac