mongoose-os-apps/shelly-homekit #602: [Flashscript] would like your opinion.

andyblac, 2021-04-01:
to those who use my script, i am adding the option to use a config file so you can run the script with saved options, how would you prefer to use it?

1. use flag `--config` to use config file if one exists
2. as default, if a config file exists just use it, and to not use it so you can use command line options use flag `--no-config`

please bare in mined that any command line options will be overruled by config, so if you have `--mode = keep` saved in config then try to use `--config --mode revert` it will use mode option from config.

I dont think it possible to give priority to commandline over config, unless anyone out there knows how ?

andyblac, 2021-04-03:
i have managed to get command line to overrule configs

```
andrews-mbp-wifi:tools andrewblackburn$ ./flash-shelly.py --timeout 360 -l -i3 -m=keep 192.168.3.46 192.168.3.47 192.168.3.20 192.168.3.43 --save-config motionsensors
Saving configuration section motionsensors to .flashrc
andrews-mbp-wifi:tools andrewblackburn$ ./flash-shelly.py --config motionsensors
Reading configuration section motionsensors from .flashrc
Looking for Shelly devices...

Host: http://192.168.3.46
Device Name: Hall Motion Sensor
Model: ShellyMotion
Device ID: shellymotionsensor-60A4239A69B0
SSID: ABFi-IoT-2.4
IP: 192.168.3.46
RSSI: -42
Up Time: 1 day, 12:21:37
Battery: 91%
Firmware: Stock 1.1.0 ✔

Host: http://192.168.3.47
Device Name: Landing Motion Sensor
Model: ShellyMotion
Device ID: shellymotionsensor-60A4239368A2
SSID: ABFi-IoT-2.4
IP: 192.168.3.47
RSSI: -33
Up Time: 12 days, 4:14:50
Battery: 94%
Firmware: Stock 1.1.0 ✔

Host: http://192.168.3.20
Device Name: Back Garden Motion Sensor
Model: ShellyMotion
Device ID: shellymotionsensor-60A423976642
SSID: ABFi-IoT-2.4
IP: 192.168.3.20
RSSI: -41
Up Time: 10:55:41
Battery: 95%
Firmware: Stock 1.1.0 ✔

Host: http://192.168.3.43
Device Name: Front Garden Motion Sensor
Model: ShellyMotion
Device ID: shellymotionsensor-60A4239752B4
SSID: ABFi-IoT-2.4
IP: 192.168.3.43
RSSI: -55
Up Time: 14:43:22
Battery: 96%
Firmware: Stock 1.1.0 ✔

Devices found: 4 Upgradeable: 0
```

as you see below adding `-i2` to command line overrules `-i3` saved in config.
```
andrews-mbp-wifi:tools andrewblackburn$ ./flash-shelly.py --config motionsensors -i2
Reading configuration section motionsensors from .flashrc
Looking for Shelly devices...

Host: http://192.168.3.46
Device Name: Hall Motion Sensor
Model: ShellyMotion
Firmware: Stock 1.1.0 ✔

Host: http://192.168.3.47
Device Name: Landing Motion Sensor
Model: ShellyMotion
Firmware: Stock 1.1.0 ✔

Host: http://192.168.3.20
Device Name: Back Garden Motion Sensor
Model: ShellyMotion
Firmware: Stock 1.1.0 ✔

Host: http://192.168.3.43
Device Name: Front Garden Motion Sensor
Model: ShellyMotion
Firmware: Stock 1.1.0 ✔

Devices found: 4 Upgradeable: 0
```

rojer, 2021-04-03:
(1), i find (2) to be confusing.

andyblac, 2021-04-06:
can i get a vote on who would use this feature. just thumbs up on main post please.

andyblac, 2021-04-07:
closing as not wanted.

closed 2021-04-07 by andyblac