mongoose-os-apps/shelly-homekit #603: Shelly1L stop working after update (v.2.8)

phramos, 2021-04-01:
Hello,

I have one Shelly1L installed with Neutral to control one light. It was working fine until I update the firmware to use 2.8 version (the previous one was 2.7.3 build 20210124-130547/2.7.3-g883914a-master). After the update, the Shelly 1L was working as expected but few minutes ago, when I turned the lights on using my homepod the light is not turning off anymore (not even if I toggle the switch). Can anyone kindly help me with this issue or guide me to downgrade to the 2.7 version? Thanks

timoschilling, 2021-04-01:
which `2.8` are you using? `2.8.0`, 2.8.1` or `2.8.2`. It should be listed in the accessory settings in the home app.

phramos, 2021-04-01:
Hey timoschilling, thanks for your fast reply. Since I updated earlier today I am using the version 20210315-093718/2.8.2-gc0f7426-master.

Thanks

andyblac, 2021-04-01:
can you here a click from the shelly when using either switch or hk, it sounds like, the relay is stuck, it can happen on Shelly1L, you may need to give the device a knock

andyblac, 2021-04-01:
> Whowwww It works! But as soon as I close the switch box it happens again. Is there anything that I can do to avoid this behavior? Many thanks

did it not happen on 2.7.3 ?,

andyblac, 2021-04-01:
you can revert to 2.7.3 by downloading zip here https://github.com/mongoose-os-apps/shelly-homekit/releases/tag/2.7.3 and uploading it via webui

phramos, 2021-04-01:
No. This has never happened before. I will try your suggested solution. Thanks

phramos, 2021-04-01:
I applied the previous version and the issue still happen. Probably there is a phisical issue with my Shelly 1L. Thanks andyblac, for the help. I am going to close the thread.

closed 2021-04-01 by phramos

andyblac, 2021-04-01:
try it on stock, if still happens (prob will it a known fault on some) report it to shelly.