mongoose-os-apps/shelly-homekit #604: website of shelly shows only "disconnected" and reloads

buttonger, 2021-04-01:
Hello,

after updating to Homekit 2.8.2 or 2.7.3 by browser or script the website shows "disconnected" and reloads several times a second. In that state I could revert to the stock-firmware. Afterwards I tried again to update to the homekit-firmware. and the browser again shows the same (disconnected and reload).
What can I do?

timoschilling, 2021-04-02:
you use post the output of `http://<ip>/rpc/Shelly.GetInfo`

buttonger, 2021-04-02:
`{"device_id": "shellyswitch25-B17760", "name": "shellyswitch25-84CCA8B17760", "app": "switch25", "model": "Shelly25", "stock_model": "SHSW-25", "host": "shellyswitch25-84CCA8B17760.local", "version": "2.8.2", "fw_build": "20210315-093826/2.8.2-gc0f7426-master", "uptime": 49, "failsafe_mode": false, "wifi_en": true, "wifi_ssid": "AUS", "wifi_pass": "***", "wifi_rssi": -63, "wifi_ip": "192.168.1.80", "wifi_ap_ssid": "shellyswitch25-84CCA8B17760", "wifi_ap_ip": "192.168.33.1", "hap_cn": 14, "hap_running": false, "hap_paired": false, "hap_ip_conns_pending": 0, "hap_ip_conns_active": 0, "hap_ip_conns_max": 12, "sys_mode": 0, "wc_avail": true, "gdo_avail": true, "debug_en": false, "sys_temp": 56, "overheat_on": false, "components": [{"id": 1, "type": 0, "name": "Shelly SW1", "svc_type": 0, "valve_type": 1, "in_mode": 1, "in_inverted": false, "initial": 3, "state": false, "auto_off": false, "auto_off_delay": 0.000, "state_led_en": -1, "out_inverted": false, "apower": 0.000, "aenergy": 0.000}, {"id": 2, "type": 0, "name": "Shelly SW2", "svc_type": 0, "valve_type": 1, "in_mode": 1, "in_inverted": false, "initial": 3, "state": false, "auto_off": false, "auto_off_delay": 0.000, "state_led_en": -1, "out_inverted": false, "apower": 0.000, "aenergy": 0.000}]}`

timoschilling, 2021-04-02:
`"uptime": 49` looks like your shelly reboots in short intervals. Please check that page multiple times and take a look how long it take until the `uptime` value starts by 0.

buttonger, 2021-04-02:
It doesn't a reboot. But I flashed the device only a few seconds before the post. 
uptime ist now "uptime": 1403

timoschilling, 2021-04-02:
please open `http://<ip>/debug/log?follow=1` wait until it's loaded and then open the ui for 1-2 min and post the log then

buttonger, 2021-04-02:
The debug-command-page is not loading.

https://imgur.com/a/N3uhGdh

The Info shows now
"uptime": 13969,

timoschilling, 2021-04-02:
@rojer any ideas?

andyblac, 2021-04-02:
we had this before this certain browser, @buttonger what browser you using ?

buttonger, 2021-04-02:
I tried Safari and Firefox. The Firmware I tried to upload via script and via browser(s)
.

andyblac, 2021-04-02:
> I tried Safari and Firefox. The Firmware I tried to upload via script and via browser(s)
> .

what happends when you do, please post test.txt file
`curl -get http://192.168.3.45 --output test.txt` (use your own ip)

buttonger, 2021-04-02:
<!DOCTYPE html>
<html lang="en">

<head>
  <meta name="viewport" content="width=device-width, initial-scale=1.0">
  <link rel="shortcut icon" type="image/x-icon" href="favicon.ico">
  
<style>
html, body {
  height: 100vh;
  padding: 0;
  margin: 5;
  background-color: #f2f1f6;
  color: #454955;
  font: 14px/1.5 '-apple-system', 'HelveticaNeue', Helvetica, Segoe UI;
}

* {
  outline: none !important;
}

h1, h2, h3, h4, h5, h6, b, th, strong {
  color: #454955;
}

input, button, div, pre, p {
  font: inherit;
}

button {
  color: white;
  padding: 0.5em 1em;
  border-radius: 18px;
  border: none;
  cursor: pointer;
  background: #0071e3;
}

.button-container{
  text-align: center;
}

button label {
  position: relative;
  cursor: pointer;
  min-width: 10em;
}

a, a:link, a:visited {
  color: #0071e3;
  text-decoration: none;
}

input[type=password], input[type=text] {
  width: 160px;
  padding: 0.2em 0.7em;
  position: relative;
  border: 1px solid #cdcdcd;
  border-color: rgba(0, 0, 0, .15);
  background-color: #ffffff;
  font-size: 14px;
}

h1 {
  margin: 0;
  padding-top: 0.5em;
  text-align: center;
}

.container {
[... 1837 lines cut ...]
      el("update_btn").onclick = function () {
        return downloadUpdate(updateURL, el("update_btn_spinner"), el("update_status"));
      };
    })
    .catch((error) => {
      console.log("Error", error);
      e.innerHTML = errMsg;
      se.className = "";
    });
}

el("update_btn").onclick = function () {
  checkUpdate();
};
el("revert_btn").onclick = function () {
  if(!confirm("Revert to stock firmware?")) return;

  el("revert_msg").style.display = "block";
  var stockURL = `https://rojer.me/files/shelly/stock/${lastInfo.stock_model}.zip`;
  downloadUpdate(stockURL, el("revert_btn_spinner"), el("revert_status"));
};

</script>
</body>
</html>

buttonger, 2021-04-09:
Do you need more information?

andyblac, 2021-04-09:
> Do you need more information?

unfortunately i'm out of ideas, you'll have to wait for @rojer to respond.

timoschilling, 2023-01-19:
Is this still an issue?

closed 2023-02-08 by github-actions[bot]