mongoose-os-apps/shelly-homekit #609: Garage door normally open sensor has wrong order to check state

TA2k, 2021-04-04:
If you run a shelly 1 as garage door and connect the sensor as normally open the order to check the state is the same like "normally closed" but should be reversed

Expected behavior:
Garage door is closed.
Garage door is opened via shelly.
Garage door is shown as moving for the entered movement time
Garage door is open
Garage sensor is connected and homekit shows the garage door as open:

Actual behavior:
Garage door is closed.
Garage door is opened via shelly.
Garage door is shown as closed
Garage door is open
Garage sensor is connected and homekit starts to shows the garage as moving
You have to wait the entered movement time then the garage is shown as open.

If the sensor set to normally open also the start of the movement time has to be set to start of the opening

https://github.com/mongoose-os-apps/shelly-homekit/blob/9729ccc42ee7f3a9a2f334a213b7ec1a922fd287/src/shelly_hap_garage_door_opener.cpp#L347

TA2k, 2021-05-05:
Is there any fix planned?

timoschilling, 2021-05-05:
@rojer what's your meaning?

marcus-aa, 2021-10-02:
> If you run a shelly 1 as garage door and connect the sensor as "normally open" the order to check the state is the same like "normally closed" but should be reversed
>
> Expected behavior: Garage door [...]

No... Shelly 1 has only one sensor and this is ALWAYS the "door is closed" position. Then this sensor (switch) can be configured if the poles are connected or opened when the door has reached the CLOSED position. This is ehat can be configured, also called NC or NO contacts, make or brake or whatever you like.

If you want to ALSO detect the "door is fully opened" position, you need a Shelly 2 or 2.5, where there are 2 inputs, both configurable for the type of switch you are using.

So the expected behavior is more like: 
Garage door is closed. 
Garage door is opened via shelly. 
Closed sensor changes state
Garage door is shown as moving for the entered movement time 
Garage door is assumed to be fully open and homekit shows the garage door as open

/M

TA2k, 2021-10-02:
Would be also ok but at the moment the movement time is shown incorrect

jimbo0207, 2021-12-10:
Same here.. The wheel in the tile keeps on spinning forever, regardless what movement time I do enter. 
when I lock the phone and go back in, state is shown correctly. 

It is correct that movement time is only there in order to change the state to open, right?

rojer, 2022-01-05:
there is no "fully open" sensor. as @marcus-aa explained, the sensor is _always_ the "dorr is closed" sensor, the normally open / normally closed reflects just how it works, whether it is open or closed circuit when not active ("normally" = not activated).

closed 2022-01-05 by rojer